espressif/arduino-esp32 #3409: Flashing ArduinoIDE .bin via Espressif ESP32 Flash Download Tool

bfaliszek, 2019-10-26:
Hi, I have a sketch of the application in the ArduinoIDE. When uploading to ESP32 with the ArduinoIDE level everything works fine. I exported it as a .bin file. I would like to use an external application to upload a .bin file to a board. ArduinoIDE compiles it every time I try to upload the sketch. It takes over 1.5 minutes with my project. I have over 100 ESP32 boards for flashing, so it will take forever for ArduinoIDE.

Using the `esptool.py --port COM4 write_flash 0x1000 my_app.bin` command, the board does not flash correctly.

Any idea how I could flash my ESP32 with .bin file from ArduinoIDE?

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:					        Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
Describe your problem here

chegewara, 2019-10-26:
0x1000 is not correct address, it should be 0x10000 for normal arduino app.

lbernstone, 2019-10-26:
Go into File->Preferences.  There is an option there to show verbose output during upload.  Run an upload, and you will see the exact esptool command the IDE uses to upload your code.

bfaliszek, 2019-12-09:
0x10000 does not work - `esptool.py --port COM4 -b 921600 write_flash 0x10000 FileFronArduinoIDE.bin`
Only uploading via ArduinoIDE works. Output during upload:

> esptool.py v2.6
> Serial port COM4
> Connecting........____
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> MAC: 24:6f:28:96:2a:94
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Compressed 8192 bytes to 47...
> 
> Writing at 0x0000e000... (100 %)
> Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4369.1 kbit/s)...
> Hash of data verified.
> Compressed 17392 bytes to 11186...
> Writing at 0x00001000... (100 %)
> Wrote 17392 bytes (11186 compressed) at 0x00001000 in 0.1 seconds (effective 933.8 kbit/s)...
> Hash of data verified.
> Compressed 1239552 bytes to 702109...
> Writing at 0x00010000... (2 %)
> ...
> Writing at 0x000b8000... (100 %)
> Wrote 1239552 bytes (702109 compressed) at 0x00010000 in 10.8 seconds (effective 918.6 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 129...
> Writing at 0x00008000... (100 %)
> Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 1638.4 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...

lbernstone, 2019-12-09:
If you go into File->Preferences in Arduino IDE, you can turn on verbose logging during upload.  This will show you the exact command sent to esptool.

bfaliszek, 2019-12-09:
@lbernstone I have "verbose logging during upload" enabled, and the only thing ArduinoIDE displays to me is the information from the previous comment.

lbernstone, 2019-12-09:
I don't know what version of the IDE you are using, but mine shows this info just above the output you show above:
```
Sketch uses 260377 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 95444 bytes (29%) of dynamic memory, leaving 232236 bytes for local variables. Maximum is 327680 bytes.
python /home/larryb/Arduino/hardware/espressif/esp32/tools/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/larryb/Arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x1000 /home/larryb/Arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /tmp/arduino_build_752071/sketch_dec06a.ino.bin 0x8000 /tmp/arduino_build_752071/sketch_dec06a.ino.partitions.bin 
```

bfaliszek, 2019-12-09:
@lbernstone Thanks. It's helped. I must upload all 4 .bin files:
/esp32/tools/partitions/boot_app0.bin at 0xe000
/esp32/tools/sdk/bin/bootloader_qio_80m.bin at 0x1000
sketch.ino.bin at 0x10000
sketch_dec.ino.partitions.bin at 0x8000

closed 2020-02-21 by stale[bot]