espressif/arduino-esp32 #3410: Enable DTLS for communication

dobraMorda, 2019-10-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			newest
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Hello!

I need to build DTLS communication with CoAP with ESP32. Static librararies for mbedtls has default disabled DTLS. I need to enable it. Could help me how can I make it? 

Best regards!

lbernstone, 2019-10-26:
The instructions for compiling your own libraries are [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).
[This](https://github.com/espressif/esp32-arduino-lib-builder) is a handy tool for downloading the necessary files.

dobraMorda, 2019-11-04:
> The instructions for compiling your own libraries are [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).
> [This](https://github.com/espressif/esp32-arduino-lib-builder) is a handy tool for downloading the necessary files.

Very thank you. Works :)

closed 2019-11-04 by dobraMorda

alorbach, 2020-03-12:
@dobraMorda thanks to this issue, I could compile the library myself. But I am fighting really hard to get DTLS running. Most of the time it is crashing as it cannot find a shared cipher:

> mbedtls/library/ssl_srv.c():2549]: 0000:  b0 1f 8e a3 58 76 41 21 b6 ba 7d 51 e2 85 97 40  ....XvA!..}Q...@
> mbedtls/library/ssl_srv.c():2549]: 0010:  57 a8 47 c3 e3 2a 9b 75 b5 c9 1b 2f eb 4d 35 06  W.G..*.u.../.M5.
> mbedtls/library/ssl_srv.c():2551]: no session has been resumed
> mbedtls/library/ssl_srv.c():2558]: server hello, chosen ciphersuite: unknown
> mbedtls/library/ssl_srv.c():2560]: server hello, compress alg.: 0x00
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> 

Can you share your mbedtls init /handshake code? That would be great thanks

alorbach, 2020-03-13:
Nevermind I got it working finally. 
I created a wrapper class based on WifiClientSecure, maybe this is help for somebody else: 

https://github.com/alorbach/wificlientsecureudp