espressif/arduino-esp32 #3413: super slow download from arduino ide board manager

ThomasWaldmann, 2019-10-26:
Does your server have a performance or other problem?

I am currently downloading the esp32 boards support from arduino ide's "boards manager" at about 20kB/s. Sometimes it is even much slower than that.

Last try, it aborted after downloading the half of the 69.248kB package and the whole procedure is taking hours already, which is rather annoying.

I have a 250Mbit/s connection here and a speed test shows ok speed for my connection, so it seems to be either the server or the connection to it in the internet.

Debian Linux "stretch", Arduino IDE 1.8.10 64bit.

ThomasWaldmann, 2019-10-26:
https://github.com/espressif/arduino-esp32/releases/download/1.0.4/esp32-1.0.4.zip

This is from the console log. So, is github slow or throttling connections?

lbernstone, 2019-10-26:
How exactly would you like this issue resolved?

ThomasWaldmann, 2019-10-26:
Does it matter? If it is resolved at all, I'm fine.

WereCatf, 2019-10-27:
> Does your server have a performance or other problem?

Github is run by Espressif, it's not "their server." Github is owned by Microsoft.

> I have a 250Mbit/s connection here and a speed test shows ok speed for my connection, so it seems to be either the server or the connection to it in the internet.

And on the other hand, I just tested it and I got 9.78MBps, or 78.24Mbps.

The issue could very well be e.g. a routing-problem at your ISP or somewhere between your ISP and Github, for example. It's not necessarily anything to do with Github itself.

ThomasWaldmann, 2019-10-27:
I can test it monday from another ISP to see if it can be reproduced.

ThomasWaldmann, 2019-12-26:
Hmm, still need to test it again. If somebody else gets to it before me, feel free.

closed 2020-03-09 by stale[bot]

fusionstream, 2020-04-19:
Obviously this is not espressif's problem but thought I'd chime in here that I'm having the same issue. Unbelievably slow. Visiting the sites is fine. Tried downloading the release directly via a download manager and that too is incredibly slow. Slower in fact. 3.4KBps is the peak and it's constantly cutting out. Each request to add a new chunk is replied with a 403.

bepitulaz, 2020-04-19:
I'm also having the same issue. Downloading is very very very slow.

canDry, 2020-04-28:
Ditto.  Getting ridiculously throttled by somebody!!

ThomasWaldmann, 2020-04-28:
Considering that likely a lot of people are seeing this: reopen issue?

chegewara, 2020-04-28:
Reopen issue wont change too much. Its problem with slow github server and solution is to use different servers provider, but its no point to use github in that case.

hikmahgumelar, 2020-05-12:
it's other server to download? Download too slow from my country

ThomasWaldmann, 2020-05-12:
@igrr reopen and try to provide a faster distribution mechanism?

lbernstone, 2020-05-12:
If you could build a worldwide CDN to provide your 50MB of free software, that'd be great.

GasAlbert, 2020-05-15:
Actually seems to be the espressif server thats extreamly slow:

s-a-tanjim, 2020-08-09:
I had the same issue. 
I downloaded those file separately from the browser(as it was faster than downloading from Arduino IDE) 
1. https://github.com/espressif/arduino-esp32/releases/download/1.0.4/esp32-1.0.4.zip
2. https://dl.espressif.com/dl/esptool-2.6.1-windows.zip
3. https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
4. https://github.com/igrr/mkspiffs/releases/download/0.2.3/mkspiffs-0.2.3-arduino-esp32-win32.zip
Download this 4file from given link. and copy those to "C:\Users\[UserName]\AppData\Local\Arduino15\staging\packages\" directory.
Then open Arduino Board Manager, Search for "esp32", and click install. As those files were already downloaded, so this time it will just install those files.
Note: If you can't find "esp32" in board manager, make sure you added this link : "https://dl.espressif.com/dl/package_esp32_index.json" into "Additional Board Manager URLs" (You will find this option in File>Preference)

atanisoft, 2020-08-09:
> 4\. Note: If you can't find "esp32" in board manager, make sure you added this link : "https://dl.espressif.com/dl/package_esp32_index.json" into "Additional Board Manager URLs" (You will find this option in File>Preference)

Note that link is not valid or supported. The correct link is https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json as documented [here](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md).

AugustoOspal, 2020-09-23:
I've just found the solution. I was trying to download the file from Argentina, but then I tried by downloading it using a VPN, and it worked perfectly.

dropthemasquerade, 2020-10-19:
> I had the same issue.
> I downloaded those file separately from the browser(as it was faster than downloading from Arduino IDE)
>
> 1. [...]

I tried this solution yesterday, but it automatic remove what I have done, and do download itself again! so I think it maybe have a mark or flag, which I need to set, and then I can skip download from internet. But I can't find that right now. Can anyone help?

andialles, 2020-12-22:
I had that issue as well, solved it by using a VPN to Hong Kong...

tomg404, 2020-12-25:
> 
> 
> I had that issue as well, solved it by using a VPN to Hong Kong...

Solved the issue for me as well!

RandomVertebrate, 2020-12-29:
The VPN does seem to help. Routing through Singapore rn.

applehat, 2021-01-04:
Just adding that I too am downloading at a somewhat reasonable speed now that I'm VPN'
d to Hong Kong

jacoblann, 2021-01-07:
> I had the same issue.
> I downloaded those file separately from the browser(as it was faster than downloading from Arduino IDE)
>
> 1. [...]

This method works really well for me! Thanks!

spielzimmer3000, 2021-01-09:
I connected to my smartphone's hotspot before installing the esp8266 package in Boards Manager and it worked very well. Connected to my WiFi at home it was unbelievably slow.

RandomVertebrate, 2021-01-10:
For me what finally worked was restarting the download late at night (3am)

ChR-iSz, 2021-01-11:
Same here. Need 3 Days to download! What a shit!

me-no-dev, 2021-01-12:
links for the upcoming releases have been switched to github. You can use the dev package manager json to give them a shot.

lbernstone, 2021-01-12:
Are we going to open a new issue for complaints about slow downloads in China, or can we reuse this one?

me-no-dev, 2021-01-12:
we will need to have separate json hosted there and pointing to the files there for CN users.

xlla, 2021-03-15:
> I had the same issue.
> I downloaded those file separately from the browser(as it was faster than downloading from Arduino IDE)
> 
> 1. https://github.com/espressif/arduino-esp32/releases/download/1.0.4/esp32-1.0.4.zip
> 2. https://dl.espressif.com/dl/esptool-2.6.1-windows.zip
> 3. https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
> 4. https://github.com/igrr/mkspiffs/releases/download/0.2.3/mkspiffs-0.2.3-arduino-esp32-win32.zip
>    Download this 4file from given link. and copy those to "C:\Users[UserName]\AppData\Local\Arduino15\staging\packages" directory.
>    Then open Arduino Board Manager, Search for "esp32", and click install. As those files were already downloaded, so this time it will just install those files.
>    Note: If you can't find "esp32" in board manager, make sure you added this link : "[https://dl.espressif.com/dl/package_esp32_index.json"[*]](https://dl.espressif.com/dl/package_esp32_index.json%22%5B*%5D) into "Additional Board Manager URLs" (You will find this option in File>Preference)
> 
> [*]Update: Use this link "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json" as documented here.

For mac user, where is the corresponding folder of "C:\Users[UserName]\AppData\Local\Arduino15\staging\packages"

asdf23, 2021-07-02:
Japan is now working as well.. it's not slow, it halts after an exact amount around 40,000kb