espressif/arduino-esp32 #3416: Stack Overflow!!!

areukiddingmestupidchineses, 2019-10-27:
Either your server is standing in a garden of a small town house wired up with unisolated copper wires, OR your server is running Windows Server 1995... I started downloading the ESP32 package yesterday. Today it's on like 20%. WHAT THE FUCK? get that fixed please.. I have a 50k network line. it's nothing on me.

stickbreaker, 2019-10-27:
@areukiddingmestupidchineses politeness is a prerequisite, GitHub is a MicroSoft service. There is nothing wrong on GitHub's end [GitHub Status](https://www.githubstatus.com/)

This is a public space, not your mother house. We do not need to hear how you talk to your mother.

Chuck.

closed 2019-10-27 by stickbreaker

areukiddingmestupidchineses, 2019-10-27:
as this is a Microsoft service that should probably kick your ass out of
the support system as you're not even caring about issues instead all you
talk about is me being "un polite" wow sorry if you got insulted why don't
you visit you mom then and cry?