espressif/arduino-esp32 #3419: SNTP crashes when using GSM

VXConsulting, 2019-10-28:
### Hardware:
Board:							TTGO_CALL 1.3
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Hello,

I am trying to use AzureIoT library => Esp32MQTTClient with TinyGSM on a TTGO TTCALL SIM800.
It works on WIfi but in GSM (2G), SNTP gives an error and crashes.

Kind Regards,

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```
void setup()
{
  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // Set-up modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);

  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.init();

 String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  // Unlock your SIM card with a PIN if needed
  if (strlen(simPIN) && modem.getSimStatus() != 3)
  {
    modem.simUnlock(simPIN);
  }

  //CONNECTION TO GPRS
  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork(240000L))
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  if (modem.isNetworkConnected())
  {
    SerialMon.println("Network connected");
  }

  SerialMon.print(F("Connecting to APN: "));
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass))
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  Serial.println(" > IoT Hub");
  
   Esp32MQTTClient_SetOption(OPTION_MINI_SOLUTION_NAME, "VXRacerIoT");
   Serial.println(" > ESP32 MQTT CLIENT INIT");
   Esp32MQTTClient_Init((const uint8_t *)connectionString, true);
  // Esp32MQTTClient_SetSendConfirmationCallback(SendConfirmationCallback);
  // Esp32MQTTClient_SetMessageCallback(MessageCallback);
}

void loop()
{
   Esp32MQTTClient_Check();
  delay(1000);
}
```

### Debug Messages:
```
[Info] Opened the serial port - COM5
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Initializing modem...

****************************
Setting Certificate: OK
****************************
Modem: SIM800 R14.18
Waiting for network... OK
Network connected
Connecting to APN: mworld.be OK
 > IoT Hub
 > ESP32 MQTT CLIENT INIT
Info: Initializing SNTP

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x400880a9 on core 1

Backtrace: 0x4008b8c0:0x3ffb1d90 0x4008baf1:0x3ffb1db0 0x400880a9:0x3ffb1dd0 0x40107a6e:0x3ffb1e10 0x40107c16:0x3ffb1e30 0x400fa068:0x3ffb1e50 0x400fa0d1:0x3ffb1e70 0x40100f85:0x3ffb1e90 0x40100faa:0x3ffb1eb0 0x400f9f2b:0x3ffb1ed0 0x400d4e22:0x3ffb1ef0 0x400d4e96:0x3ffb1f10 0x400d4056:0x3ffb1f30 0x400d3cca:0x3ffb1f70 0x400dff9f:0x3ffb1fb0 0x400883bd:0x3ffb1fd0

Rebooting...
```

atanisoft, 2019-10-28:
SNTP has a dependency on wifi or eth. Without one of those active it will not be possible to start it. There is no support for using gsm modems as an internet connection by default, it may be possible with esp-idf using LwIP PPP support though.

VXConsulting, 2019-10-28:
Thanks, it is clearer :-)

I am trying to run the pppos_client example on TTGO TCALL card. 
Not working.
I believe I missed  RTS and CTX pin configuration.
I did not find them.

closed 2020-01-10 by stale[bot]