espressif/arduino-esp32 #3420: error while receiving data from a tcp connection

lupine37, 2019-10-28:
### Hardware:
Board:							?ESP32 DEV Module?
Core Installation version:			?ESP32-WROM-32?
IDE name:						?Arduino IDE?
Flash Frequency:					?80Mhz?
PSRAM enabled:                                    ?no? ?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10?

### Description:
Describe your problem here
I am receiving long bytes of data  from a tcp connection. I formed a byte array to store the data and then print the data. When I try to print the data from the array ESP32 resets.  It worked with a smaller byte array size. The compile detail says I have 300kb space for local variable.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

#include <WiFi.h>

WiFiClient client;

int ledPin = 5;
String recvString = "";

const int port = 8888;  //Port number
const IPAddress server(192, 168, 1, 129);

//Server connect to WiFi Network
const char *ssid = "MW40VD_19E7";  //Enter your wifi SSID
const char *password = "Cazp2x6U7t3A9jXg";  //Enter your wifi Password

void writeString(String stringData) {
        for (int i = 0; i < stringData.length(); i++) {
                client.write(stringData[i]);
        }
}

uint8_t recvWithStartEndMarker() {
  boolean endData = false;
  const byte numChars = 32;
  char receivedChars[numChars];
  static boolean recvInProgress = false;
  static byte ndx = 0;
  char startMarker = '<';
  char endMarker = '>';
  char rc;
      while(client.available()>0){
        rc = client.read();
        if (rc == startMarker) {
          recvInProgress = true;
         }
        else if (recvInProgress == true) {
          if (rc != endMarker) {
              receivedChars[ndx] = rc;
              ndx++;
         
              if (ndx >= numChars) {
              ndx = numChars - 1;
              }
          }
          else {
            ndx = 0;
            recvInProgress = false;
            uint8_t dataRecv = atoi(receivedChars);
            return dataRecv;
          }
         
        }
      }
}

void imgbyteArray(){
  uint8_t receivedByte[96000];
  uint32_t starttime = millis();
  int i = 0;
  long calc_data = 0;
  int a = 0;
  int b = 0;
  int c = 0;
//  while(i < 4) {
//    if (client.available()) {
//      uint8_t imgData = recvWithStartEndMarker();
//      calc_data = (imgData * pow(256, i)) + calc_data;
//      i++;
//    }
//  }
//  Serial.println(String(calc_data));
  while(i < 96000 && (millis() - starttime < 22000)) {
    if (client.available()) {
      uint8_t imgData = recvWithStartEndMarker();
      receivedByte[i++] = imgData;
    }
  }
  Serial.println(i);
  for (int j = 0; j < i; j++) {
    printHex(receivedByte[j], 2);
  }
 
}

void printHex(int num, int precision) {
    char tmp[16];
    char format[128];
 
    sprintf(format, "%%.%dX", precision);
 
    sprintf(tmp, format, num);
    Serial.print(tmp);
}

void setup()
{
  Serial.begin(115200);
  pinMode(ledPin, OUTPUT);
  digitalWrite(ledPin, LOW);
  Serial.println();

  WiFi.begin(ssid, password); //Connect to wifi
 
  // Wait for connection 
  Serial.println("Connecting to Wifi");
  while (WiFi.status() != WL_CONNECTED) {   
    delay(500);
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);

  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

}

void loop() {
  char imgchr;
    if (client.connect(server, port)) {
    Serial.println("Client Connected");
    digitalWrite(ledPin, HIGH);
  }
  else {
    Serial.println("No Connection");
    delay(500);
    return;
  }
   while(client.connected()){
       while(client.available()>0){
              imgbyteArray();
       }
      while(Serial.available()>0)
      {
         client.write(Serial.read());
      }
   }
    client.stop();
    Serial.println("Client disconnected");   
    digitalWrite(ledPin, LOW);
 
}
```

### Debug Messages:
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
```

lbernstone, 2019-10-28:
The code you posted doesn't create any large array.  A couple coding practices should help you.
1) Don't pass strings b/w functions.  This causes a lot of copies of large objects.  That is what pointers are designed for.
2) Declaring a variable in a function (a task actually) will use stack memory.  This is quite limited.  Either declare them globally, or use malloc to allocate heap memory to them within the function.

lupine37, 2019-10-31:
Thanks for reply, I will into that and what about the issue of restarting while I while start to receive data.

closed 2020-01-13 by stale[bot]