espressif/arduino-esp32 #3421: preferences.begin gives runtime error:  begin(): nvs_open failed: NOT_INITIALIZED

tehuster, 2019-10-28:
### Hardware:
Board:							ESP32-WROOM-32
Core Installation version:			where can i find this?
IDE name:						Platform.io
Flash Frequency:					where can i find this?
PSRAM enabled:                                     where can i find this?
Upload Speed:				       115200 
Computer OS:                                         Windows 10

### Description:
When i use preferences.begin i get a runtime error: [E][Preferences.cpp:49] begin(): nvs_open failed: NOT_INITIALIZED. 

### Sketch: 
```
#include <Arduino.h>
#include <Preferences.h>
#include <Puzzle.h>

Preferences preferences;

void Puzzle::BlinkLed()
{
    Serial.println("Turn led on..");
    Serial.print("Wait for ");
    Serial.println(blinkTime);
    Serial.println("Turn led off");    
}

void Puzzle::setBlinkTime(int bTime)
{
    Puzzle::blinkTime = bTime;    
    preferences.begin("puzzle", false);
    preferences.putUInt("blinkTime", Puzzle::blinkTime);
    preferences.end();
}

int Puzzle::getBlinkTime()
{
    return Puzzle::blinkTime;
}

void Puzzle::loadVariables()
{
    Serial.println("Getting prefernces variables");
    preferences.begin("puzzle", false);
    Puzzle::blinkTime = preferences.getUInt("blinkTime", 0);
    preferences.end();
}
```
```
#include <Arduino.h>
#include <TCP.h>
#include <JSON.h>
#include <Request.h>

TCP tcp;
JSON json;
Puzzle puzzle;
Request request(puzzle);  

long previousMillis = 0; 

int blinkTime = 1000;

void setup()
{
    Serial.begin(115200);
    delay(10);

    tcp.connectWifi();
    json.createStatus();   
}

void loop()
{   
    if(tcp.newMessage)
    {
        request.request = json.parseJson(tcp.messageRX);
        request.handleRequest();
        tcp.newMessage = false;
    }
    
    unsigned long currentMillis = millis();
    if(currentMillis - previousMillis > tcp.interval) {
        previousMillis = currentMillis; 
        tcp.sendData(json.jsonTX);
    }
}

```
### Debug Messages:
```
[E][Preferences.cpp:49] begin(): nvs_open failed: NOT_INITIALIZED
```

tehuster, 2019-10-28:
I figured out it's because i need to initialize the storage driver. I tried to initialize it by adding this to preferences.begin: 
```
    esp_err_t err_init = nvs_flash_init();
    if (err_init == ESP_ERR_NVS_NO_FREE_PAGES || err_init == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err_init = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err_init );
```
This however triggered to esp to go in a reboot loop.

tehuster, 2019-10-28:
@lbernstone helped me on gitter.im. It's a namespace issue!

zkarcher, 2019-11-10:
I'm experiencing this issue now, and seeing `nvs_open failed: NOT_INITIALIZED`

What was the solution? Thanks!

zkarcher, 2019-11-10:
I figured it out. For anyone experiencing this issue: Your `Preferences` object must be declared in your top-level `.ino` file.

In the `init()` method, call `begin()`, and never call `end()`:

`prefs.begin("your-namespace");`

I had a separate class (not the main .ino sketch) which created the Preferences object. I can't explain *why* this fails, but moving the Preferences object to the top-level .ino file fixed the issue.

If you want to use the Preferences object in a class: You can pass it in as a pointer. I'm storing the pointer, and my class instance can `get` and `put` preferences data, without issues.

closed 2020-01-24 by stale[bot]

veshus, 2020-04-27:
> @lbernstone helped me on gitter.im. It was a namespace issue, nothing to do with the library it was a user issue.

Can you tell, what kind of issue? I have the same problem, and I'm stuck on it.

lbernstone, 2020-04-27:
Read @zkarcher comment or post a new issue.

mark-locatechs, 2020-06-29:
nvs_open in Preferences::begin will only work if nvs_flash_init is called. You have to make sure nvs_flash_init is executed before your code.
By arduino-esp32, nvs_flash_init is executed in initArduino which is just before the setup function.

sauvant, 2021-11-30:
I have had the very same problem ("nvs_open failed: NOT_INITIALIZED") when encapsulating preferences.begin() in a class structure. Meanwhile I found out that it only occurs when "begin()" is called in the constructor. Totally makes sense together with the latest comment from mark-locatechs: constructing of the objects happens too early, Arduino framework is not worked through yet. I experienced similar behaviour earlier, will be more careful doing stuff in the constructors in future...

c---, 2023-10-07:
nvs_flash_init and some of the other comments above can work but they don't actually need to be used. It will work like it's suppose to if you do everything in the "correct" order at the "correct" time.

This really needs to be fixed. It should "just work" instead of needing to worry about the order of function calls and order of object instantiations. The current implementation is broken.