espressif/arduino-esp32 #3422: Flash Read ERROR 1000 - ets_main.c 371

desmatron, 2019-10-29:
### Hardware:
Board:							ESP32 Dev Module - ESP32D0WDQ6 (revision 1)
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Flash Mode:					        DIO
PSRAM enabled:                                      no
Upload Speed:					115200 
Computer OS:                                         Windows 10

### Description:
Uploading this sketch goes fine but listening later on COM3 through Arduino IDE Monitor, I get continuosly this error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

### Sketch: 
```cpp

#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#define SEALEVELPRESSURE_HPA (1013.25)

Adafruit_BME280 bme;

unsigned long delayTime;

void setup() {
  Serial.begin(9600);
  Serial.println(F("BME280 test"));

  bool status;

  status = bme.begin(0x76);  
  if (!status) {
    Serial.println("Could not find a valid BME280 sensor, check wiring!");
    while (1);
  }

  Serial.println("-- Default Test --");
  delayTime = 1000;

  Serial.println();
}

void loop() { 
  printValues();
  delay(delayTime);
}

void printValues() {
  Serial.print("Temperature = ");
  Serial.print(bme.readTemperature());
  Serial.println(" *C");
    
  Serial.print("Pressure = ");
  Serial.print(bme.readPressure() / 100.0F);
  Serial.println(" hPa");

  Serial.print("Approx. Altitude = ");
  Serial.print(bme.readAltitude(SEALEVELPRESSURE_HPA));
  Serial.println(" m");

  Serial.print("Humidity = ");
  Serial.print(bme.readHumidity());
  Serial.println(" %");

  Serial.println();
}
```

lbernstone, 2019-10-29:
Do you have something connected to gpio6-11?  They are reserved. [PINMAP](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap).
Try uploading with board parameters: Flash Mode DIO, Flash Freq 40, Upload speed 115200

desmatron, 2019-10-29:
> Do you have something connected to gpio6-11? They are reserved. [PINMAP](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap).
> Try uploading with board parameters: Flash Mode DIO, Flash Freq 40, Upload speed 115200

No nothing connected on those pins, I've 3.3v + GND + SDA on GPI021 and SCL on GPI022

Changed to DIO, but after upload the problem persists the same:

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`
`flash read err, 1000`
`ets_main.c 371`

lbernstone, 2019-10-29:
Check if your chip is limited to 160MHz. https://github.com/espressif/arduino-esp32/issues/3350#issuecomment-541537391

desmatron, 2019-10-29:
No chip limitation found, but I've checked [pinmap ](https://github.com/espressif/arduino-esp32/blob/master/docs/esp32_pinmap.png)instead and I see that is different from the one provided in the ESP32 pinmap I bought. Just trying now to follow your pinmap I'm facing this on monitor:

_rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac_

atlask1, 2019-11-28:
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

This error happen when you have nothing flashed on .

closed 2020-02-10 by stale[bot]

wotupfoo, 2021-04-24:
I have the same issue.
My non BLE program runs fine programmed over serial or over esp-prog.
When I run with the BLE program (which is quite a bit bigger) I get this reboot error.
I get it with serial programming and with esp-prog.

This is the BLE program size.

Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.3% (used 30564 bytes from 327680 bytes)
Flash: [========  ]  75.1% (used 984834 bytes from 1310720 bytes)
Building .pio\build\esp32dev\firmware.bin

chegewara, 2021-04-25:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

erichards-nvidia, 2021-04-26:
> https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

thx but I'm not touching GPIO0 or GPIO2 or the Serial Rx/Tx pins. I'm also not touching the flash pins - in fact on a 30pin ESP32 board they don't even get exposed.
And as I said, other non-BLE programs program and run just fine for both serial bootloader and jtag programming.
Further, the Serial bootloader and the JTAG programming pass the Verify test.

It's really weird. I suspect there is a dependency on another GPIO when you enable BLE that is not documented. I guess i'll take off the SPI0 and SPI1 devices one by one to see if it fixes it. I'll get back to you.

Thx for responding.