espressif/arduino-esp32 #3423: Ble Beacon Scan  ISSUE

miracle0419, 2019-10-29:
I have trying BLE SCAN by ESP32.
but, The result is that one device B cannot scan.
It is current BLEscan value, and Device B  cannot scan even if the setting is changed.
PLS let me no if you have any idea.

  pBLEScan->setInterval(50);
  pBLEScan->setWindow(30);  // less or equal setInterval value
  foundDevices = pBLEScan->start(5 /*scanTime*/, false);

[image]

sachinrana2k11, 2019-11-06:
hi man, it works just you have to change something, look this code it will help https://github.com/sachinrana2k11/Arnowa_NODES_CODES/blob/master/BLE_SCAN_ESP32/src/main.cpp

miracle0419, 2020-01-03:
Thanks for your update. but, still have problem.

Below, company all product  not found.
http://www.skylabmodule.com/products-list/bluetooth-beacons/

As I know this product used  nRF52832 chipset of Nordic.

sachinrana2k11, 2020-01-03:
Can you poat some pictures of the outputs. What are you getting? While scanning those beacons.

miracle0419, 2020-01-03:
Hi Please see below my result. Also, I trying source code for es-idf-v3.2.3/ble_ibeacon and arduino.

chegewara, 2020-01-04:
Hi,
from 1st screenshot you can see that nRF connect does not recognize iBeacon B as a real iBeacon, which means its advertising format is incorrect. This is probably the reason you cant find it with scanner.

miracle0419, 2020-01-09:
Tow testing device works fine on the nrf connect by Nordic.

[image]

closed 2020-03-23 by stale[bot]

tekinyasin02, 2020-04-18:
Hello. I have this problem. Did you solve it?

yuzuka4573, 2020-10-29:
@tekinyasin02 maybe not
I had a similar situation happen to me. I'm trying with the library samples "BLE_iBeacon" and "BLE_Beacon_Scanner", but the scanner doesn't detect BLE_iBeacon.  
Android apps (nRF Connect and Beacon Analyzer) do detect it as an iBeacon. Conversely, BLE_Beacon_Scanner detects commercially available iBeacons, but I still haven't figured out what's causing it.

siifuu, 2021-03-26:
@yuzuka4573 Hello! I have the exact same issue, have you figured out its cause?

yuzuka4573, 2021-03-30:
@siifuu 
Partially solved. If you don't use Deepsleep, the iBeacon will send successfully, but the error itself doesn't go away.( https://github.com/espressif/arduino-esp32/issues/4462 )