espressif/arduino-esp32 #3426: undefined reference to `mbedtls_chachapoly_init`

gmag11, 2019-10-29:
### Hardware:
Board:							LolinD32
Core Installation/update date:		30/10/2019
IDE name:						Visual Studio 2019
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
I'm developing a communication system on ESP32. I'm trying to test chachapoly with mbedtls. I've written a simple program to test encryption and decryption.

When trying to compile it I get several "undefined reference" linking errors:
```
ESP32crypt.cpp.o: (.literal._Z5setupv+0x40): undefined reference to mbedtls_chachapoly_starts
ESP32crypt.cpp.o: (.literal._Z5setupv+0x44): undefined reference to mbedtls_chachapoly_setkey
ESP32crypt.cpp.o: (.literal._Z5setupv+0x48): undefined reference to mbedtls_chachapoly_encrypt_and_tag
ESP32crypt.cpp.o: (.literal._Z5setupv+0x4c): undefined reference to mbedtls_chachapoly_free
 
ESP32crypt.cpp.o: In function setup()
ESP32crypt.ino:37: undefined reference to mbedtls_chachapoly_init
ESP32crypt.ino:38: undefined reference to mbedtls_chachapoly_starts
ESP32crypt.ino:41: undefined reference to mbedtls_chachapoly_setkey
ESP32crypt.ino:42: undefined reference to mbedtls_chachapoly_encrypt_and_tag
ESP32crypt.ino:43: undefined reference to mbedtls_chachapoly_free
 
collect2.exe*: error: ld returned 1 exit status
```

Is chachapoly encryption supported on esp32 by mbedtls? If not what is the preferred way to implement this?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "mbedtls/chachapoly.h"

void setup()
{
	uint8_t iv[12] =		{ 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, 0x2B, \
							  0x38, 0xE6, 0x9A, 0x83 };
	uint8_t key[32] =		{ 0xB3, 0xD9, 0x1B, 0xF8, 0xA2, 0x3F, 0xA7, 0xEB, \
							  0x3E, 0x26, 0xD5, 0x34, 0x40, 0xE9, 0x0A, 0x12, \
							  0x73, 0x03, 0x0C, 0x6E, 0xB8, 0x69, 0xA0, 0x7B, \
							  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
	uint8_t aad[21] =		{ 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0x04, 0x65, 0x0E, \
							  0xDD, 0x94, 0x97, 0xB3, 0x84 };
	uint8_t message[49]=	{ 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0xFC, 0x6F, 0x3E, \
							  0xEE, 0x04, 0x5B, 0x35, 0xDE, 0x04, 0xF2, 0xCE, \
							  0x45, 0x31, 0x37, 0x26, 0xD9, 0xEB, 0xAA, 0x32, \
							  0x6B, 0x36, 0x9C, 0x78, 0x82, 0xCD, 0xD6, 0x71, \
							  0xFE, 0x79, 0xB9, 0xDA, 0x23, 0xF9, 0x8F, 0x77, \
							  0x96 };
	uint8_t tag[16] =			{ 0x74, 0xBF, 0x73, 0x37, 0x27, 0x58, 0xF8, 0x9E, \
							  0xD2, 0xF3, 0x84, 0xB6, 0x4B, 0xC5, 0xE9, 0x04 };
	uint8_t cryptMsg[49] = { 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0xA0, 0x0F, 0xCE, \
							  0xE0, 0x2A, 0xA0, 0xD0, 0x7C, 0x98, 0x79, 0x59, \
							  0xA2, 0x1E, 0x25, 0xC6, 0x53, 0x7D, 0xE7, 0x44, \
							  0x82, 0x62, 0x74, 0x32, 0x3E, 0xF4, 0xF3, 0x52, \
							  0xB3, 0x8A, 0xFA, 0x7F, 0x93, 0x3A, 0xB5, 0xAD, \
							  0x9E };
	uint8_t buffer[49];
	uint8_t tagBuffer[16];
	mbedtls_chachapoly_context ctx;

	Serial.begin (115200);

	mbedtls_chachapoly_init (&ctx);
	if (mbedtls_chachapoly_starts (&ctx, iv, MBEDTLS_CHACHAPOLY_ENCRYPT)) {
		Serial.printf ("Start error");
	}
	mbedtls_chachapoly_setkey (&ctx, key);
	if (mbedtls_chachapoly_encrypt_and_tag (&ctx, sizeof (message), iv, aad, sizeof (aad), message, buffer, tagBuffer)) {
		Serial.printf ("Encrypt error");
	}
	mbedtls_chachapoly_free (&ctx);

	Serial.print ("Output: ");
	for (int i = 0; i < sizeof (message); i++) {
		Serial.printf ("%02x ", buffer[i]);
	}
	Serial.print ("Tag: ");
	for (int i = 0; i < sizeof (tag); i++) {
		Serial.printf ("%02x ", tagBuffer[i]);
	}
}

void loop(){}

```

gmag11, 2019-11-02:
I've tried an IDF project on platformio with same resulting errors

lbernstone, 2019-11-03:
This should be opened in esp-idf.  There is no option in the Makefiles to enable it.  I think if you manually go into mbedtls/include/mbedtls/config.h and enable the options there, it will work for you.

gmag11, 2019-11-04:
This is the conde I'm using.

```
#include "mbedtls/chachapoly.h"
#include "driver/uart.h"
#include "esp_log.h"

void setup()
{
	uint8_t iv[12] =		{ 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, 0x2B, \
							  0x38, 0xE6, 0x9A, 0x83 };
	uint8_t key[32] =		{ 0xB3, 0xD9, 0x1B, 0xF8, 0xA2, 0x3F, 0xA7, 0xEB, \
							  0x3E, 0x26, 0xD5, 0x34, 0x40, 0xE9, 0x0A, 0x12, \
							  0x73, 0x03, 0x0C, 0x6E, 0xB8, 0x69, 0xA0, 0x7B, \
							  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
	uint8_t aad[21] =		{ 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0x04, 0x65, 0x0E, \
							  0xDD, 0x94, 0x97, 0xB3, 0x84 };
	uint8_t message[49]=	{ 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0xFC, 0x6F, 0x3E, \
							  0xEE, 0x04, 0x5B, 0x35, 0xDE, 0x04, 0xF2, 0xCE, \
							  0x45, 0x31, 0x37, 0x26, 0xD9, 0xEB, 0xAA, 0x32, \
							  0x6B, 0x36, 0x9C, 0x78, 0x82, 0xCD, 0xD6, 0x71, \
							  0xFE, 0x79, 0xB9, 0xDA, 0x23, 0xF9, 0x8F, 0x77, \
							  0x96 };
	uint8_t tag[16] =			{ 0x74, 0xBF, 0x73, 0x37, 0x27, 0x58, 0xF8, 0x9E, \
							  0xD2, 0xF3, 0x84, 0xB6, 0x4B, 0xC5, 0xE9, 0x04 };
	uint8_t cryptMsg[49] = { 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0xA0, 0x0F, 0xCE, \
							  0xE0, 0x2A, 0xA0, 0xD0, 0x7C, 0x98, 0x79, 0x59, \
							  0xA2, 0x1E, 0x25, 0xC6, 0x53, 0x7D, 0xE7, 0x44, \
							  0x82, 0x62, 0x74, 0x32, 0x3E, 0xF4, 0xF3, 0x52, \
							  0xB3, 0x8A, 0xFA, 0x7F, 0x93, 0x3A, 0xB5, 0xAD, \
							  0x9E };
	uint8_t buffer[49];
	uint8_t tagBuffer[16];
	mbedtls_chachapoly_context ctx;

    const uart_port_t uart_num = UART_NUM_0;
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS,
        .rx_flow_ctrl_thresh = 122,
    };
    // Configure UART parameters
    ESP_ERROR_CHECK(uart_param_config(uart_num, &uart_config));
    const int uart_buffer_size = (1024 * 2);
    QueueHandle_t uart_queue;
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_0, uart_buffer_size, \
                                        uart_buffer_size, 10, &uart_queue, 0));
	mbedtls_chachapoly_init (&ctx);
	if (mbedtls_chachapoly_starts (&ctx, iv, MBEDTLS_CHACHAPOLY_ENCRYPT)) {
		
		ESP_LOGI ("I","Start error");
	}
	mbedtls_chachapoly_setkey (&ctx, key);
	if (mbedtls_chachapoly_encrypt_and_tag (&ctx, sizeof (message), iv, aad, sizeof (aad), message, buffer, tagBuffer)) {
		ESP_LOGI ("I","Encrypt error");
	}
	mbedtls_chachapoly_free (&ctx);

	ESP_LOGI ("I","Output: ");
	ESP_LOG_BUFFER_HEX ("I", buffer, sizeof (message));

	ESP_LOGI ("I","Tag: ");
	ESP_LOG_BUFFER_HEX ("I", tagBuffer, sizeof (tag));
}

void app_main(void){
	setup();
}
```

`config.h` has chachapoly enabled with `#define MBEDTLS_CHACHAPOLY_C`

Compile result is this:
```
Linking .pio\build\esp32dev\firmware.elf
.pio\build\esp32dev\src\esp32cryptidf_main.o:(.literal.setup+0x44): undefined reference to `mbedtls_chachapoly_init'
.pio\build\esp32dev\src\esp32cryptidf_main.o:(.literal.setup+0x48): undefined reference to `mbedtls_chachapoly_starts'
.pio\build\esp32dev\src\esp32cryptidf_main.o:(.literal.setup+0x4c): undefined reference to `mbedtls_chachapoly_setkey'
.pio\build\esp32dev\src\esp32cryptidf_main.o:(.literal.setup+0x50): undefined reference to `mbedtls_chachapoly_encrypt_and_tag'
.pio\build\esp32dev\src\esp32cryptidf_main.o:(.literal.setup+0x54): undefined reference to `mbedtls_chachapoly_free'
.pio\build\esp32dev\src\esp32cryptidf_main.o: In function `setup':
esp32cryptidf_main.c:(.text.setup+0xb7): undefined reference to `mbedtls_chachapoly_init'
esp32cryptidf_main.c:(.text.setup+0xc9): undefined reference to `mbedtls_chachapoly_starts'
esp32cryptidf_main.c:(.text.setup+0xef): undefined reference to `mbedtls_chachapoly_setkey'
esp32cryptidf_main.c:(.text.setup+0x11b): undefined reference to `mbedtls_chachapoly_encrypt_and_tag'
esp32cryptidf_main.c:(.text.setup+0x137): undefined reference to `mbedtls_chachapoly_free'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

lbernstone, 2019-11-04:
Please open an issue upstream.  There's not much I can do for you here.

gmag11, 2019-11-05:
I'll do, thank you

closed 2019-11-05 by gmag11

ln-12, 2021-07-22:
@gmag11 Did you find a solution?

ln-12, 2021-07-22:
@lbernstone could you please take a look at https://github.com/espressif/esp-idf/issues/7306#issuecomment-884911052?

gmag11, 2021-07-24:
> @gmag11 Did you find a solution?

Finally I used Crypto Arduino library https://github.com/rweather/arduinolibs