espressif/arduino-esp32 #3428: arduino as component of esp-idf : BLE library missing references

manuprendlair, 2019-10-30:
Hello,

I build an esp-idf project with arduino as component with visualGDB.
All work well with blink and different wifi project I did.
Yesterday I try the ble uart exemple. and I got some missing refferent when trying to compile.
To solve it I added to esp-idf\v3.2\components\bt\CMakeLists.txt :

in list(APPEND COMPONENT_SRCS :
				   "bluedroid/bta/hf_client/bta_hf_client_api.c"
				   "bluedroid/bta/hf_client/bta_hf_client_main.c"
				   "bluedroid/bta/hf_client/bta_hf_client_act.c"
				   "bluedroid/bta/hf_client/bta_hf_client_rfc.c"
				   "bluedroid/bta/hf_client/bta_hf_client_sco.c"
				   "bluedroid/bta/hf_client/bta_hf_client_sdp.c"
				   "bluedroid/bta/hf_client/bta_hf_client_cmd.c"
				   "bluedroid/bta/hf_client/bta_hf_client_at.c"
and
in if(CONFIG_BLUEDROID_ENABLED)
        list(APPEND COMPONENT_PRIV_INCLUDEDIRS :
                                   bluedroid/bta/hf_client/include

Why this append ? I doubt that I make the good correction at the right place...

Regards.
Emmanuel from France.

me-no-dev, 2019-10-30:
maybe you did not enable BT in menuconfig? you should not add those lines to Arduno's CMakeLists :) it's from a different lib. BTW, use IDF3.2/3.3 if you want all to work

manuprendlair, 2019-10-30:
me-no-dev, thanks for your answer.
So what I did:
 - I take sdkconfig from the espressif github arduino core (arduino-esp32/tools/sdk/sdkconfig) and replace my own in my project.
- I use the ESP-IDF specific comit you use for building the arduino core (comit ID: 977854975)

and at compile time I have:
`bta_hf_client_int.h: No such file or directory	line 93	in C:\SysGCC\esp32\esp-idf\v3.2-arduino\components\bt\bluedroid\main\bte_init.c`

And if I look to the build output I have:

```
...
[295/972] Building C object bt/CMakeFiles/bt.dir/bluedroid/main/bte_init.c.obj
FAILED: bt/CMakeFiles/bt.dir/bluedroid/main/bte_init.c.obj 
C:\SysGCC\esp32\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.2-18-g977854975\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/api/include/api -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/ar/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/av/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/dm/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/gatt/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/hh/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/jv/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/sdp/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/bta/sys/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/device/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/hci/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/osi/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/external/sbc/decoder/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/external/sbc/encoder/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/btc/profile/esp/blufi/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/btc/profile/esp/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/btc/profile/std/a2dp/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/btc/profile/std/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/btc/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/btm/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/gap/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/gatt/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/l2cap/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/sdp/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/smp/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/avct/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/avrc/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/avdt/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/a2dp/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/rfcomm/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/stack/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/bt/bluedroid/common/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/nvs_flash/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/spi_flash/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/mbedtls/port/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/mbedtls/mbedtls/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/lwip/include/apps -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/lwip/lwip/src/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/lwip/port/esp32/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/lwip/port/esp32/include/arch -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/lwip/include_compat -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/vfs/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/esp32/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/driver/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/esp_ringbuf/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/tcpip_adapter/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/esp_event/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/log/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/newlib/platform_include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/newlib/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/freertos/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/app_trace/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/heap/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/soc/esp32/include -IC:/SysGCC/esp32/esp-idf/v3.2-arduino/components/soc/include -g   -Og -std=gnu99 -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -fstack-protector -ggdb [... cut ...]
So this is why I add some line to cmakelist as I explain .With this patch I have no compile error, but when I run the code I have a fail : 
[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error. 

I found this error come from 

`
errRc = esp_bt_controller_init(&bt_cfg); (line 354)`

and in esp_bt_controller_init method, the error come from esp-idf file : components\bt\bt.c, line 1040:

```
    if (btdm_dram_available_region[0].mode == ESP_BT_MODE_IDLE) {
        return ESP_ERR_INVALID_STATE;
    }
```

manuprendlair, 2019-10-30:
So..I found the problem... I made an error on the esp-idf version use to build the arduino core!
for the current 1.04 it the the comit id : d3e562907 not the one I mention in my early message!

I change the esp-idf to the correct version and....it's compile without problem!

Sorry me no dev for disturbance!

closed 2019-10-30 by manuprendlair

youssef-smarto, 2019-11-03:
@manuprendlair im trying to build my esp32 project with arduino (to use the juicy BLE library) as a component but i always endup with errors like 
```
/home/<user>/esp/<project>/components/esp-idf/components/partition_table/Makefile.projbuild:60: warning: undefined variable 'quote'
Building partitions from /home/<user>/esp/<project>/components/arduino/tools/partitions/default.csv...
Partitions defined in '/home/<user>/esp/<project>/components/arduino/tools/partitions/default.csv' occupy 4.0MB of flash (4194304 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.
make: *** [/home/<user>/esp/<project>/components/esp-idf/components/partition_table/Makefile.projbuild:61: /home/<user>/esp/<project>/build/default.bin] Error 2
```
could you please put a link or much better list the steps you followed ?
that would help alot...

atanisoft, 2019-11-03:
@youssef-smarto you need to edit your sdkconfig to set the board's flash size. It is incorrectly set to 2MB and your partition csv is using 4MB. You should be able to run `make menuconfig` and find the setting in the section referenced by the compiler output.