espressif/arduino-esp32 #3429: ESP32 not loading sketch

desmatron, 2019-10-30:
### Hardware:
Board: ESP32 Dev Module - ESP32D0WDQ6 (revision 1)
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 40Mhz
Flash Mode: DIO
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10

### Description:
I'm having many problem with this board, now after a hard reset through:

`esptool.exe --chip esp32 --port COM3 --baud 115200 --before default_reset --after hard_reset erase_flash`

I'm always having this message on COM3, when before the same code worked (just for web server):

_rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8_

additionally sensor BME280 is not visible from the board (SDA on 21 and SCL on 22)
_[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can' t init sda=0, scl=1
Found 0 device(s)._

(pull up are onboarded on BME280)

just uploading a sample sketch is not working at all

### Sketch:  

```cpp

/*
 LEDC Software Fade

 This example shows how to software fade LED
 using the ledcWrite function.

 Code adapted from original Arduino Fade example:
 https://www.arduino.cc/en/Tutorial/Fade

 This example code is in the public domain.
 */

// use first channel of 16 channels (started from zero)
#define LEDC_CHANNEL_0     0

// use 13 bit precission for LEDC timer
#define LEDC_TIMER_13_BIT  13

// use 5000 Hz as a LEDC base frequency
#define LEDC_BASE_FREQ     5000

// fade LED PIN (replace with LED_BUILTIN constant for built-in LED)
#define LED_PIN            5

int brightness = 0;    // how bright the LED is
int fadeAmount = 5;    // how many points to fade the LED by

// Arduino like analogWrite
// value has to be between 0 and valueMax
void ledcAnalogWrite(uint8_t channel, uint32_t value, uint32_t valueMax = 255) {
  // calculate duty, 8191 from 2 ^ 13 - 1
  uint32_t duty = (8191 / valueMax) * min(value, valueMax);

  // write duty to LEDC
  ledcWrite(channel, duty);
}

void setup() {
  // Setup timer and attach timer to a led pin
  ledcSetup(LEDC_CHANNEL_0, LEDC_BASE_FREQ, LEDC_TIMER_13_BIT);
  ledcAttachPin(LED_PIN, LEDC_CHANNEL_0);
}

void loop() {
  // set the brightness on LEDC channel 0
  ledcAnalogWrite(LEDC_CHANNEL_0, brightness);

  // change the brightness for next time through the loop:
  brightness = brightness + fadeAmount;

  // reverse the direction of the fading at the ends of the fade:
  if (brightness <= 0 || brightness >= 255) {
    fadeAmount = -fadeAmount;
  }
  // wait for 30 milliseconds to see the dimming effect
  delay(30);
}
```

atanisoft, 2019-10-30:
how is this different than https://github.com/espressif/arduino-esp32/issues/3424, it appears to be a direct copy/paste of your previous issue which appears to be an unresolved hardware issue and not a code issue.  This issue is likely stemming from the same.

desmatron, 2019-10-30:
> how is this different than #3424, it appears to be a direct copy/paste of your previous issue which appears to be an unresolved hardware issue and not a code issue. This issue is likely stemming from the same.

No because before the scanner was loaded successfully and in that case didn't see the sensor at all, now it's not loading any sketch but only showing that msg

atanisoft, 2019-10-30:
From your sketch above it doesn't init Serial *OR* print anything to it, so I'm not certain what you are expecting on the console.

The rest of the code looks like an exact copy of the LEDSoftwareFade example. If you have an LED connected to GPIO 5 and it does not cycle on/off, something is likely wrong with the hardware not the code.

desmatron, 2019-10-30:
The sketch is going to dimmer the led, but after uploading it nothing happens and I'm seeing that msg

atanisoft, 2019-10-30:
Check your wiring as the above message is the expected output when it loads the code from flash and starts execution of it.

closed 2020-01-12 by stale[bot]