espressif/arduino-esp32 #3430: How to check if Serial.write() sent data ?

GeorgeFlorian, 2019-10-30:
Board: ESP32-EVB

I am trying to send some commands to a radar through a serial connection.
Right now I am capable of reading what the radar sends to the ESP through the same serial.

I've received the command and parameter list from the Radar's manufacturer and I have enabled Serial Port Commands from the Radar settings.

A command looks like this: `SET detectionDirection 0`

As such, I've written in code: `USE_SERIAL1.write("SET detectionDirection 0");`

But it doesn't change the settings of the Radar.

How can I check if I actually sent any data at all ?

Also, why use `USE_SERIAL1.write` and not `USE_SERIAL1.print` ?

lbernstone, 2019-10-30:
George,
I think I have mentioned this before.  You can put a led+resistor on the pin parallel to the line going to your radar device.  You will see the light flash when data moves across the line.

stickbreaker, 2019-10-30:
@GeorgeFlorian `Serial.flush()` will hang until all of queued tx data is sent out the tx pin.  It will also dump any data in the rxQueue. so I would use this code:
```c++
while (Serial.available() >0 ){ // handle input data
   char ch = Serial.read();
 // handle input data
}
Serial.flush(); // wait for tx queue to be sent

//all data has been sent out of uart
```

Chuck.

atanisoft, 2019-10-30:
Note that Serial.flush() will also clear the RX buffer...

GeorgeFlorian, 2019-10-30:
But 

> @GeorgeFlorian `Serial.flush()` will hang until all of queued tx data is sent out the tx pin. It will also dump any data in the rxQueue. so I would use this code:
>
> ```c++
> while (Serial.available() [...]

But I don't want to read, I want to write. I want to send data.

stickbreaker, 2019-10-30:
@GeorgeFlorian I just made a branch [stickbreaker-patch-1](https://github.com/espressif/arduino-esp32/tree/stickbreaker-patch-1) that changes `Serial.flush()` by adding a boolean parameter such that `Serial.flush(true)` does not reset the rx Queue.  

you can try it by following the Development branch instruction [here](https://github.com/espressif/arduino-esp32#installation-instructions) and choosing `stickbreaker-patch-1` instead of `master`

Your code would just be:
```c++
//send command
Serial.flush(true); // wait for all data to be sent out UART
// next operatoin.
```

Chuck.

GeorgeFlorian, 2019-10-30:
> @GeorgeFlorian I just made a branch [stickbreaker-patch-1](https://github.com/espressif/arduino-esp32/tree/stickbreaker-patch-1) that changes `Serial.flush()` by adding a boolean parameter such that [...]

So my code should look like:
```
USE_SERIAL1.write("SET detectionDirection 0");
USE_SERIAL1.flush(true);
```

So send then flush ?

But let's assume that `USE_SERIAL1.write("SET detectionDirection 0");` is the first operation on the Serial. Isn't the serial buffer supposed to be empty ? Why to flush it ?

stickbreaker, 2019-10-30:
Your original question was "How to check if Serial.Write() sent data?". Perhaps we miss interpreted you question? 

There is no way for the data no to be sent after you execute a `Serial.write()`, the only question would be "has the data I added to the output queue been sent out the tx pin?" Since the CPU operates at 240mhz and 115200 baud is 11,500 characters per second the CPU could execute 20,000 instruction per byte sent out the port. 

This means that when you execute "
`USE_SERIAL1.write("SET detectionDirection 0");`
That just one of the two CPU's could execute 24*20,000 or 480,000 instructions before the complete message has been sent out the hardware. 

`Serial.flush(true)` would cause the processor to STALL until that complete message has exited the esp32. It was my understanding that you needed to reconfigure the hardware after the message was sent, and you need to know when message had exited the esp32.

Chuck.

GeorgeFlorian, 2019-10-30:
Ok. Thank you ! I will try it as soon as possible and will keep you posted.

GeorgeFlorian, 2019-10-31:
> @GeorgeFlorian I just made a branch [stickbreaker-patch-1](https://github.com/espressif/arduino-esp32/tree/stickbreaker-patch-1) that changes `Serial.flush()` by adding a boolean parameter such that [...]

I've encountered a problem while trying to use your custom branch. I am using PlatformioIDE and I can't seem to figure out how to make the branch work. I've asked a question about this on their forum. Also, from one reason or another, the same code with the same libraries doesn't work on Arduino.

While waiting for their answer, I have received an answer from the Radar's manufacturer regarding how is the string supposed to look. I need `\r\n` at the and of the string to make it work.
Right now `USE_SERIAL1.write("SET detectionDirection 1\r\n");` works perfectly.

The thing is that since I've opened this thread I've started having doubts about using `flush()`.

What happens if the serial connection is used, at the same time, to read and write ? This can happen, because the radar is set to send data whenever it has any and a user can also send commands through the web-interface to the radar to change a setting.

I thought that using `flush()` before and after any `read()` and `write()` can solve this issue.

What do you think ?

stickbreaker, 2019-10-31:
@GeorgeFlorian 
The only reason to use `flush()` is if you need to wait for a transmit message to be sent out the pins before you do some other operation.  

The current version of `Flush()` will DELETE any received data waiting in the input queue.

Based on:
>What happens if the serial connection is used, at the same time, to read and write ? This can happen, because the radar is set to send data whenever it has any and a user can also send commands through the web-interface to the radar to change a setting.

I would not recommend using `flush()` unless you use the `stickbreaker-patch-1` branch.  You would loose received data each time `flush()` is executed.

Based on this information, I don't think you need to use `flush()` at all.  normally `flush()` is only needed when a hardware reconfig is sequenced with a tx data stream.  Such as a half duplex RS485 data stream, or re configuring the UART to use different pins or baud rate.

You just had a protocol error.

Chuck.

closed 2020-01-13 by stale[bot]

zekageri, 2022-04-07:
I'm using a modbus protocol to communicate with many devices on the bus continously.
My baudrate is 115200 and continously writing and reading to and from multiple devices on the bus. For this iam using an rs232 to 485 adapter.
Before i send my serial data i must pull a pin to high, after i sent it i must pull it low. 

The problem is that my modbus communication is inside a task, running on whatever core it is running on.
When i pull my pin high before sending the data and start sending, everything is fine. But i don't know when the data sent out and after flush i got timeout messages because the device did not get my data, because the esp pulled the pin low too soon.
The flush is not working properly for me.

```C
void mSystem::startWrite(){
    mBusLock();
    digitalWrite(MBUS_TX_EN, HIGH);
}

void mSystem::endWrite(){
    Serial1.flush(true); // true does not help.
    vTaskDelay(0);  // vTaskDelay 0 or 1 does not help.
    digitalWrite(MBUS_TX_EN, LOW);  // <-- this line executes too soon. The bytes are still in transmit and the adapter is shutting down before the data sent out.
    mBusUnLock();
}
void mSystem::rawWrite(uint8_t * data, int length){
    int crc16 = getCRC_16(data, length);
    startWrite();
    Serial1.write( data, length );
    Serial1.write( lowByte(crc16) );
    Serial1.write( highByte(crc16) );
    endWrite();
}
```

I'm on PIO using this package:

```
platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board       = esp32dev
framework   = arduino

platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

VojtechBartoska, 2022-04-07:
@zekageri Consider opening new issue if needed.

zekageri, 2022-04-07:
The problem is the same. But okay. Sorry for hijacking

VojtechBartoska, 2022-04-07:
Let's wait if others contributors here can share their ideas :)

zekageri, 2022-04-07:
Thank you for your response.
I don't want to use modbus library because there is a lot of stuff in the protocol which is not standard for us.
I already using hardwareserial and setting the pins for myself.

```C
#define MBUS_BAUD 115200
#define MBUS_RX 34  //   6.  PIN
#define MBUS_TX 15  //   23. PIN
#define MBUS_TX_EN 5
Serial1.begin(MBUS_BAUD,SERIAL_8N1,MBUS_RX,MBUS_TX);
```

This is my setup. The "modbus" frame is a litt bit modified for my use case so i can not use any library.
I have implemented the critical sections which will handle every possible scenario, but i need a way to detect if the hardware really sent my bytes. :/

Thank you for the examples. I will look at them!