espressif/arduino-esp32 #3431: Can't read anything from the Serial (0)

Speedy2k, 2019-10-30:
I am using Arduino,
I have a sketch that need to read Serial data, my sketch upload well to the ESP32 so i know my rx and tx pin and communications are ok. 

I set the baudrate in both the Arduino terminla to 115200 and in my program: Serial.begin(115200);

I have a function that read check for serial data:
void readSerial(){
  if(((millis() - lastReceivedChar) > 10) && (idx > 0)){
    idx = 0;
    serialBuffer[idx] = '0';
    return;
  }
  if(Serial.available()){
    delay(10);
    Serial.print("Available char in serial buffer: ");
    Serial.println(Serial.available());
  }
  if(Serial.available()){  
    char c = Serial.read();
    Serial.print("Serial Buffer: \"");
    Serial.write(c);
    Serial.println("\"");
    lastReceivedChar = millis();
    if (c != termChar){
      if (idx < length){
        serialBuffer[idx] = c;
        idx += 1;
      }
    }else{
      serialBuffer[idx] = '\0';
      idx = 0;
      haveNewData = true;
    }
  }
}

That is called every loop. I have added some debug code to it so i can see what is received from the serial and There is nothing received.

If i write "Hi! How are you?" in the arduino terminal and click send, the only print i got in  the terminal is that:

> Available char in serial buffer: 1
> Serial Buffer: "

I don't know what to do with that? I have tried forcing setting the pin and config of the Serial 0 but it change nothing!

### Hardware:
Board:							Adafruit Huzzah 32
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz?
PSRAM enabled:                                    don't know
Upload Speed:					921600
Computer OS:                                        Windows 10

Speedy2k, 2019-10-30:
I know this is not a site to debug my code! The fact is i receive nothing from the serial port! The code is there to show what i am doing! This exact same code work on a ESP8266 or a ATMEGA2560!

Benik3, 2019-11-15:
Why you don't use function Serial.readBytesUntil() where you can set termination char, max length and timeout?

closed 2020-01-28 by stale[bot]