espressif/arduino-esp32 #3436: esp_bt_controller_init error 259 on esp-idf project with arduino as component

manuprendlair, 2019-10-30:
I have an esp-idf project with arduino as components. 
I try the ble uart example. Evrything compile now.
But on serial console I have this message when I run the example:

`
[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error. 
`

I found this error come from 

`
errRc = esp_bt_controller_init(&bt_cfg); (line 354)`

and in esp_bt_controller_init method, the error come from esp-idf file : components\bt\bt.c, line 1040:

```
    if (btdm_dram_available_region[0].mode == ESP_BT_MODE_IDLE) {
        return ESP_ERR_INVALID_STATE;
    }
```
I use the sdkconfig from the espressif/esp 32 arduino core and (this time) the esp-idf component is the same who was use to compile the arduino core.

If I test the ble example in a pure arduino project, the example work and I am able to discover the uart service on my phone

Now I dont know where to search.

Emmanuel 
From France.

manuprendlair, 2019-10-30:
This message come when this line is executed
BLEDevice::init("UART Service");

chegewara, 2019-10-30:
First idea is memory shortage.

manuprendlair, 2019-10-31:
@chegewara , the same sketch run on a native Arduino sketch for ESP32.
Normaly all my esp-idf project setup should be same as the one for the actual esp32 arduino core
..but it look it's not.. :-( But I am really not an expert esp-ide toolchain!
Hope me no dev could have an idea of what's happend because I really not know where to search!

chegewara, 2019-11-01:
Strange, code looks like one of example, easy code.

manuprendlair, 2019-11-01:
Ok, so I start from the default sdkconfig and run menuconfig directly this time (not by visualgdb).
I activate ble and bluedroid with default value, here my sdkconfig:

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y
CONFIG_PARTITION_TABLE_SINGLE_APP=y
# CONFIG_PARTITION_TABLE_TWO_OTA is not set
# CONFIG_PARTITION_TABLE_CUSTOM is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_singleapp.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000
CONFIG_PARTITION_TABLE_MD5=y
CONFIG_ENABLE_ARDUINO_DEPENDS=y
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_ARDUINO_RUN_CORE0 is not set
CONFIG_ARDUINO_RUN_CORE1=y
# CONFIG_ARDUINO_RUN_NO_AFFINITY is not set
CONFIG_ARDUINO_RUNNING_CORE=1
# CONFIG_ARDUINO_EVENT_RUN_CORE0 is not set
CONFIG_ARDUINO_EVENT_RUN_CORE1=y
# CONFIG_ARDUINO_EVENT_RUN_NO_AFFINITY is not set
CONFIG_ARDUINO_EVENT_RUNNING_CORE=1
# CONFIG_ARDUINO_UDP_RUN_CORE0 is not set
CONFIG_ARDUINO_UDP_RUN_CORE1=y
# CONFIG_ARDUINO_UDP_RUN_NO_AFFINITY is not set
CONFIG_ARDUINO_UDP_RUNNING_CORE=1
# CONFIG_DISABLE_HAL_LOCKS is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE is not set
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE is not set
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
# CONFIG_ARDUHAL_LOG_COLORS is not set
# CONFIG_ARDUHAL_ESP_LOG is not set
CONFIG_ARDUHAL_PARTITION_SCHEME_DEFAULT=y
# CONFIG_ARDUHAL_PARTITION_SCHEME_MINIMAL is not set
# CONFIG_ARDUHAL_PARTITION_SCHEME_NO_OTA is not set
# CONFIG_ARDUHAL_PARTITION_SCHEME_HUGE_APP is not set
# CONFIG_ARDUHAL_PARTITION_SCHEME_MIN_SPIFFS is not set
CONFIG_ARDUHAL_PARTITION_SCHEME="default"
# CONFIG_AUTOCONNECT_WIFI is not set
# CONFIG_ARDUINO_SELECTIVE_COMPILATION is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_NONE is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_ERROR is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_WARN is not set
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
# CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
CONFIG_LOG_BOOTLOADER_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
# CONFIG_SECURE_BOOT_ENABLED is not set
[... 664 lines cut ...]
[0;32mI (425) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (426) boot: Disabling RNG early entropy source...[0m
[0;32mI (426) cpu_start: Pro cpu up.[0m
[0;32mI (430) cpu_start: Starting app cpu, entry point is 0x4008101c[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (440) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (447) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (453) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (459) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (465) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (471) heap_init: At 3FFC8FD0 len 00017030 (92 KiB): DRAM[0m
[0;32mI (477) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (484) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (490) heap_init: At 4009216C len 0000DE94 (55 KiB): IRAM[0m
[0;32mI (496) cpu_start: Pro cpu start user code[0m
[0;32mI (516) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
�uart over ble start!
Create the BLE Device..
[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error
Create the BLE Server..
```
:-(

I realy dont know where to search now

manuprendlair, 2019-11-03:
I try to build the esp-idf ble spp client example and it worked on my hardware.
I add the arduino core as componenent and compile the esp-idf spp client. It worked
I change the code to the arduino ble uart example and its failed with:

`[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error
`
...

chegewara, 2019-11-03:
Ok, i just tested code from your repository, with esp-idf v3.2 and it works, with 1 exception. I had to disable classic BT in menuconfig.
Maybe, im not sure, there is some bug in library or just not enough memory to run both, ble and classic BT.

manuprendlair, 2019-11-04:
@chegewara : I understand you havent any message on serial console about ble init? this right?
How do you compile it? Directly from esp-idf command line tool chain ?
Could you give me information about (version)? 
and about esp-idf sdk is it the last release of v3.2 or the one with git ID I mentioned ?

chegewara, 2019-11-04:
With classic BT enabled in menuconfig BLE just didnt start, but when i disabled it then all works, my android phone can see it.
This is esp-idf version i used (most recent as of today):
```
commit 90af68bd3c1941d9f5513bee22a12de7250208d7
Merge: 093214c 0afad4c
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Mon Oct 14 21:46:22 2019 +0800

    Merge branch 'bugfix/alt_exit_idf_monitor_v3.2' into 'release/v3.2'
    
    idf_monitor: Exit with CTRL+X in menu (v3.2)
    
    See merge request espressif/esp-idf!6292
```
I am using VS code and since your code is not platformio ready, i had to use command line `make`.

manuprendlair, 2019-11-04:
ok, I will test this.

manuprendlair, 2019-11-04:
Thanks for your return chegewara <https://github.com/chegewara>

manuprendlair, 2019-11-04:
@chegewara  : Bluetooth classic option is disable and its not work..
Do you use my sdkconfig file without option BT classsic or do you rebuild one?
(could you send me the sdkconfig who work?)

I will try to compile the code with your sdk configuration.

Reagrds

manuprendlair, 2019-11-04:
So, I use the esp-idf installer under windows, I install the esp-idf 3.2 release branch and launch manualy a build
when I run IDF monitor, I got:
```

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6608
ho 0 tail 12 room 4
load:0x40078000,len:10752
load:0x40080400,len:6664
entry 0x40080778
I (72) boot: Chip Revision: 1
I (73) boot_comm: mismatch chip revision, expect 1, found 0
I (40) boot: ESP-IDF v3.2.2-176-g90af68bd3 2nd stage bootloader
I (40) boot: compile time 21:25:22
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: mismatch chip revision, expect 1, found 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2d4a8 (185512) map
I (176) esp_image: segment 1: paddr=0x0003d4d0 vaddr=0x3ffbdb60 size=0x02b04 ( 11012) load
I (180) esp_image: segment 2: paddr=0x0003ffdc vaddr=0x40080000 size=0x00034 (    52) load
0x40080000: _WindowOverflow4 at C:/Users/ManuF/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1779

I (183) esp_image: segment 3: paddr=0x00040018 vaddr=0x400d0018 size=0x7719c (487836) map
0x400d0018: _flash_cache_start at ??:?

I (363) esp_image: segment 4: paddr=0x000b71bc vaddr=0x40080034 size=0x003cc (   972) load
0x40080034: _WindowOverflow4 at C:/Users/ManuF/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1783

I (363) esp_image: segment 5: paddr=0x000b7590 vaddr=0x40080400 size=0x11dc4 ( 73156) load
I (410) boot: Loaded app from partition at offset 0x10000
I (410) boot: Disabling RNG early entropy source...
I (411) cpu_start: Pro cpu up.
I (414) cpu_start: Starting app cpu, entry point is 0x4008101c
0x4008101c: call_start_cpu1 at C:/Users/ManuF/Desktop/esp-idf/components/esp32/cpu_start.c:250

I (0) cpu_start: App cpu up.
I (425) heap_init: Initializing. RAM available for dynamic allocation:
I (432) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (438) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (444) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (450) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (456) heap_init: At 3FFC8FC8 len 00017038 (92 KiB): DRAM
I (462) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (469) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (475) heap_init: At 400921C4 len 0000DE3C (55 KiB): IRAM
I (481) cpu_start: Pro cpu start user code
I (500) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
uart over ble start!
Create the BLE Device..
[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error
Create the BLE Server..
```

So it's also not working. I am no able to analyze the monitor log. If somebody could help. Thanks

manuprendlair, 2019-11-06:
@chegewara : Could you share with me your working project ? (with arduino component)

chegewara, 2019-11-07:
I will, when i find few spare minutes.

manuprendlair, 2019-11-07:
Thanks for your support Chegewara!

chegewara, 2019-11-08:
Ok, this time i had some troubles before i make it to work. Looks like last time i did it automatically:
```
Bluetooth controller  --->Bluetooth controller mode (BR/EDR/BLE/DUALMODE) (BLE Only)  ---> BLE Only
```
And remember to disable Classic BT too.

manuprendlair, 2019-11-08:
@chegevara: Hum I did evry things, just not find why it failed evry time for me.
Did you able to run it again ?

chegewara, 2019-11-08:
Yes, this is what i did. Pull cod from your repository and tested few options in menuconfig until i was able to make it work, then i reverted BT classic options to make sure all is needed.
[sdkconfig.txt](https://github.com/espressif/arduino-esp32/files/3823702/sdkconfig.txt)

manuprendlair, 2019-11-08:
In my repository there are no ESP32 Arduino core. Did you use the last available in Espressif repository?

chegewara, 2019-11-08:
Yes, most recent arduino cloned.

manuprendlair, 2019-11-08:
so, I install the esp-idf 3.2 release with esp-idf-tools-setup-2.1.
I download the code from my repository.
I put your sdkconfig
run an esp-idf command prompt, then idf.py build
It compile just right, But when I ran the code.. always the same error..
`[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error `
I test the code on another hardware, same problem

I think I will give up ..:-(
'cause I dont know where to search

Last things, @chegewara could you try to run my bin on your hardware ?
[ble_uart.zip](https://github.com/espressif/arduino-esp32/files/3826343/ble_uart.zip)

chegewara, 2019-11-10:
https://github.com/chegewara/ESP_IDF_ARDUINO_BLE_TEST

manuprendlair, 2019-11-11:
@chegewara Did you have time to try my bin?
I took you ESP_IDF_ARDUINO_BLE_TEST fork and add the asp 32 arduino core that I found in your repository. then compile from esp-idf tools
- first the compile fail falling to this problem : https://github.com/espressif/arduino-esp32/issues/3209
-> so I kept the up to date BLEDevice.cpp
- secondly I got a missing main function, when I ran menuconfig I saw that the option for use setup and loop function. so I correct it. compile finish ok.
But when I ran it I got the same error:

`[E][BLEDevice.cpp:356] init(): esp_bt_controller_init: rc=259 Unknown ESP_ERR error `

...

chegewara, 2019-11-12:
In that case i really dont know what to say if code from my repository works on my device but you still getting error.
```
/**
 * @brief Initialize the %BLE environment.
 * @param deviceName The device name of the device.
 */
/* STATIC */ void BLEDevice::init(std::string deviceName) {
	if(!initialized){
		initialized = true; // Set the initialization flag to ensure we are only initialized once.

		esp_err_t errRc = ESP_OK;
#ifdef ARDUINO_ARCH_ESP32
		if (!btStart()) {
			errRc = ESP_FAIL;
			return;
		}
#else
		errRc = ::nvs_flash_init();
		if (errRc != ESP_OK) {
			log_e("nvs_flash_init: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}

#ifndef CLASSIC_BT_ENABLED
		esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);  
#endif
		esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
		errRc = esp_bt_controller_init(&bt_cfg);
		if (errRc != ESP_OK) {
			log_e("esp_bt_controller_init: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}

#ifndef CLASSIC_BT_ENABLED
		errRc = esp_bt_controller_enable(ESP_BT_MODE_BLE);
		if (errRc != ESP_OK) {
			log_e("esp_bt_controller_enable: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}
#else
		errRc = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
		if (errRc != ESP_OK) {
			log_e("esp_bt_controller_enable: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}
#endif
#endif

		esp_bluedroid_status_t bt_state = esp_bluedroid_get_status();
		if (bt_state == ESP_BLUEDROID_STATUS_UNINITIALIZED) {
			errRc = esp_bluedroid_init();
			if (errRc != ESP_OK) {
				log_e("esp_bluedroid_init: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
				return;
			}
		}

		if (bt_state != ESP_BLUEDROID_STATUS_ENABLED) {
			errRc = esp_bluedroid_enable();
			if (errRc != ESP_OK) {
				log_e("esp_bluedroid_enable: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
				return;
			}
		}

		errRc = esp_ble_gap_register_callback(BLEDevice::gapEventHandler);
		if (errRc != ESP_OK) {
			log_e("esp_ble_gap_register_callback: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}

#ifdef CONFIG_GATTC_ENABLE   // Check that BLE client is configured in make menuconfig
		errRc = esp_ble_gattc_register_callback(BLEDevice::gattClientEventHandler);
		if (errRc != ESP_OK) {
			log_e("esp_ble_gattc_register_callback: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}
#endif   // CONFIG_GATTC_ENABLE

#ifdef CONFIG_GATTS_ENABLE  // Check that BLE server is configured in make menuconfig
		errRc = esp_ble_gatts_register_callback(BLEDevice::gattServerEventHandler);
		if (errRc != ESP_OK) {
			log_e("esp_ble_gatts_register_callback: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}
#endif   // CONFIG_GATTS_ENABLE

		errRc = ::esp_ble_gap_set_device_name(deviceName.c_str());
		if (errRc != ESP_OK) {
			log_e("esp_ble_gap_set_device_name: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		};

#ifdef CONFIG_BLE_SMP_ENABLE   // Check that BLE SMP (security) is configured in make menuconfig
		esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE;
		errRc = ::esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
		if (errRc != ESP_OK) {
			log_e("esp_ble_gap_set_security_param: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		};
#endif // CONFIG_BLE_SMP_ENABLE
	}
	vTaskDelay(200 / portTICK_PERIOD_MS); // Delay for 200 msecs as a workaround to an apparent Arduino environment issue.
} // init
```

manuprendlair, 2019-11-13:
@chegewara This code is a full native esp-idf function to run ble right ? Do you check my repository with this code or with the arduino code in ble_uart.cpp ?

chegewara, 2019-11-13:
Yes, my repository i pasted link above is fork of your repository, just changed sdkconfig with `make menuconfig`.

Also I posted here BLEDevice::init() function (code above) just in case i have in  my environment somehow changed library (just dont remember) so you could compare with library you have.

manuprendlair, 2019-11-13:
ok, I will test all that. Thanks a lot!

chegewara, 2020-01-02:
I dont know, any news? Last post is this:

> ok, I will test all that. Thanks a lot!

I cant help more, because i cant reproduce your problem.

manuprendlair, 2020-01-02:
Ohhh sorry, I forgot a comment... where is my head ?? ;-)
So not any suggestion work! I am at the same point.
even with visualGDB or native Espressif tools, compilation was ok but fail with the same message at startup!
So for now I continue my project with wifi, but eat much power than bluetooth (I think...because not able to test as now)

closed 2020-03-16 by stale[bot]

GuyMinibrew, 2020-09-29:
I had the same issue, took me a long time to figure it out, it's a missing "#define ARDUINO_ARCH_ESP32", which has something to do with "-DARDUINO_ARCH_ESP32" not configured correctly.

miqmago, 2021-09-26:
Seems the problem is in the `esp32-hal-misc.c`. It releases all bluetooth memory because `btInUse()` returns false. It says this function is overriden in `esp32-hal-bt.c` but the fact is that it's calling the one defined in `esp32-hal-misc.c`:

```c
#ifdef CONFIG_BT_ENABLED
//overwritten in esp32-hal-bt.c
bool btInUse() __attribute__((weak));
bool btInUse(){ return false; }
#endif

void initArduino()
{
#ifdef CONFIG_APP_ROLLBACK_ENABLE
    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            if (verifyOta()) {
                esp_ota_mark_app_valid_cancel_rollback();
            } else {
                log_e("OTA verification failed! Start rollback to the previous version ...");
                esp_ota_mark_app_invalid_rollback_and_reboot();
            }
        }
    }
#endif
    //init proper ref tick value for PLL (uncomment if REF_TICK is different than 1MHz)
    //ESP_REG(APB_CTRL_PLL_TICK_CONF_REG) = APB_CLK_FREQ / REF_CLK_FREQ - 1;
#ifdef F_CPU
    setCpuFrequencyMhz(F_CPU/1000000);
#endif
#if CONFIG_SPIRAM_SUPPORT
    psramInit();
#endif
    esp_log_level_set("*", CONFIG_LOG_DEFAULT_LEVEL);
    esp_err_t err = nvs_flash_init();
    if(err == ESP_ERR_NVS_NO_FREE_PAGES){
        const esp_partition_t* partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, NULL);
        if (partition != NULL) {
            err = esp_partition_erase_range(partition, 0, partition->size);
            if(!err){
                err = nvs_flash_init();
            } else {
                log_e("Failed to format the broken NVS partition!");
            }
        }
    }
    if(err) {
        log_e("Failed to initialize NVS! Error: %u", err);
    }
#ifdef CONFIG_BT_ENABLED
    if(!btInUse()){
        esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
    }
#endif
    init();
    initVariant();
}
```

I've tried to define my own `btInUse()` function, but it seems not possible to override it, even from `src/main.cpp`.

The only solution I've found up to now is to call a function from `esp32-hal-bt.c`. Right now I've selected `btStarted()` which seems very innocent and then it seems the `esp32-hal-bt.c` is linked and the `btInUse()` is overriden, so it makes to not release all bluetooth and it makes the magic to happen.

chegewara, 2021-09-26:
> Seems the problem is in the `esp32-hal-misc.c`. It releases all bluetooth memory because `btInUse()` returns false. It says this function is overriden in `esp32-hal-bt.c` but the fact is that it's calling the one defined in `esp32-hal-misc.c`

if that would be true, then bluetooth would not work to anyone ever.

Maybe it is required to init BT in setup, so make use that BT from `esp32-bt-hal.c`, just like you said, but for sure there is some magic.

miqmago, 2021-09-27:
For sure this is quite strange. I'm using platformIO:
```
[platformio]
default_envs = esp32dev

[env]
platform = espressif32
framework = 
	arduino
	espidf
monitor_speed = 115200
platform_packages = framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
board_build.partitions = partitions/partitions_singleapp_large.csv
build_flags = 
    -DCORE_DEBUG_LEVEL=4
    -DESP32=1
lib_deps = 
	me-no-dev/ESP Async WebServer@^1.2.3
	me-no-dev/ESPAsyncTCP@^1.2.2

[env:esp32dev]
board = esp32dev
```

I've digged inside code and haven't seen a rational explanation. Even I've tried to change the order in CMakeLists.txt of the repo. What I'm sure is that if I don't call a function from `esp32-bt-hal.c`, the `btInUse()` definition comes from `esp32-hal-misc.c`, I've placed logs everywhere to see which function is being called and traced to see if there was another call to `esp_bt_controller_mem_release(ESP_BT_MODE_BTDM)` but the only one comes from `initArduino()`.

Perhaps it might be due to compiler options, or even because of `__attribute__((weak))` processement.

artoowang, 2022-12-10:
> Right now I've selected `btStarted()` which seems very innocent and then it seems the `esp32-hal-bt.c` is linked and the `btInUse()` is overriden, so it makes to not release all bluetooth and it makes the magic to happen.

Thank you, sir! I was completely puzzled by the init error, but what you suggested resolved the issue like magic.

parksj10, 2023-06-11:
I can confirm this is still an issue in 4.4.5, thanks @miqmago for the solution!