espressif/arduino-esp32 #3437: Connect between 2 ESP32

gamer421, 2019-10-30:
Is it possible to pair 2 ESP32 and be able to send data between them?

Thanks

lbernstone, 2019-10-30:
There are probably 100 ways to do this.  Maybe explain which you mean.

gamer421, 2019-10-30:
> There are probably 100 ways to do this. Maybe explain which protocol you mean.

I am sorry, I am new to this. 
So basically I have two ESP32 and I want to connect them using bluetooth serial, I want to be able to send data from one ESP32 to the other one to activate physical sensor readings.
I haven't found anything explaining on how to connect two ESP32 using bluetooth. The only thing I find is to teach you how to connect an ESP32 to a smartphone through a bluetooth terminal app.
I really just need help to connect them. I can't find a way because there's no interface like the bluetooth terminal app where you just click the device you want to connect to.

Thanks

gamer421, 2019-11-01:
Anyone? Please?

beegee-tokyo, 2019-11-01:
Serial
Bluetooth
WiFi
I2C
SPI

You choose, hit Google Search, and will get solutions

gamer421, 2019-11-01:
> Serial
> Bluetooth
> WiFi
> I2C
> SPI
> 
> You choose, hit Google Search, and will get solutions

Like I said in the previous post, I couldn't find any information on how to communicate between 2 ESP32 using bluetooth classic.

beegee-tokyo, 2019-11-01:
Issues are for bugs and mistakes in the Arduino - ESP32 framework, not a training/help forum. Check the examples here https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial/examples

gamer421, 2019-11-01:
> Issues are for bugs and mistakes in the Arduino - ESP32 framework, not a training/help forum. Check the examples here https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial/examples

I am new at this, I'm sorry... Anyway I don't see a reason for you to act like that, but thank you for all the help.

gamer421, 2019-11-01:
Would appreciate if anyone else could help me do this.

atanisoft, 2019-11-01:
@gamer421 I'd suggest posting this in the forum on esp32.com instead of here since as @beegee-tokyo mentioned this tracker is for bugs in the arduino-esp32 code only.

gamer421, 2019-11-01:
> @gamer421 I'd suggest posting this in the forum on esp32.com instead of here since as @beegee-tokyo mentioned this tracker is for bugs in the arduino-esp32 code only.

Ok thank you.

closed 2019-11-01 by gamer421

me-no-dev, 2019-11-03:
you have the examples in the link above. they will connect two ESPs together over BT serial. All you have to do is properly configure both sketches. If you are new, this will be a good example for  you to get on with reading example code. You will need this in the future.