espressif/arduino-esp32 #3438: esp32 arduinoOTA  sethostname() not working

Kiuman, 2019-10-31:
Hi,
I'm using `ArduinoOTA.sethostname("esp32-entrance-device");` and the hostname in the Arduino IDE Network ports is always  showing as esp32-[MAC address].
Maybe I'm a little bit rusty with my code abilities, but isn't _hostname missing in the ArduinoOTAClass::ArduinoOTAClass() definition?

Link to code: https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/ArduinoOTA.cpp

Thanks!

EsserPrototyping, 2019-10-31:
The following worked, but i don´t think you really need the disconnect and config parts..

In my case it wasnt the ESP32, it was my router. Check that you don´t have the mac or ip assigned anywhere in the router and change the lease time of dhcp to something like one minute.. now the router should see the ESP with it´s new hostname.. 

`
char hname[] = "ESP32_1234";

void setup () {

  WiFi.disconnect();  // probalbly not needed
  WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE); // probalbly not needed
  WiFi.setHostname(hname);
  WiFi.mode(WIFI_STA);

  WiFi.begin(ssid, password);

}
`

Kiuman, 2019-11-06:
Well thanks to both, indeed `ArduinoOTA.setHostaname(hostname);` (from ArduinoOTA.h) doesn't work as it is described, but `WiFi.setHostname(hostname);` (from WiFi.h) does work as intended. 
Issue closed.

closed 2019-11-06 by Kiuman

mepster, 2020-11-03:
@EsserPrototyping thanks for the tip!

Turns out for me it was

`WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);`

that was needed... although I actually used

`WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);`

I found in addition that `WiFi.mode(WIFI_STA);` and `WiFi.disconnect();` were not necessary.

(All of the above must go before `WiFi.begin(ssid, password);`)