espressif/arduino-esp32 #3439: avr/dtostrf() cannot be #included

dsyleixa, 2019-10-31:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Adafruit Feather ESP32
Core Installation version:			 
IDE name:						Arduino 1.8.9
Flash Frequency:					80MHz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					     921600
Computer OS:                                        Windows 7/64 pro

### Description:
Describe your problem here
avr/dtostrf() cannot be included by
`#include <avr/dtostrf.h>`

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(1000);
  Serial.println();
  
  float f=1234.567;
  char s[20]="";
  dtostrf(f,8,3,s);
  Serial.println(f);
  Serial.println(s);
  Serial.println();
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

```

### Debug Messages:
```

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_732431 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -verbose D:\Arduino\portable\sketchbook\dtostrftest000\dtostrftest000.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_732431 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -verbose D:\Arduino\portable\sketchbook\dtostrftest000\dtostrftest000.ino
Using board 'featheresp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/heap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/idf_test" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/jsmn" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/json" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/libsodium" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/log" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/lwip" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mbedtls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mdns" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/micro-ecc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mqtt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/newlib" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nghttp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nvs_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/openssl" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/protobuf-c" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/protocomm" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/pthread" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/sdmmc" "-ID:\\arduino\\portable\\packages\\esp32\\hardwa [... cut ...] P32\"" "-DARDUINO_VARIANT=\"feather_esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2\\cores\\esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2\\variants\\feather_esp32" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_732431\\sketch\\dtostrftest000.ino.cpp" -o nul
dtostrftest000:1:25: error: avr/dtostrf.h: No such file or directory

compilation terminated.

exit status 1
avr/dtostrf.h: No such file or directory

```

lbernstone, 2019-10-31:
A quick search in the repository shows dtostrf is in stdlib_noniso.h

dsyleixa, 2019-10-31:
ok, that is kind of unexpected path but it surprisingly works...
#include <stdlib_noniso.h>

why doesn't 
#include <avr/dtostrf.h>
work? it's finally there and it works for all SAM and SAMD cores, as stated?

atanisoft, 2019-10-31:
No include is necessary to have the dtostrf function declared/usable on the esp32. It seems that the "avr/dtostrf.h" file is specific to SAM/SAMD and does not exist outside of that architecture.

dsyleixa, 2019-11-01:
yes, surprisingly you are right!
there always was a need to
#include <avr/dtostrf.h>
for all non-AVR-boards (e.g. ARM Cortex M0, M3, M4) so I took that line also for ESP.

Alright, thank you, issue solved.

dsyleixa, 2019-11-02:
hello, will it be possible to increase include file compatibility with other architectures, like commited for esp8288?
https://github.com/esp8266/Arduino/pull/6700
https://github.com/esp8266/Arduino/issues/6695#issuecomment-549024415

me-no-dev, 2019-11-03:
no. this makes no sense. you can easily check if you are on the architecture that requires it and include it. You can see in the official Arduino API that they are deprecated: https://github.com/arduino/ArduinoCore-API/tree/master/api/deprecated-avr-comp/avr

dsyleixa, 2019-11-03:
ok, but ftm avr/dtostrf.h is still there in the avr folder and dtostrf(). is also part of MANY libs...

closed 2019-11-03 by dsyleixa