espressif/arduino-esp32 #3440: WiFi Starting Problem

afnan, 2019-11-01:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                       ?Windows 10

### Description:
This is an intermittent problem. I was unable to find a pattern for this to occur. Given that the device works most of the time I believe the code is not at fault. I never had this issue before but the latest version of ESP32 for Arduino seems to cause this because was working previously and i did not modify the code. 

### Sketch: 
```cpp
boolean setupWiFi() {
  //  This is here to force the ESP32 to reset the WiFi and initialise correctly.
  uint8_t connectAttempts = 0;

restartConnection: Serial.print("WIFI status = ");

  Serial.println(WiFi.getMode());
  WiFi.disconnect(true);
  delay(1000);
  WiFi.mode(WIFI_STA);
  delay(1000);
  WiFi.enableAP(false);
  WiFi.enableSTA(true);
  Serial.print("WIFI status = ");
  Serial.println(WiFi.getMode());
  Serial.print("Attempt to connect # =>");
  Serial.println(connectAttempts);
  // End silly stuff !!!
  wifiMulti.addAP("BATMEN", "_R3se@ch2019!");    //Add Multiple WiFi connections.
  wifiMulti.addAP("No Device(s) Found", "hydrogenbomb699");    //Add Multiple WiFi connections.
  if ((wifiMulti.run() == WL_CONNECTED)) {
    Serial.println("WiFi Connected");
    return true;
  }//If connection was a success
  else {
    Serial.println("Could not connect to WiFI");
    if (connectAttempts <= 5)//Make 5 attempts before sleeping
    {
      timerWrite(timer, 0); //reset timer (feed watchdog)
      connectAttempts++;;
      goto restartConnection;
    }
    LetsSleep();
    return false;
  }
  //   WiFi.begin(ssid, password);
  //
  //    while (WiFi.status() != WL_CONNECTED) {
  //        delay(500);
  //        Serial.print(".");
  //    }
}
```

### Debug Messages:
```
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: BATMEN
E (25175) wifi: esp_wifi_scan_start 1245 wifi not start
[D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
E (25185) wifi: esp_wifi_disconnect 1153 wifi not start
[E][WiFiSTA.cpp:257] disconnect(): disconnect failed!
[D][WiFiMulti.cpp:198] run(): [WIFI] start scan
E (25199) wifi: esp_wifi_scan_start 1245 wifi not start
Could not connect to WiFI
E (25707) wifi: esp_wifi_disconnect 1153 wifi not start
[E][WiFiSTA.cpp:257] disconnect(): disconnect failed!

 
```

afnan, 2019-11-03:
@me-no-dev any clues on this one?

smacyas, 2019-11-07:
Let me say my concerns about the work of the WiFi stack. I noticed that the esp32 does not always correctly process the request to update dynamic keys and does not reconnect to the network. On pictures it is visible that Mikrotik updated keys, and esp32 is not present and inaccessible in a network, only restart helps.
[image]
[image]
[image]

afnan, 2019-11-07:
> Let me say my concerns about the work of the WiFi stack. I noticed that the esp32 does not always correctly process the request to update dynamic keys and does not reconnect to the network. On [...]

Not sure how it is linked to the issue but good to see that graph :)

closed 2020-01-20 by stale[bot]

fellek, 2020-04-19:
Confirm this behavior. Have you found any solution for that issue?