espressif/arduino-esp32 #3443: Download for Board-manager is impossibly slow

Ridgeback09, 2019-11-01:
Can't download because it takes hours...

atanisoft, 2019-11-01:
@Ridgeback09 a bit more info would be needed than simply it is slow..

@me-no-dev FYI, another case of possible GH CDN issues.

Ridgeback09, 2019-11-02:
@atanisoft 
[Download_Clip.m4v.zip](https://github.com/espressif/arduino-esp32/files/3800401/Download_Clip.m4v.zip)

I've added the Board-manager URL to Arduino ("https://dl.espressif.com/dl/package_esp32_index.json") and the download won't finish because it takes forever to load just a few kB.

Attached a short video where you can see how slow it is.

Ridgeback09, 2019-11-02:
After waiting forever I get this error which says: "CRC isn't matching. Can be a temporary problem, please try again later."

[image]

dnguyen76, 2019-11-03:
New procedure for Arduino 1.8.0

You must use now this link 

Stable release link:  **https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json**

(see https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)

Ridgeback09, 2019-11-03:
Thank you,
Worked like a charm!

closed 2019-11-03 by Ridgeback09

Joel777, 2020-03-27:
Incredibly slow now using the above "Stable release link".  Is there a newer link or procedure?

dnguyen76, 2020-03-27:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
is upto now the link to use ( see https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)

lbernstone, 2020-03-27:
The internet may be slow for a couple weeks.
https://www.worldometers.info/coronavirus/#countries

Joel777, 2020-03-27:
Thanks.  Death doesn't normally slow down internet performance, though.
Here's a link for monitoring internet performance worldwide: http://www.internettrafficreport.com/
I've been doing a lot of online meetings and video uploads & downloads with very little noticeable performance degradation, so I'm surprised the Arduino Boards Manager is so incredibly slow.  Thank you for the new link, dnguyen76, but it's still slow.  I'll just wait it out.

lbernstone, 2020-03-27:
Death rates are probably unchanged or a slight tick down, as people are driving and murdering less.
Trouble ticket times, on the other hand, are way up.

AugustoOspal, 2020-09-23:
I've just found the solution. I was trying to download the file from Argentina, but then I tried by downloading it using a VPN, and it worked perfectly.