espressif/arduino-esp32 #3444: strange unfunctionality/incompatibility of ESP32 code vs. ESP8266 code

dsyleixa, 2019-11-02:
the following code can be compiled for ESP32 and runs,
nonetheless it hangs up and creates no accessable website.
Tis code is working for ESP8266 though (code attached, see below!)

### Hardware:
Board:							Adafruit Feather ESP32
Core Installation version:			    1.0.2 + 1.0.4
IDE name:						Arduino 1.8.9
Flash Frequency:					80MHz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					     921600
Computer OS:                                        Windows 7/64 pro

### Description:
Describe your problem here
avr/dtostrf() cannot be included by
`#include <avr/dtostrf.h>`
this code works with ARM Cortex SAMD21, SAMD51 and M3/Due though.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>

// Version 0.0.2a

const char* ssid = "WLAN-3YA7LD";
const char* password = "18658446694458594657";

WebServer server(80);

//-----------------------------------------------------
  //HTML und Daten für das Eingabe-/Ausgabefeld
  //einfügen im globalen Scope
  //-----------------------------------------------------
  //erstellt von (c): Moppi    @roboternetz.de
  //-----------------------------------------------------
  //Breite und Höhe des ges. Feldes
    const int tableWidth = 10;
    const int tableHeight = 10;
  //Füllzeichen für Felder
    const String fillChar = "X";
  //Parameter-Array enthält den Status aller Felder
  //der Ein-/Ausgabematrix (0= AUS, 1= AN)
    bool matrix[tableHeight*tableWidth]={0};
  //Status, ob Änderungen vorliegen
    volatile static bool toChange = 0;
    volatile static bool toLearn = 1;
    volatile static bool toDetect = 0;
    volatile static bool toReset = 0;
    volatile static bool toSave = 0;
    volatile static bool toClear = 0;
  //HTML
    String body = "";
    String s1="<style>\nform{margin:0}\n#un{width:0;height:0;visibility:hidden}\n#bu{width:10em;height:2em;}\n#rw{display:table-row;}\n#ce{display:table-cell;}\n#ip{width:1.5em;text-align:center;font-weight:bold}\n</style>\n<script>dge=document.getElementById; function oc(ob){if(ob.value=='')ob.value='"+fillChar+"';else ob.value='';}</script>\n<div role=\"presentation\" style=\"display:table\"><div id='rw'><div id='ce' style=\"padding-right:1em\"><form method=\"get\"><div role=\"presentation\" style=\"margin-bottom:1em;display:table\">";
    String s2="<div id=\"rw\">";
    String s3="<div id=\"ce\"><input onclick=\"oc(this);\" type=\"text\" name=\"b";
    String s3_1="\" value=\"";
    String s3_2="\" maxlength=\"1\" id=\"ip\"></div>";
    String s4="</div>";
    String s5="</div><div id='ce'><input id=\"bu\" type=\"submit\" value=\"Change:Ok\"><p></form>";
    String s5_0="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Clear\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"clear\"></form>"; 
    String s5_1="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Learn\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"learn\"></form>";
    String s5_2="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Detect\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"detect\"></form>";
    String s5_3="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Reset\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"reset\"></form>";
    String s5_4="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Save\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"save\"></form></div></div></div>";
String par="b";
  //-----------------------------------------------------

float bufferArray[100];

//-----------------------------------------------------
//-----------------------------------------------------
void setup() {

  Serial.begin(115200);
  delay(1000);
  Serial.println();

  //-----------------------------------------------------
  //Anmeldung am Netzwerk
  //-----------------------------------------------------
  Serial.println("connecting by IPAddress(192,168,2,99)");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password); //begin WiFi connection
  WiFi.config(IPAddress(192, 168, 2, 99), IPAddress(192, 168, 2, 1), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));

  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  //-----------------------------------------------------
  //Auswertung und Erstellung des Ein-/Ausgabefeldes
  //-----------------------------------------------------
  //Einbinden in setup() und in loop() nutzen durch
  //Aufruf von "server.handleClient();"
  //benötigt Objekt "server", erstellt im globalen Scope
  //mittels:       ESP8266WebServer server(80);
  //benötigt auch: #include <ESP8266WebServer.h>
  //-----------------------------------------------------
  //erstellt von (c): Moppi    @roboternetz.de
  //-----------------------------------------------------

  memset(matrix , 0, sizeof(matrix));
  server.on("/",[]()
   {
    //Auswerten des Query-String vom Browser
    int n = 0; 
    if(server.hasArg("bn")){
      if(server.arg("bn") == "clear") toClear=1; else toClear=0;
      if(server.arg("bn") == "save") toSave=1; else toSave=0;
      if(server.arg("bn") == "reset") toReset=1; else toReset=0;
      if(server.arg("bn") == "detect") toDetect=1; else toDetect=0;
      if(server.arg("bn") == "learn") toLearn=1; else toLearn=0;      
    }
    for(int x=0; x < tableHeight; x++){
      for(int j=0; j < tableWidth; j++){
        if(server.hasArg(par+String(n))){
          String a = server.arg(par+String(n));
          bool ic = matrix[n];
          if (a == "") matrix[n] = 0; else matrix[n] = 1; 
          if (ic != matrix[n]) toChange = 1;
          n++;
        }
      }
    }
    //Aufbau des Feldes
    String v; body = s1; n = 0;
    for(int x=0; x < tableHeight; x++){
      body += s2;
      for(int j=0; j < tableWidth; j++){ 
        if (matrix[n]) v = fillChar; else v = "";
        body += s3 + String(n) + s3_1 + v + s3_2;
        n++;
      }
      body += s4;
    }
    body += s4+s5+s5_0+s5_1+s5_2+s5_3+s5_4;
    //Senden der Seite an den Browser
    server.send(200, "text/html", body);
   }
  );
  //-----------------------------------------------------

  //-----------------------------------------------------
  //Webserver starten
  //-----------------------------------------------------
  server.begin();
  Serial.println("Web server started!");

}

uint32_t timestamp=0, sec=0;
uint8_t modeLearn=0, modeDetect=0;

//-----------------------------------------------------
//-----------------------------------------------------
void loop() {  
  server.handleClient();
  // Aktionen nach ButtonClick:
  if(toClear) {
    Serial.println((String)"toClear="+toClear);
    memset(matrix , 0, sizeof(matrix));
    toClear=0;
  }
  if (toChange) {
    Serial.println("new data:");
    for (int i = 0; i < 100; i++) {
       bufferArray[i] = (float)matrix[i];
       Serial.println((String)i + ": " + bufferArray[i]);
    }
    toChange=0;
  }
  if(toLearn){       
    modeLearn=1;
    modeDetect=0;
    Serial.println((String)"modeLearn="+modeLearn);
    toLearn=0; 
  }
  else 
  if(toDetect) {      
    modeLearn=0;
    modeDetect=1;
    Serial.println((String)"modeDetect="+modeDetect);
    toDetect=0;  
  }
  if(toReset) {
    toLearn=0;
    toDetect=1;
    Serial.println((String)"toReset="+toReset);
    toReset=0;
  }
  if(toSave) {
    toLearn=0;
    toDetect=1;
    Serial.println((String)"toSave="+toSave);
    toSave=0;
  }
  if(millis()-timestamp >= 1000) {
    sec++;
    timestamp +=1000;
    Serial.println(sec);
  }
}

```

### Debug Messages:
``` 
//ESP32:

load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

connecting by IPAddress(192,168,2,99)

..
Connected to WLAN-3YA7LD
IP address: 192.168.2.99
Web server started!
modeLearn=1
1
2
3
4
5
^^^^^^^^^^^^^^^^
for core 1.0.2: freezes, no website in the browser at url 192.168.2.99\
for core 1.0.4: no freezing, counter increments, but also no website in the browser at url 192.168.2.99\

```

dsyleixa, 2019-11-02:
this is the working ESP8266 code version:
```

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

// Version 0.0.2a

const char* ssid = "WLAN-3YA7LD";
const char* password = "18658446694458594657";

ESP8266WebServer server(80);

//-----------------------------------------------------
  //HTML und Daten für das Eingabe-/Ausgabefeld
  //einfügen im globalen Scope
  //-----------------------------------------------------
  //erstellt von (c): Moppi    @roboternetz.de
  //-----------------------------------------------------
  //Breite und Höhe des ges. Feldes
    const int tableWidth = 10;
    const int tableHeight = 10;
  //Füllzeichen für Felder
    const String fillChar = "X";
  //Parameter-Array enthält den Status aller Felder
  //der Ein-/Ausgabematrix (0= AUS, 1= AN)
    bool matrix[tableHeight*tableWidth]={0};
  //Status, ob Änderungen vorliegen
    volatile static bool toChange = 0;
    volatile static bool toLearn = 1;
    volatile static bool toDetect = 0;
    volatile static bool toReset = 0;
    volatile static bool toSave = 0;
    volatile static bool toClear = 0;
  //HTML
    String body = "";
    String s1="<style>\nform{margin:0}\n#un{width:0;height:0;visibility:hidden}\n#bu{width:10em;height:2em;}\n#rw{display:table-row;}\n#ce{display:table-cell;}\n#ip{width:1.5em;text-align:center;font-weight:bold}\n</style>\n<script>dge=document.getElementById; function oc(ob){if(ob.value=='')ob.value='"+fillChar+"';else ob.value='';}</script>\n<div role=\"presentation\" style=\"display:table\"><div id='rw'><div id='ce' style=\"padding-right:1em\"><form method=\"get\"><div role=\"presentation\" style=\"margin-bottom:1em;display:table\">";
    String s2="<div id=\"rw\">";
    String s3="<div id=\"ce\"><input onclick=\"oc(this);\" type=\"text\" name=\"b";
    String s3_1="\" value=\"";
    String s3_2="\" maxlength=\"1\" id=\"ip\"></div>";
    String s4="</div>";
    String s5="</div><div id='ce'><input id=\"bu\" type=\"submit\" value=\"Change:Ok\"><p></form>";
    String s5_0="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Clear\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"clear\"></form>"; 
    String s5_1="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Learn\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"learn\"></form>";
    String s5_2="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Detect\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"detect\"></form>";
    String s5_3="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Reset\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"reset\"></form>";
    String s5_4="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Save\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"save\"></form></div></div></div>";
String par="b";
  //-----------------------------------------------------

float bufferArray[100];

//-----------------------------------------------------
//-----------------------------------------------------
void setup() {

  Serial.begin(115200);
  delay(1000);
  Serial.println();

  //-----------------------------------------------------
  //Anmeldung am Netzwerk
  //-----------------------------------------------------
  Serial.println("connecting by IPAddress(192,168,2,99)");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password); //begin WiFi connection
  WiFi.config(IPAddress(192, 168, 2, 99), IPAddress(192, 168, 2, 1), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));

[... 145 lines cut ...]
8
9
10
11
12
13
14
toClear=1
15
16
17
18
19
20
21
modeDetect=1
22
23
24

....

a.s.o., all systems are functioning!

```

dsyleixa, 2019-11-02:
this is the working ESP8266 code version:
website available, all button presses working fine!
(same code, just ESP8266 libs changed to ESP32 libs)

```

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

// Version 0.0.2a

const char* ssid = "WLAN";
const char* password = "1234657";

ESP8266WebServer server(80);

//-----------------------------------------------------
  //HTML und Daten für das Eingabe-/Ausgabefeld
  //einfügen im globalen Scope
  //-----------------------------------------------------
  //erstellt von (c): Moppi    @roboternetz.de
  //-----------------------------------------------------
  //Breite und Höhe des ges. Feldes
    const int tableWidth = 10;
    const int tableHeight = 10;
  //Füllzeichen für Felder
    const String fillChar = "X";
  //Parameter-Array enthält den Status aller Felder
  //der Ein-/Ausgabematrix (0= AUS, 1= AN)
    bool matrix[tableHeight*tableWidth]={0};
  //Status, ob Änderungen vorliegen
    volatile static bool toChange = 0;
    volatile static bool toLearn = 1;
    volatile static bool toDetect = 0;
    volatile static bool toReset = 0;
    volatile static bool toSave = 0;
    volatile static bool toClear = 0;
  //HTML
    String body = "";
    String s1="<style>\nform{margin:0}\n#un{width:0;height:0;visibility:hidden}\n#bu{width:10em;height:2em;}\n#rw{display:table-row;}\n#ce{display:table-cell;}\n#ip{width:1.5em;text-align:center;font-weight:bold}\n</style>\n<script>dge=document.getElementById; function oc(ob){if(ob.value=='')ob.value='"+fillChar+"';else ob.value='';}</script>\n<div role=\"presentation\" style=\"display:table\"><div id='rw'><div id='ce' style=\"padding-right:1em\"><form method=\"get\"><div role=\"presentation\" style=\"margin-bottom:1em;display:table\">";
    String s2="<div id=\"rw\">";
    String s3="<div id=\"ce\"><input onclick=\"oc(this);\" type=\"text\" name=\"b";
    String s3_1="\" value=\"";
    String s3_2="\" maxlength=\"1\" id=\"ip\"></div>";
    String s4="</div>";
    String s5="</div><div id='ce'><input id=\"bu\" type=\"submit\" value=\"Change:Ok\"><p></form>";
    String s5_0="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Clear\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"clear\"></form>"; 
    String s5_1="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Learn\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"learn\"></form>";
    String s5_2="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Detect\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"detect\"></form>";
    String s5_3="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Reset\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"reset\"></form>";
    String s5_4="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Save\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"save\"></form></div></div></div>";
String par="b";
  //-----------------------------------------------------

float bufferArray[100];

//-----------------------------------------------------
//-----------------------------------------------------
void setup() {

  Serial.begin(115200);
  delay(1000);
  Serial.println();

  //-----------------------------------------------------
  //Anmeldung am Netzwerk
  //-----------------------------------------------------
  Serial.println("connecting by IPAddress(192,168,2,99)");
  WiFi.mode(WIFI_STA);
[... 148 lines cut ...]
8
9
10
11
12
13
14
toClear=1
15
16
17
18
19
20
21
modeDetect=1
22
23
24

....

a.s.o., all systems are functioning!

```

atanisoft, 2019-11-02:
Can you try moving WiFi.config() to before WiFi.begin()? I suspect what is happening is it connects and uses DHCP (default until config is called with static IP) and then the call to config resets it but the address is not used as it has already connected to WiFi.

dsyleixa, 2019-11-02:
core 1.0.2:

```cpp
//-----------------------------------------------------
  //Anmeldung am Netzwerk
  //-----------------------------------------------------
  Serial.println("connecting by IPAddress(192,168,2,99)");
  WiFi.mode(WIFI_STA);
   
  WiFi.config(IPAddress(192, 168, 2, 99), IPAddress(192, 168, 2, 1), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
  WiFi.begin(ssid, password); //begin WiFi connection
```
   
now the counter is incrementing (stops intermediately each 20s shortly, then going on,
but still no website available
```

load:0x40080400,len:5868
entry 0x4008069c

connecting by IPAddress(192,168,2,99)

..
Connected to WLAN-3YA7LD
IP address: 192.168.2.99
Web server started!
modeLearn=1
1
2
3
4
5
6
7
8
9
10
11
...
```

dsyleixa, 2019-11-02:
core 1.0.2:

```cpp
//-----------------------------------------------------
  //Anmeldung am Netzwerk
  //-----------------------------------------------------
  Serial.println("connecting by IPAddress(192,168,2,99)");
  WiFi.mode(WIFI_STA);
   
  WiFi.config(IPAddress(192, 168, 2, 99), IPAddress(192, 168, 2, 1), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
  WiFi.begin(ssid, password); //begin WiFi connection
```
   
now the counter is incrementing (stops intermediately each 20s shortly, then going on,
but still no website available
```

load:0x40080400,len:5868
entry 0x4008069c

connecting by IPAddress(192,168,2,99)

..
Connected to WLAN 
IP address: 192.168.2.99
Web server started!
modeLearn=1
1
2
3
4
5
6
7
8
9
10
11
...
```

atanisoft, 2019-11-02:
Can you turn on debug logging via the boards menu with the esp32 dev module selected as the board type?

dsyleixa, 2019-11-02:
here we go...:
Serial console:

```

load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

connecting by IPAddress(192,168,2,99)
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START

.[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.2.99, MASK: 192.168.2.1, GW: 192.168.2.1
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.2.99, MASK: 192.168.2.1, GW: 192.168.2.1
.
Connected to WLAN 
IP address: 192.168.2.99
Web server started!
modeLearn=1
1
2
3
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
4
5
6
7
8
9
// edit: here opening 192.168.2.99/
// edit: shortly stops a couple of sec, then proceeding...
10
11
12
13
14
15
16
17
18
19
```

atanisoft, 2019-11-02:
Is this with 1.0.4? It looks like the server did accept a connection but there was no data to read and it got disconnected.

dsyleixa, 2019-11-02:
no, that is 1.0.2 again, 1.0.4 is pending...

dsyleixa, 2019-11-02:
core 1.0.4:
```

load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

connecting by IPAddress(192,168,2,99)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.2.99, MASK: 192.168.2.1, GW: 192.168.2.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.2.99, MASK: 192.168.2.1, GW: 192.168.2.1
.
Connected to WLAN-3YA7LD
IP address: 192.168.2.99
Web server started!
modeLearn=1
1
2
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
3
4
5
6
7
8
9
10
// edit: again here opening 192.168.2.99/
// edit: shortly stops a couple of sec, then proceeding...
11
12
13
14
15
16
17
18
19
20
21
22
23

```

atanisoft, 2019-11-02:
From the output it looks like the server is behaving correctly but there is a problem in receiving data from the browser. It is not clear what the issue might be though unfortunately.

atanisoft, 2019-11-02:
This is very similar to #1921 for that single output line between 2 and 3.

dsyleixa, 2019-11-02:
esp8266webserver performs correctly though...

atanisoft, 2019-11-02:
There is a big difference between esp32 and esp8266 when it comes to the tcp/ip stack and the WiFi library that both WebServer codes use. I don't see what the issue might be in the code unfortunately.

As with #1921 and other issues related to WebServer it would be highly recommended to shift to EspAsyncWebServer which works on both platforms and frees up the loop() method from running any part of the server.

dsyleixa, 2019-11-02:
can the code above remain the same? (just exchanging EspWebServer  by EspAsyncWebServer ?)

atanisoft, 2019-11-02:
Virtually the same. You would just need to drop the server.handleClient() line from the loop() method.

dsyleixa, 2019-11-02:
to put it then to where...?

atanisoft, 2019-11-02:
You will need to install the library, you can find it here: https://github.com/me-no-dev/ESPAsyncWebServer

You will also need EspAsyncTCP for the esp8266 and AsyncTCP for esp32, both should be linked from the above.

You won't need the server.handleClient() line at all.

dsyleixa, 2019-11-02:
after install,
#include <ESPAsyncWebServer.h>
muddles all up about WiFi.h

In file included from C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_513526\esp32_html_tablectrl002b.ino:3:0:

D:\arduino\portable\sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory

compilation terminated.

Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Nicht benutzt: D:\arduino\libraries\WiFi
Bibliothek WiFi in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi  wird verwendet
Bibliothek ESPAsyncWebServer in Version 1.2.3 im Ordner: D:\arduino\portable\sketchbook\libraries\ESPAsyncWebServer  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\FS  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.

Is there no way to make my simple website work using the standard WebServer.h?
Perhaps changing my code or fixing WebServer.h ibugs?

dsyleixa, 2019-11-02:
after installing both libs, then
#include <ESPAsyncWebServer.h>
muddles all up about WiFi.h

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <ESPAsyncWebServer.h>

// Version 0.0.2b: async

ESPAsyncWebServer server(80);
```

```

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=debug,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_780026 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_547671\esp32_html_tablectrl002b.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=debug,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_780026 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_547671\esp32_html_tablectrl002b.ino
Using board 'featheresp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\ [... cut ...]
Bibliothek ESPAsyncWebServer in Version 1.2.3 im Ordner: D:\arduino\portable\sketchbook\libraries\ESPAsyncWebServer  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\FS  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.
```

Is there no way to make my simple website work using the standard WebServer.h?
Perhaps changing my code or fixing WebServer.h bugs?

Finally esp8266webserver performs correctly with that code though!

atanisoft, 2019-11-02:
Did you install AsyncTCP as well? It is a dependency for esp32, the EspAsyncTCP is only for esp8266. There are differences in the two architectures thus two different libs.

As for fixing WebServer, that would take a bit more work than simply using a more performant and stable server.

dsyleixa, 2019-11-02:
this one?
https://github.com/me-no-dev/AsyncTCP

atanisoft, 2019-11-02:
Yup

dsyleixa, 2019-11-02:
WebServer server(80);
error: WebServer  does not neme a type

ESPAsyncWebServer server(80);
error: ESPAsyncWebServer does not neme a type

finally:
AsyncWebServer server(80);
```
using the type

     AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload, ArBodyHandlerFunction onBody);

                              ^

D:\arduino\portable\sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:426:30: note:   candidate expects 5 arguments, 2 provided

Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Nicht benutzt: D:\arduino\libraries\WiFi
Bibliothek WiFi in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi  wird verwendet
Bibliothek ESPAsyncWebServer in Version 1.2.3 im Ordner: D:\arduino\portable\sketchbook\libraries\ESPAsyncWebServer  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\FS  wird verwendet
Bibliothek AsyncTCP in Version 1.1.1 im Ordner: D:\arduino\portable\sketchbook\libraries\AsyncTCP  wird verwendet
exit status 1
'class AsyncWebServer' has no member named 'hasArg'
```

atanisoft, 2019-11-02:
Swap hasArg for hasParam and if you call arg() swap it for param().

dsyleixa, 2019-11-02:
exit status 1
'class AsyncWebServer' has no member named 'hasArg'

exit status 1

dsyleixa, 2019-11-02:
exit status 1
'class AsyncWebServer' has no member named 'hasArg'

exit status 1
'class AsyncWebServer' has no member named 'hasParam'

dsyleixa, 2019-11-02:
exit status 1
'class AsyncWebServer' has no member named 'hasArg'

exit status 1
'class AsyncWebServer' has no member named 'hasParam'

exit status 1
'class AsyncWebServer' has no member named 'Arg'

dsyleixa, 2019-11-02:
perhaps it woud b easier to fix the WebServer.h bugs... :-/

atanisoft, 2019-11-02:
@dsyleixa not likely. There are params passed into the callback handler and the methods you are looking for are not on the server but the request that is passed into the handler.

dsyleixa, 2019-11-03:
so you suspect there are issues about my program for either platform?
But why does it work with esp8266 + esp8266webserver.h  though?
And how to achieve then code compatibility between  esp8266WebServer.h and (esp32) WebServer.h?

atanisoft, 2019-11-03:
I don't think it is your code exactly. There is an issue of some sort in the WifiClient code as seen from the debug log message that has yet to be identified likely.

The shift to EspAsyncWebServer though is a better option in a number of ways though as it offloads a lot of the processing for both of your target platforms.

dsyleixa, 2019-11-04:
as both are standard ESP32 libs coming with the ESP AArduino API: : any ideas about 
WebServer.h and /or WifiClient.h
to make them compatible to related esp8266 libs, for code compatibility?

rel1ct, 2019-11-05:
It remains to answer the question: - Who needs it?
What does the alternative(AsyncWebServer) do not suit you?
Make a simple test of the firmware download speed in AsyncWebServer and WebServer. You will be surprised.
created by **@me-no-dev**
```
#include "WiFi.h"
#include "WebServer.h"
#include "ESPAsyncWebServer.h"
#include "Update.h"
char* ssid = "********";
char* pass = "********";
WebServer server(80);
AsyncWebServer server1(81);
const char* serverIndex = "<form method='POST' action='/update' enctype='multipart/form-data'>"
                          "<input type='file' name='update'><br><br>"
                          "<input type='submit' value='Update'>"
                          "</form>";
unsigned long t_start,t_stop;

double calcSpeed(unsigned long ms, size_t len){
  return (double)(len * 125) / (double)(ms * 16);
}

void setup(){
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
//------------   WebServer.h
  server.on("/update", HTTP_GET, []() {
    server.sendHeader("Connection", "close");
    server.send(200, "text/html", serverIndex);
  });
  server.on("/update", HTTP_POST, []() {
    server.sendHeader("Connection", "close");
    server.send(200, "text/html", serverIndex);
    //delay(500);
    //ESP.restart();
  }, []() {
    HTTPUpload& upload = server.upload();
    if (upload.status == UPLOAD_FILE_START) {
      t_start = millis();
      Serial.println("*** Test upload WebServer.h ***");
      Serial.printf("Upload: %s\n", upload.filename.c_str());
    }
    if (upload.status == UPLOAD_FILE_WRITE) {
      //
    }
    if (upload.status == UPLOAD_FILE_END) {
      t_stop = millis();
      Serial.print("SDK: "); Serial.println(ESP.getSdkVersion());
      Serial.print("Time UPLOAD: "); Serial.print((t_stop - t_start) / 1000); Serial.println(" sec.");
      Serial.print("Speed UPLOAD: "); Serial.print(calcSpeed(t_stop - t_start, upload.totalSize)); Serial.println(" Kbit/s");
      Serial.printf("Upload Success: %u bytes\n", upload.totalSize);
    }
  });
//------------   ESPAsyncWebServer.h
  server1.on("/update", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/html", serverIndex);
  });
  server1.on("/update", HTTP_POST, [](AsyncWebServerRequest *request){
    request->send(200, "text/html", serverIndex);
    //delay(500);
    //ESP.restart();
  },[](AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
    if(!index){
      t_start = millis();
      Serial.println("*** Test upload ESPAsyncWebServer.h ***");
      Serial.printf("Upload Start: %s, %lu\n", filename.c_str(), millis());
    }
    if(final){
      t_stop = millis();
      Serial.print("SDK: "); Serial.println(ESP.getSdkVersion());
      Serial.print("Time UPLOAD: "); Serial.print((t_stop - t_start) / 1000); Serial.println(" sec.");
      Serial.print("Speed UPLOAD: "); Serial.print(calcSpeed(t_stop - t_start, index+len)); Serial.println(" Kbit/s");
      Serial.printf("Upload Success: %u bytes\n", index+len);
    }
  });
  
  server.begin();
  server1.begin();
}

void loop(){
  server.handleClient();
}
```

dsyleixa, 2019-11-05:
Re: who needs it:
people programming the esp8266 and already have lots of code running with the satandard esp8266 libs,
and now want to migrate their code to esp32.

OTOH, ESPAsyncWebServer.h is no espressif/arduino-esp32 standard lib

rel1ct, 2019-11-07:
Your code for AsyncWebServer.
Try to run.
```cpp
#include "WiFi.h"
#include "WiFiClient.h"
#include "ESPAsyncWebServer.h"

const char* ssid = "WLAN";
const char* pass = "12345657";

AsyncWebServer server(80);

const int tableWidth = 10;
const int tableHeight = 10;
const String fillChar = "X";
bool matrix[tableHeight*tableWidth]={0};
volatile static bool toChange = 0;
volatile static bool toLearn = 1;
volatile static bool toDetect = 0;
volatile static bool toReset = 0;
volatile static bool toSave = 0;
volatile static bool toClear = 0;
String body = "";
String s1="<style>\nform{margin:0}\n#un{width:0;height:0;visibility:hidden}\n#bu{width:10em;height:2em;}\n#rw{display:table-row;}\n#ce{display:table-cell;}\n#ip{width:1.5em;text-align:center;font-weight:bold}\n</style>\n<script>dge=document.getElementById; function oc(ob){if(ob.value=='')ob.value='"+fillChar+"';else ob.value='';}</script>\n<div role=\"presentation\" style=\"display:table\"><div id='rw'><div id='ce' style=\"padding-right:1em\"><form method=\"get\"><div role=\"presentation\" style=\"margin-bottom:1em;display:table\">";
String s2="<div id=\"rw\">";
String s3="<div id=\"ce\"><input onclick=\"oc(this);\" type=\"text\" name=\"b";
String s3_1="\" value=\"";
String s3_2="\" maxlength=\"1\" id=\"ip\"></div>";
String s4="</div>";
String s5="</div><div id='ce'><input id=\"bu\" type=\"submit\" value=\"Change:Ok\"><p></form>";
String s5_0="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Clear\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"clear\"></form>"; 
String s5_1="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Learn\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"learn\"></form>";
String s5_2="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Detect\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"detect\"></form>";
String s5_3="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Reset\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"reset\"></form>";
String s5_4="<form method=\"get\"><input id=\"bu\" type=\"submit\" value=\"Save\"><input id=\"un\" type=\"text\" name=\"bn\" value=\"save\"></form></div></div></div>";
String par="b";

float bufferArray[100];
uint32_t timestamp=0, sec=0;
uint8_t modeLearn=0, modeDetect=0;

void WiFiEvent(WiFiEvent_t event){
  switch(event){
    case SYSTEM_EVENT_STA_START:
      WiFi.config(IPAddress(192, 168, 2, 99), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
      break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println();
  WiFi.onEvent(WiFiEvent);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  Serial.println("");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  memset(matrix , 0, sizeof(matrix));
  server.on("/", work);
  server.begin();
  Serial.println("Web server started!");
}

void loop() {  
  if(toClear) {
    Serial.println((String)"toClear="+toClear);
    memset(matrix , 0, sizeof(matrix));
    toClear=0;
  }
  if (toChange) {
    Serial.println("new data:");
    for (int i = 0; i < 100; i++) {
       bufferArray[i] = (float)matrix[i];
       Serial.println((String)i + ": " + bufferArray[i]);
    }
    toChange=0;
  }
  if(toLearn){       
    modeLearn=1;
    modeDetect=0;
    Serial.println((String)"modeLearn="+modeLearn);
    toLearn=0; 
  }
  else 
  if(toDetect) {      
    modeLearn=0;
    modeDetect=1;
    Serial.println((String)"modeDetect="+modeDetect);
    toDetect=0;  
  }
  if(toReset) {
    toLearn=0;
    toDetect=1;
    Serial.println((String)"toReset="+toReset);
    toReset=0;
  }
  if(toSave) {
    toLearn=0;
    toDetect=1;
    Serial.println((String)"toSave="+toSave);
    toSave=0;
  }
  if(millis()-timestamp >= 1000) {
    sec++;
    timestamp +=1000;
    Serial.println(sec);
  }
}

void work(AsyncWebServerRequest *request){
  int n = 0; 
  if(request->arg("bn")){
    if(request->arg("bn") == "clear") toClear=1; else toClear=0;
    if(request->arg("bn") == "save") toSave=1; else toSave=0;
    if(request->arg("bn") == "reset") toReset=1; else toReset=0;
    if(request->arg("bn") == "detect") toDetect=1; else toDetect=0;
    if(request->arg("bn") == "learn") toLearn=1; else toLearn=0;      
  }
  for(int x=0; x < tableHeight; x++){
    for(int j=0; j < tableWidth; j++){
      if(request->arg(par+String(n))){
        String a = request->arg(par+String(n));
        bool ic = matrix[n];
        if (a == "") matrix[n] = 0; else matrix[n] = 1; 
        if (ic != matrix[n]) toChange = 1;
        n++;
      }
    }
  }
  String v; body = s1; n = 0;
  for(int x=0; x < tableHeight; x++){
    body += s2;
    for(int j=0; j < tableWidth; j++){ 
      if (matrix[n]) v = fillChar; else v = "";
      body += s3 + String(n) + s3_1 + v + s3_2;
      n++;
    }
    body += s4;
  }
  body += s4+s5+s5_0+s5_1+s5_2+s5_3+s5_4;
  request->send(200, "text/html", body);
}
```

dsyleixa, 2019-11-07:
```
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/pthread" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/sdmmc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/smartconfig_ack" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/soc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spi_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spiffs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcp_transport" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcpip_adapter" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ulp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/vfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wear_levelling" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wifi_provisioning" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wpa_supplicant" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/xtensa-debug-module" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-face" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32-camera" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/in [... cut ...]
Bibliothek WiFi in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi  wird verwendet
Bibliothek WebServer in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer  wird verwendet
Bibliothek ESPmDNS in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\FS  wird verwendet
exit status 1
'this_ip' was not declared in this scope
```

dsyleixa, 2019-11-07:
IMO AvancedWebServer is no option, too many code incompatibilities.

rel1ct, 2019-11-07:
1. Sketch uses 728362 bytes (55%) of program storage space. Maximum is 1310720 bytes.
Global variables use 40220 bytes (12%) of dynamic memory, leaving 287460 bytes for local variables. Maximum is 327680 bytes.

2. Good luck in porting the code, no one will do this except you.
Judging by the activity, this is not interesting to anyone (no need).
I also used WebServer, but switched to AsyncWebServer, I do not regret it.

dsyleixa, 2019-11-07:
I get just errors using ESPAsyncWebServer, see above.
And as to post https://github.com/espressif/arduino-esp32/issues/3444#issuecomment-549122525 it probably doesn't concern WebServer but WifiClient.h

rel1ct, 2019-11-07:
You have posted the code.
You were offered (atanisoft) an alternative option (AsyncWebServer).
I rewrote your code for AsyncWebServer and it works.
I have no idea about the code that you are trying to compile.

dsyleixa, 2019-11-07:
to me the code  https://github.com/espressif/arduino-esp32/issues/3444#issuecomment-551175944
don't work, I posted the error messages.
https://github.com/espressif/arduino-esp32/issues/3444#issuecomment-551177992

atanisof stated, that it's probably a WbCleint.h isuue anyway, not a Webserver.h issue.
The code works fine though with esp8266 libbs for my esp8266.

rel1ct, 2019-11-07:
Search and fix bugs in the code that bother you, but no one will.

dsyleixa, 2019-11-08:
the code is fine, otherwise it wouldn't work with esp8266, as stated  the issue is propbably caused by  one of the esp32 libs.

rel1ct, 2019-11-08:
Logic says the opposite.
WebServer   code from the example = does not work.
AsyncWebServer   code from the example = works.
Problem with esp32 libs or WebServer?
Perhaps WebServer is poorly adapted to work with esp32.

dsyleixa, 2019-11-08:
esp32 WifiClient Code  from my example = does not work.
esp8266 WifiClient Code  from my example = does work.
AsyncWebServer code from the example cannot be compiled, and why to change it if the error is from WiFiClient?

see debug logs!
https://github.com/espressif/arduino-esp32/issues/3444#issuecomment-549046120

dsyleixa, 2019-11-08:
anyway, the issues are about esp32 libraries, either with WebServer or with WebClient, 
but which one ever: it should be fixed...!

atanisoft, 2019-11-08:
@dsyleixa the error you posted doesn't appear to match the code posted by @rel1ct, there is no reference to "this_ip" or "gateway" in the posted code, only in your error output, nor a call to WiFi.config() with a comment after the line. So I don't think you have compiled the posted code but some modified version there of.

As for fixing WiFiClient, feel free to submit a PR with what you find that makes it work so it can be reviewed. I don't know where the issue might be in the code unfortunately.

dsyleixa, 2019-11-08:
sorry guys, I have to apologize and I stand corrected:
there was a c+p mistake inthe tested Async WS code.

Indeed it works now as it was supposed to!

Nonetheless me and perhaps thousands of people have tons of well-working esp8266WebServer program code and  cannot migrate with that code to esp32.

So the TOP issue remains the same:

@esp32 devs:
please make ESP32 WebServer libs compatible to ESP8266!

atanisoft, 2019-11-08:
> please make ESP32 WebServer libs compatible to ESP8266!

From an API perspective I believe they are compatible. If you have specific examples where they are not compatible at the API level please post them. There are a couple known differences in the WiFi library (setHostname vs hostname as example), but I'm not aware of any in the WebServer libraries.

Incompatibility at the functional (runtime) side of things can be expected as the two architectures are vastly different. Many of the issues come down to how the code is used but there are definitely issues (both on esp32 and esp8266) that should be resolved provided someone is able to dig into them and identify the right fix.

dsyleixa, 2019-11-08:
I am just starting with esp32, but all my esp8266 codes are working well.
The code in the TOP post is currently the only one I can show that it does not work with ESP32 WebServer, as shown.

atanisoft, 2019-11-08:
> Nonetheless me and perhaps thousands of people have tons of well-working ESP8266WebServer program code and cannot migrate with that code to esp32.

You might be surprised on that one, a lot of people shifted over to ESPAsyncWebServer simply due to the fact that it offloaded all of the processing from the loop() method (not to mention the increased performance!).

> Compared with ESP8266WebServer, ESPAsyncWebServer is totally weird and cumbersome.

They should be nearly identical at a functional level but ESPAsyncWebServer does provide additional functionality that is not available in either ESP8266WebServer or the ESP32 WebServer code.

dsyleixa, 2019-11-08:
the code snippet 

```
void WiFiEvent(WiFiEvent_t event){
  switch(event){
    case SYSTEM_EVENT_STA_START:
      WiFi.config(IPAddress(192, 168, 2, 99), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
      break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println();
  WiFi.onEvent(WiFiEvent);
...
...
...
void work(AsyncWebServerRequest *request){
  int n = 0; 
  if(request->arg("bn")){
    if(request->arg("bn") == "clear") toClear=1; else toClear=0;
    if(request->arg("bn") == "save") toSave=1; else toSave=0;
    if(request->arg("bn") == "reset") toReset=1; else toReset=0;
    if(request->arg("bn") == "detect") toDetect=1; else toDetect=0;
    if(request->arg("bn") == "learn") toLearn=1; else toLearn=0;      
  }
  for(int x=0; x < tableHeight; x++){
    for(int j=0; j < tableWidth; j++){
      if(request->arg(par+String(n))){
        String a = request->arg(par+String(n));
        bool ic = matrix[n];
        if (a == "") matrix[n] = 0; else matrix[n] = 1; 
        if (ic != matrix[n]) toChange = 1;
        n++;
      }
    }
  }
  String v; body = s1; n = 0;
  for(int x=0; x < tableHeight; x++){
    body += s2;
    for(int j=0; j < tableWidth; j++){ 
      if (matrix[n]) v = fillChar; else v = "";
      body += s3 + String(n) + s3_1 + v + s3_2;
      n++;
    }
    body += s4;
  }
  body += s4+s5+s5_0+s5_1+s5_2+s5_3+s5_4;
  request->send(200, "text/html", body);
}
```

is completely diferent from my esp8266 code, and t's really a mess to have to re-write all code from esp8266 to that weird AsyncWebServer API code.

So the TOP issue remains the same:

@ESP32 devs:
please make ESP32 WebServer libs compatible to ESP8266!

atanisoft, 2019-11-08:
> The code in the TOP post is currently the only one I can show that it does not work with ESP32 WebServer, as shown.

Aside from calling WiFi.config() prior to WiFi.begin() I don't see anything obvious that would explain why it doesn't work as written (for either ESP32 or ESP8266), which I know you tested above and captured debug logs with, which points to issue https://github.com/espressif/arduino-esp32/issues/1921 but there is no consensus on the fix for that one.

dsyleixa, 2019-11-08:
I tesed all your proposed changes about WiFi.config() prior to WiFi.begin() , all remains the same:
https://github.com/espressif/arduino-esp32/issues/3444#issuecomment-549042053

no access to the website by calling the url in Firefox possible!

atanisoft, 2019-11-08:
Until there is a fix for https://github.com/espressif/arduino-esp32/issues/1921 I suspect you won't have it working as you expect. Also note that the ESPAsyncWebServer code works on both ESP8266 and ESP32, though I understand not wanting to rewrite it.

dsyleixa, 2019-11-08:
if  ESP32 WebServer would work the the website would be expected to be accessed, but it can't:
So ESP32 WebServer appearently is broken, and so my request 
@ESP32 devs:
please fix ESP32 WebServer libs and make them compatible to ESP8266!

rel1ct, 2019-11-08:
> please fix ESP32 WebServer libs and make them compatible to ESP8266!

Why don't you want to deal with this problem yourself?
Why should someone solve your problems for you?

dsyleixa, 2019-11-08:
Re: who needs it:
people need it who are already programming the esp8266 and already have lots of code running with the standard esp8266 libs,
and now want to migrate their code to esp32, using code-compatible libs.

OTOH, ESPAsyncWebServer.h is no espressif/arduino-esp32 standard lib and it's functions also are not compatible to standard esp32 WebServer functions and related Arduino examples.

rel1ct, 2019-11-08:
"I do not want to do this, let other people solve my problems for me!"
You trying to say that?
With this position, no one here will help you.

[12 more replies]

closed 2020-02-17 by stale[bot]