espressif/arduino-esp32 #3445: WiFi stack size configuration

gmag11, 2019-11-02:
### Hardware:
Board:							LolinD32
Core Installation/update date:		30/10/2019
IDE name:						Visual Studio 2019
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:

I am developing a project with ESP-NOW that is working as STA and AP too. It seems that stack is getting full as I am getting this error

```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (wifi) 
```

How could I increase stack size for WiFi module?

lbernstone, 2019-11-02:
The code is at https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L92.
It is odd that you are using so much stack, though.  Are you doing anything weird with the object that might allocate something in stack that could be handled with a malloc/new?

gmag11, 2019-11-02:
Probably. I'm porting code from ESP8266 to ESP32. I've tried to increase it but I've got same result. I guess you are right.
I'll try to rewrite code only for ESP32.
Only one question more before closing this: Do you mean it is better to use malloc than static memory allocation? I use some ~250kB buffers to handle ESP-NOW communication and to make it easier to manage I'm doing everything static. Heap is about 250kB free,

gmag11, 2019-11-02:
Output from exception decoder is this:
```
0x400e6eb6: Print::write(char const*) at C:\Users\gmartin\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h line 66
0x400e7061: Print::printNumber(unsigned long, unsigned char) at C:\Users\gmartin\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 271
0x400e7114: Print::print(long, int) at C:\Users\gmartin\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 124
0x400e7128: Print::println(int, int) at C:\Users\gmartin\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 104
0x400d67f9: processRxData(unsigned char*, unsigned char const*, unsigned char, unsigned short, bool) at C:\Users\gmartin\Documents\Arduino\EnigmaIOT_ESP32Gw/EnigmaIOT_ESP32Gw.ino line 292
0x40178576: std::_Function_handler ::_M_invoke(std::_Any_data const&, unsigned char*&&, unsigned char const*&&, unsigned char&&, unsigned short&&, bool&&) at c:\users\gmartin\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x400d32c5: EnigmaIOTGatewayClass::processDataMessage(unsigned char const*, unsigned char const*, unsigned int, Node*) at c:\users\gmartin\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d41d9: EnigmaIOTGatewayClass::manageMessage(unsigned char const*, unsigned char const*, unsigned char) at C:\Users\gmartin\AppData\Local\Temp\arduino_build_530351\sketch\EnigmaIOTGateway.cpp line 719
0x400d4389: EnigmaIOTGatewayClass::rx_cb(unsigned char*, unsigned char*, unsigned char) at C:\Users\gmartin\AppData\Local\Temp\arduino_build_530351\sketch\EnigmaIOTGateway.cpp line 592
0x400d732d: Espnow_halClass::rx_cb(unsigned char*, unsigned char*, unsigned char) at C:\Users\gmartin\AppData\Local\Temp\arduino_build_530351\sketch\espnow_hal.cpp line 57
0x40088b9d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Last call from my code is processRxData function that calls another method to convert mac address to char string.
Code stops just before calling it.

gmag11, 2019-11-04:
I've found what was filling up all stack space. I was creating a static JSONArduino object to hold an array. It is considerably big, around 2 kB. I've noticed how different is memory management in ESP8266 and ESP32, thank you for your comment @lbernstone 

Creating it as dynamic solves the issue, that of course is not related to ESP32 Arduino core.

closed 2019-11-04 by gmag11