espressif/arduino-esp32 #3449: Unable to make HTTPS requests while using BLE at the same time

andreimarinescu, 2019-11-05:
### Hardware:
Board:							ESP32 Dev Kit board
Core Installation version:			Using version 1.0.4 of the ESP32 board with Arduino IDE (also tried 1.0.0 and 1.0.1, 1.0.0 didn't allow for an easy setting of partitions, 1.0.1 exhibited the same issue).
IDE name:						Arduino IDE (via VScode extension)
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Manjaro Linux 5.3.x

Partition scheme is set to huge app (3MB flash, no OTA).

### Description:

I'm trying to poll a HTTPS URL while scanning some BLE devices. The issue is that after 3-6 successful HTTP calls, I constantly keep getting a "connection refused" error from the HTTP client.

I don't even need to start scanning for BLE, I just need to init the BLEDevice. Removing the init call makes requests succesful 100% of the time (tested while running for 5 minutes).

I've created a minimal file outlining how the issue is replicated. I also tried initializing the BLEDevice object in a separate task on core 0, however that only made the "connection refused" messages to show up more quickly (after 1 call usually).

Code is below, based on the BLE and WifiClientSecure examples:

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))

```cpp
#include <Adafruit_NeoPixel.h>
#include <BLEDevice.h>
#include <HTTPClient.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include "secrets.h"

const char* rootCACertificate = \
"-----BEGIN CERTIFICATE-----\n" \
"ROOTCACERT" \
"-----END CERTIFICATE-----\n";

void setClock() {
  configTime(0, 0, "pool.ntp.org", "time.nist.gov");

  Serial.print(F("Waiting for NTP time sync: "));
  time_t nowSecs = time(nullptr);
  while (nowSecs < 8 * 3600 * 2) {
    delay(500);
    Serial.print(F("."));
    yield();
    nowSecs = time(nullptr);
  }

  Serial.println();
  struct tm timeinfo;
  gmtime_r(&nowSecs, &timeinfo);
  Serial.print(F("Current time: "));
  Serial.print(asctime(&timeinfo));
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  esp_log_level_set("*", ESP_LOG_VERBOSE);
  Serial.println("Starting Arduino BLE Client application...");
  Serial.print("setup() running on core ");
  Serial.println(xPortGetCoreID());

  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    
  // wait for WiFi connection
  Serial.print("Waiting for WiFi to connect...");
    while (WiFi.status() != WL_CONNECTED) {
[... 161 lines cut ...]
[HTTPS] begin...
[HTTPS] GET...
[HTTPS] GET... failed, error: connection refused

Waiting 2s before the next round...
loop() running on core 1
192.168.1.9
[HTTPS] begin...
[HTTPS] GET...
[HTTPS] GET... failed, error: connection refused

Waiting 2s before the next round...
loop() running on core 1
192.168.1.9
[HTTPS] begin...
[HTTPS] GET...
[HTTPS] GET... failed, error: connection refused

Waiting 2s before the next round...
loop() running on core 1
192.168.1.9
[HTTPS] begin...
[HTTPS] GET...
[HTTPS] GET... failed, error: connection refused
```

andreimarinescu, 2019-11-05:
After figuring out that I should have enabled verbose logging from the compilation options, I was able to see a complete log of how the request is done. Seems that there's a memory allocation issue:

````
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 87232
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-16) BIGNUM - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -16
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[D][HTTPClient.cpp:1018] connect(): failed connect to <REDACTED_DOMAIN>:443
[W][HTTPClient.cpp:1318] returnError(): error(-1): connection refused
e[D][HTTPClient.cpp:383] disconnect(): tcp is closed

[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
````

chegewara, 2019-11-06:
`BIGNUM - Memory allocation failed`
There is not enough memory to `Performing the SSL/TLS handshake...` and `error(-1): connection refused`.

sachinrana2k11, 2019-11-06:
hi, the BT stack with wifi connectivity is too bulky, it cant fit in the 4mb module. you have to buy a 16mb flash module. or you have to set up the partition in the 4mb module so that these 2 can be fit accordingly in this 4mb module.

andreimarinescu, 2019-11-06:
@sachinrana2k11 thanks. I actually already ran into that problem and modified the partition scheme so that it allocates 3MB of flash for program memory. This solved the issue with storage, but got me into the RAM issues above.

@chegewara I've seen the memory allocation failed message, however given that some requests succeed at the beginning of each boot, I'm wondering if we're not also facing a memory leak here. Is there any good reason why this would work only 4-6 times before running out of memory?
Additionally to the memory leak, would the only fix for having both ble and https functions be to have external PSRAM? I'd have thought it would be a pretty common scenario with an ESP32.

chegewara, 2019-11-06:
Try to add logging to track memory in app.

closed 2020-01-19 by stale[bot]

mefeinberg, 2020-06-25:
I have run into this issue and have a work around.   The issue is that BLE device uses a lot of heap space, so later calls to ssl_connect (and/or other ssl routines) do not have enough heap space.  The workaround I first tried should have worked but didn't.  Essentially I call  BLEDevice::deinit(false); write before I try to make my https request.   This actually works for the https (ssl_connect) request but even though I immediately call    BLEDevice::init(""); to reinitialize the BLEDevice, future scans don't find any BLE device.  The total workaround that works properly is right before I call the deinit I actually release the memory associated with Bluetooth classis.  I do that by       esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT).   For some reason when I do this the https works, and when I then do the init for BLE, the scans work.  So the code looks something like this with print statements to see heap

     Serial.printf("\n****Before BLEDevice::deinit ESP.getFreeHeap() %u\n",ESP.getFreeHeap());
      esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
      BLEDevice::deinit(false);
      Serial.printf("\n****After BLEDevice::deinit ESP.getFreeHeap() %u\n",ESP.getFreeHeap());

Then call https and then after done with ssl 

  if (!(BLEDevice::getInitialized())){
    Serial.printf("Calling BLEDevice::init\n");
    BLEDevice::init("");
    initializeBLEScan(pBLEScan); 
  }
  
Note in my case I didn't need classic bluetooth, so it works for me.  Bottom line the underlying issue is running out of heap space.  Ideally init/deinit sequence would work but release of classic bluetooth memory seems help the init/deinit sequence.  Hope this gives more clarity.  The easy way to reproduce is just watch heap size when making https request when BLE is init'ed and then when not..

gotfredsen, 2020-08-17:
I have a solution running BLE, HTTPS, SPIFSS, and OTA on one standard 4Mb ESP32-WROOM-32. But I do the following:

1. I run BLE first.
2. Then occasionally I upload data using HTTPS. In order to do that, I do a `BLEDevice::deinit(false);` first.
3. I also do a OTA in the same event.
4. And set the RTC clock at the same time using SNTP.
5. Then I do a `deepsleep`, which takes care of memory leaks, etc.
6. To save power I do switch off a lot of stuff all the time.

Before the `BLEDevice::deinit(false);` I had the same BIGNUM problem. I hope this add a bit insight...