espressif/arduino-esp32 #3450: ps_malloc fails when used in a c++ class method

mikemoretti3, 2019-11-05:
If you try to ps_malloc within a c++ class method, it fails because psramInit hasn't been called yet (nor has arduinoInit).

psramInit should actually be called in a class method somewhere at startup instead of inside arduinoInit.  This will allow ps_malloc to be used inside a c++ class method.

### Hardware:
Board:							any esp32 board
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.10
Flash Frequency:					doesn't matter
PSRAM enabled:                                      yes
Upload Speed:					doesn't matter
Computer OS:                                         doesn't matter

### Description:
If you try to ps_malloc within a c++ class method, it fails because psramInit hasn't been called yet (nor has arduinoInit).

psramInit should actually be called in a class method somewhere at startup instead of inside arduinoInit.  This will allow ps_malloc to be used inside a c++ class method.

### Sketch: 
```cpp

//Change the code below by your sketch
#include <Arduino.h>

class test1 {
private:
  static uint8_t *buff;
  static void initTest1(void) {
    buff = (uint8_t *)ps_malloc(128);
    // buff will now be NULL because psramInit was not called by this time
  }
}

void setup() {
}

void loop() {
}
```

### Debug Messages:

lbernstone, 2019-11-05:
If you select a board with PSRAM (ESP32 WROVER module), and set debug to verbose, you should get some info that either indicates psramInit is enabled or failed.

mikemoretti3, 2019-11-05:
It doesn't even get that far.  None of the uart stuff prints no matter what I set the level to.  I think it's because it's in an class method.

stickbreaker, 2019-11-05:
It is my understanding that PsRam can't be accessed until the OS is live. The static methods execute before the OS is live.

mikemoretti3, 2019-11-05:
yeah, i just found that out. all the locking in multi_heap requires the os to be running.

closed 2019-11-05 by mikemoretti3