espressif/arduino-esp32 #3451: WebServer - HTTP request headers array contains only 1 element

jordanj-pl, 2019-11-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
I am trying to read HTTP headers from request. The array always contain 1 element regardless actual number of headers sent with request. Exemplary code is below:

Please note the interesting thing. When printing array of headers only 1 item is available i.e. Authorization but when trying to access Host header by method hostHeader() it returns valid value.

```
class HTTPServer {

  private:
    WebServer *server;

public:
HTTPServer() {
      this->server = new WebServer(80);
      this->server->begin();

      this->server->on("/aaatest", [this]() {

        int argsNum = this->server->args();
        for(int i=0; i<argsNum; i++) {
          Serial.println(this->server->argName(i) + " : " + this->server->arg(i));
        }
        
        int headersNum = this->server->headers();
        Serial.println("Headers num: " + String(headersNum));
        Serial.println("Header [Accept]: " + this->server->header("Accept"));
        Serial.println("Header [Host]: " + this->server->hostHeader());
        Serial.println(this->server->hasHeader("Accept"));
        for(int i=0; i<headersNum; i++) {
          Serial.println(this->server->headerName(i) + " : " + this->server->header(i));
        }

      });

}

}

```

Here is the result:

```
00:38:25.359 -> Headers num: 1
00:38:25.359 -> Header [Accept]: 
00:38:25.359 -> Header [Host]: 10.0.0.165
00:38:25.359 -> 0
00:38:25.359 -> Authorization : 
```

closed 2020-01-19 by stale[bot]

TheRealKasumi, 2022-12-09:
Hey guys, 3 years later this still seems to be an issue or maybe we are using it wrong. I encountered the exact same problem described above with core version 2.0.5.

Looking into this for a few seconds on the side it seems like `_headerKeysCount` is always 1 when `_parseRequest` is called in `Parsing.cpp`. Further down the headers and values are correctly parsed and `_collectHeader` is called with the header and it's value as parameters:

```
bool WebServer::_collectHeader(const char *headerName, const char *headerValue)
{
  for (int i = 0; i < _headerKeysCount; i++)
  {
    if (_currentHeaders[i].key.equalsIgnoreCase(headerName))
    {
      _currentHeaders[i].value = headerValue;
      return true;
    }
  }
  return false;
}
```

However with `_headerKeysCount` being 1 and `_currentHeaders` having one element... we are not getting far here.

TheRealKasumi, 2022-12-09:
Alright, looking further into this, it doesn't seem to be a bug but is intended like this. Headers are not parsed by default, maybe having memory consumption in mind. The following works:

```
#include <WebServer.h>

WebServer srv(80);

const char *headerKeys[2] = {"foo", "bar"};
srv.collectHeaders(headerKeys, 2);

// Now when receiving requests the headers above are parsed correctly
```

It probably would be good to have an example for that. It's not documented I think and can be quite confusing. Reason is, that all args are parsed by default and here the usage is different/unexpected.

WebDust21, 2023-09-18:
> It probably would be good to have an example for that. It's not documented I think and can be quite confusing.

Goodness sakes, what a lifesaver!  Works a treat!
And yes, this is completely undocumented.  (Such a shame that the WebServer is not well documented.)

99% of the examples online are for the AsyncWebServer, not the plain-jane WebServer.