espressif/arduino-esp32 #3452: esp32 uploading multipart data to server

sheryarshahzad, 2019-11-06:
Hi guys I am using esp32 as http client to send multipart data which is basically a jpg file stored in spiffs. I need to send this file to server with the following information.

Host: some url
Content-Type: multipart/form-data
Accept: */*
Content-Disposition: form-data; name="file"; filename="timestamp.jpg"

I am doing this

`
file = SPIFFS.open("/timestamp.jpg");
String debugLogData;

  while (file.available()) {
    debugLogData += char(file.read());
  }
  file.close();

  Serial.println("=====================================");
  Serial.println(debugLogData);
  Serial.println("=====================================");

    HTTPClient http;
 
    http.begin("https://id=some url");
   
    http.addHeader("Content-Disposition", "form-data; name=\"file\"; filename=\"timestamp.jpg\"");
    http.addHeader("Content-Type", "multipart/form-data");             
    http.addHeader("Accept", "*/*");
   

    int httpResponseCode = http.POST(debugLogData); 

    http.end(); 

`

Can anyone suggest something?
Regards,

giobauermeister, 2019-11-13:
I'm also trying to do something similar

radass1996, 2020-01-18:
I have same problem, i was able to POST image from postman but from ESP32 i cant. [qtp1063410741-16] WARN org.eclipse.jetty.util.MultiPartInputStreamParser - Badly formatted multipart request
[qtp1063410741-16] ERROR spark.webserver.MatcherFilter - 
java.io.IOException: Missing initial multi part boundary

closed 2020-04-01 by stale[bot]