espressif/arduino-esp32 #3457: Using arduino-esp32 as a component fails on ESP-IDF v4.1

neilpanchal, 2019-11-07:
### Hardware:
Board:							ESP32-Devkit-C
Core Installation version:			Master branch, `v4.1-dev-815-ga45e99853`
IDE name:						IDF component
Computer OS:                                        macOS Catalina 10.15.1

### Description:
Trying to use arduino-esp32 as a component, using [this](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) guide. 

How to reproduce the errors:

1. Setup the latest master branch of ESP-IDF
2. Make sure hello_world project from ESP-IDF/examples/get-started/hello_world builds
3. Use [this](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) guide to add arduino-esp32 as a component
4. Configure Arduino component to work either as setup/loop functions or without. It doesn't matter.
5. Run `idf.py build` or `make` - both fail, but with different errors:

`make` fails with:
```
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c: In function 'base64_decode_block_signed':
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c:42:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x03f) << 2;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c:43:7: note: here
       case step_b:
       ^~~~
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c:53:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x00f) << 4;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c:54:7: note: here
       case step_c:
       ^~~~
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c:64:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x003) << 6;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/radon/Dev/esp/hello_world/components/arduino/cores/esp32/libb64/cdecode.c:65:7: note: here
       case step_d:
       ^~~~
make[1]: *** No rule to make target `cores/esp32/libb64/cencode.o', needed by `libarduino.a'.  Stop.
make: *** [component-arduino-build] Error 2
```

Running `idf.py build` fails with the following errors:

```Executing action: all (aliases: build)
Running ninja in directory /Users/radon/Dev/esp/hello_world/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Error at /Users/radon/Dev/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  /Users/radon/Dev/esp/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  /Users/radon/Dev/esp/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  /Users/radon/Dev/esp/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/radon/Dev/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja
/usr/local/Cellar/cmake/3.15.5/bin/cmake -S/Users/radon/Dev/esp/hello_world -B/Users/radon/Dev/esp/hello_world/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1```

lbernstone, 2019-11-07:
IDF v4 is not yet supported.  When it is, this will compile.

elad30300, 2019-12-22:
Hi, do you know when support for version 4.x will be released?

atanisoft, 2019-12-22:
@elad30300 there is no timeline set for it yet as there has not been a stable release yet (only alpha and beta release so far). IDF v3.3 will soon be integrated though.

frankdownunder, 2020-02-08:
@atanisoft   Its at release candidate stage now, does that mean work on supporting it might start soon?

atanisoft, 2020-02-08:
@frankdownunder there is a v4.0 branch on this repo that should work with it now. There isn't a timeline for going beyond the current v4.0 rc.

TimeWarDoctor, 2020-03-08:
I tried re-installing esp-idf with v4.0 branch and still got:
CMake Error at C:/Derrick/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'ethernet'.

gpeyton, 2020-04-13:
When will this be resolved? Still have this issue with v4.1
https://esp32.com/viewtopic.php?f=13&t=15148&sid=0b9ca47a32dcccd0f0110dde6f608bc0

cruse123, 2020-06-21:
I tried switching to `idf-release/v4.0` branch and it works with esp-idf `release/v4.0`:
`git checkout -b idf-release/v4.0 remotes/origin/idf-release/v4.0 --`

There are a few more errors during the compilation:
1. `error: 'rtc_gpio_desc' undeclared (first use in this function)`
Solution: Run menuconfig and select `Component config > Driver configurations > RTCI0 configuration -> [*] Support array ``rtc_gpio_desc`` for ESP32`

2. `error: #error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"`
Solution: `Component config -> mbedTLS -> TLS Key Exchange Methods -> [*] Enable pre-shared-key ciphersuites`

PS: Also I think it's to heavy to clone this repo into EVERY esp-idf project, so I use `mklink` (in Windows) and all go smoothly.

LouieAdamian, 2020-08-14:
This is still an issue on idf version 4.0.1 (stable). The full error is
`  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  $IDF_PATH/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  $IDF_PATH/tools/cmake/build.cmake:425 (__build_expand_requirements)
  $IDF_PATH/tools/cmake/project.cmake:348 (idf_build_process)
`

lbernstone, 2020-08-14:
> IDF v4 is not yet supported. When it is, this will compile.

closed 2020-10-28 by stale[bot]

Bodenseematze, 2020-12-08:
The problem still exists with the released esp-idf v4.1 and also with RC of v4.2
So this issue is now present more than a year :-(

Is there any chance that it will ever be fixed?
Is there a workaround to make it work?

chegewara, 2020-12-08:
There is workaround:
- use arduino-esp32 branch idf-release/v4.2 with
- esp-idf master

Yes, i know that idf-release/v4.2 should match esp-idf v4.2 and maybe it will any time soon, but now, due to some API breaking changes in esp-idf, this combination works, with just small CMakeLists.txt changes.

sauvant, 2021-01-29:
> There is workaround:
> 
> * use arduino-esp32 branch idf-release/v4.2 with
> * esp-idf master
> 
> Yes, i know that idf-release/v4.2 should match esp-idf v4.2 and maybe it will any time soon, but now, due to some API breaking changes in esp-idf, this combination works, with just small CMakeLists.txt changes.

Can you please elaborate in how to do that?
Thanks in advance!

chegewara, 2021-01-29:
> > There is workaround:
> > 
> > * use arduino-esp32 branch idf-release/v4.2 with
> > * esp-idf master
> > 
> > Yes, i know that idf-release/v4.2 should match esp-idf v4.2 and maybe it will any time soon, but now, due to some API breaking changes in esp-idf, this combination works, with just small CMakeLists.txt changes.
> 
> Can you please elaborate on how to do that?
> Thanks in advance!

At the moment idf master branch is not usable with arduino as component anymore. There if few breaking changes in esp-idf that are not addressed yet. To use idf-release/v4.2 as a component you need to use commit before https://github.com/espressif/esp-idf/commit/d1222ce158761146fc5a015f178ccb665c3b55b8

Thanks to @atanisoft for finding it.

karelchik, 2021-05-05:
We are struggling to use Arduino-ESP32 as ESP-IDF component. Some of the errors: 
"failed to resolve component 'Ethernet' ", 
"failed to resolve component 'arduino_tinyusb' ".

As far as we are concerned this should be a version problem. Can someone help us out on how to make it work? Maybe @chegewara can comment on this?

chegewara, 2021-05-05:
This is missing tinyusb component:
https://github.com/espressif/esp32-arduino-lib-builder/tree/master/components/arduino_tinyusb
but i have no idea why you are missing ethernet.
I believe @lbernstone knows more

karelchik, 2021-05-05:
Thank you for commenting. To clarify - Ethernet and arduino_tinyusb weren't missing at the same time, sorry for that. Ethernet missing has not been a problem lately, I just thought that it would add some context. 

Anyway, we added the tinyusb component. `idf.py menuconfig` was now successful, yay! But when trying to build the following error occurred
`../components/arduino/cores/esp32/esp32-hal-gpio.h:78:44: error: 'SOC_GPIO_PIN_COUNT' was not declared in this scope`.

The steps we have taken so far:
* using ESP-IDF v4.2.1
* followed steps here: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
* added tinyusb to components

chegewara, 2021-05-05:
You should try esp-idf 4.3

lbernstone, 2021-05-06:
You can see the exact versions used in the 2.0.0alpha release in my [dockerfile](https://github.com/lbernstone/arduino-esp32-dockerfiles/blob/master/Dockerfile-2.0.0a)

AlexeyYukhin, 2021-05-06:
Has anyone managed to build ESP32 idf project with arduino as a component? 

Earlier I used 

platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0

And it worked. But now if I delete downloaded package it will not download it... so our project if in dangerous

my-dudhwala, 2023-09-14:
same error appeared in IDF 4.4.5