espressif/arduino-esp32 #3461: SW_RESET loop

nurmayadi, 2019-11-08:
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
ets Jun  8 2016 00:22:57

martinius96, 2019-11-08:
Without description of problem don't wait reply. 
Go erase flash with esptool and upload your code again.
Use command esptool.py erase_flash

nurmayadi, 2019-11-25:
Thank you, mr. Martinius96,
I managed to overcome the problem.
In addition to removing flash, use the command esptool.py erase_flash.

Some problems may not occur in ESP32, there may be problems with the computer or IDE used to program.

As I experienced, (using the Arduino IDE) the debug program console told us that an error had occurred in esptool.py. although the program can be uploaded, ESP32 does not function properly and instead displays the "SW_RESET" message.

You can try updating the esptool.py library or ESP SDK in the Arduino IDE.
because this might be an error (which I don't know yet) on esptool.py on the Arduino IDE.

This is based on my experience that also uses "mos tools" to program ESP32 and the program runs well, but when I program with Arduino IDE, the problem arises.

it worked on my computer.

closed 2020-02-07 by stale[bot]