espressif/arduino-esp32 #3462: Hii guy could someone help me ? I am new user of esp32  and i always have the following error whenver I upload the code on esp32 board .                          esptool.py v2.6 Serial port COM3 Connecting... Traceback (most recent call last):   File "esptool.py", line 2959, in <module>   File "esptool.py", line 2952, in _main   File "esptool.py", line 2653, in main   File "esptool.py", line 460, in connect   File "esptool.py", line 440, in _connect_attempt   File "esptool.py", line 379, in sync   File "esptool.py", line 322, in command   File "esptool.py", line 285, in write   File "site-packages\serial\serialwin32.py", line 323, in write serial.serialutil.SerialTimeoutException: Write timeout Failed to execute script esptool Une erreur est survenue lors du transfert du croquis

Charost, 2019-11-08:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-11-08:
Please fix your title.  It should be one line.
https://github.com/espressif/arduino-esp32/issues/333#issuecomment-303476651

mrdc, 2019-11-09:
@Charost 
Try to use Google first. It's not arduino-esp32 or Arduino IDE issue.

@lbernstone 

I think he should be banned for this.

me-no-dev, 2019-11-09:
ban is a bit harsh, no?

@Charost the actual error is on the next line that you did not paste... can's help much here

closed 2019-11-09 by me-no-dev

Azkarehman, 2021-01-30:
Im facing same issue :) 

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "AI Thinker ESP32-CAM"

Sketch uses 2100647 bytes (66%) of program storage space. Maximum is 3145728 bytes.

Global variables use 53552 bytes (16%) of dynamic memory, leaving 274128 bytes for local variables. Maximum is 327680 bytes.

esptool.py v2.6

Serial port COM4

Connecting.....

Traceback (most recent call last):

  File "esptool.py", line 2959, in <module>

  File "esptool.py", line 2952, in _main

  File "esptool.py", line 2653, in main

  File "esptool.py", line 460, in connect

  File "esptool.py", line 440, in _connect_attempt

  File "esptool.py", line 379, in sync

  File "esptool.py", line 322, in command

  File "esptool.py", line 285, in write

  File "site-packages\serial\serialwin32.py", line 323, in write

serial.serialutil.SerialTimeoutException: Write timeout

Failed to execute script esptool

An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.