espressif/arduino-esp32 #3465: Timer interrupts within classes

VishalDhayalan, 2019-11-08:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation version:			Arduino-ESP32 v1.04 / Espressif 32 v1.11.1 in Platform.io
IDE name:						Arduino IDE / Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I'm trying to use timer interrupts and alarms inside a class (in the class definition). The function to call is a method in the class. But when passing that into `timerAttachInterrupt()` I'm getting this error: `argument of type "void (Motor::*)()" is incompatible with parameter of type "void (*)()"` .

I've tried using <FunctionalInterrupt.h> by passing the method with std::bind(), like it's done in the functional interrupts example but that doesn't work either, giving the error: `no suitable conversion function from "std::_Bind<std::_Mem_fn<void (Motor::*)()> (Motor *)>" to "void (*)()" exists` .

How can timer interrupts be done within classes? I can't make the timer ISR method static as I want it to be object dependant to process the attribute values for the particular object.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <FunctionalInterrupt.h>

class Motor {
  private:
    int pwm_ch, IN1, IN2;
    float _PPR, reduction_ratio;
    volatile unsigned long enc_pulse;
    hw_timer_t * RPM_timer = NULL;

    void IRAM_ATTR encoder_ISR() {
      enc_pulse++;
    }

    void IRAM_ATTR calc_RPM() {
      // RPM calculation code
    }

  public:
    Motor(int pwm_channel, int pwm_pin, int IN1_pin, int IN2_pin, int enc_pin, float PPR, float ratio) {
      pwm_ch = pwm_channel;
      IN1 = IN1_pin;
      IN2 = IN2_pin;
      _PPR = PPR;
      reduction_ratio = ratio;
      enc_pulse = 0;

      ledcSetup(pwm_ch, 5000, 8);
      ledcAttachPin(pwm_pin, pwm_ch);
      pinMode(IN1, OUTPUT);
      pinMode(IN2, OUTPUT);
      coast();

      pinMode(enc_pin, INPUT);
      attachInterrupt(digitalPinToInterrupt(enc_pin), std::bind(&Motor::encoder_ISR, this), FALLING);
      RPM_timer = timerBegin(0, 80, true);
      timerAttachInterrupt(RPM_timer, std::bind(&Motor::calc_RPM, this), true);   // THIS DOESN'T WORK
      timerAlarmWrite(RPM_timer, 1000000, true);
      timerAlarmEnable(RPM_timer);
    }
    
    void forward(int pwm_speed) {
      digitalWrite(IN1, LOW);
      digitalWrite(IN2, HIGH);
      ledcWrite(pwm_ch, pwm_speed);
    }

    void reverse(int pwm_speed) {
      digitalWrite(IN1, HIGH);
      digitalWrite(IN2, LOW);
      ledcWrite(pwm_ch, pwm_speed);
    }

    void coast() {
      digitalWrite(IN1, LOW);
      digitalWrite(IN2, LOW);
      ledcWrite(pwm_ch, 0);
    }
};

Motor drive_R = Motor(0, 23, 18, 19, 35, 11, 34.02);
Motor drive_L = Motor(1, 5, 16, 17, 34, 11, 34.02);

void setup() {
   Serial.begin(115200);
}

void loop() {
   // Some project specific code
}
```

### Debug Messages:
```
src\RC-CONTROL-WORKING.cpp: In constructor 'Motor::Motor(int, int, int, int, int, float, float)':
src\RC-CONTROL-WORKING.cpp:88:78: error: cannot convert 'std::_Bind_helper<false, void (Motor::*)(), Motor*>::type {aka std::_Bind<std::_Mem_fn<void (Motor::*)()>(Motor*)>}' to 'void (*)()' for argument '2' to 'void timerAttachInterrupt(hw_timer_t*, void (*)(), bool)'
       timerAttachInterrupt(RPM_timer, std::bind(&Motor::calc_RPM, this), true);
                                                                              ^
```

VishalDhayalan, 2019-11-12:
Can someone please help with this or at least let me know if it's possible to use timer interrupts within classes as described above? Thanks.

nakanaela, 2020-01-17:
@VishalDhayalan Take a look at https://github.com/espressif/arduino-esp32/tree/master/libraries/Ticker to see if that may help with using timer interrupts with your class. From that library, they have the timer within the class defined as

`esp_timer_handle_t _timer;`

Attachment to the timer is done with

```c++
  esp_timer_create_args_t _timerConfig;  // Timer config struct
  // see ticker.h for explanation on how to pass arguments to the callback
  _timerConfig.arg = reinterpret_cast<void*>(arg);        
  // the callback function, see above
  _timerConfig.callback = callback;
  // How the time is deployer                               
  _timerConfig.dispatch_method = ESP_TIMER_TASK; 
  _timerConfig.name = "Ticker";  // This can be anything, helpful to make it meaningful to your class

  esp_timer_create(&_timerConfig, &_timer); // Pass configuration to the timer 
  // Call should the timer be periodic or not.
  esp_timer_start_periodic(_timer, milliseconds * 1000ULL); 
  //esp_timer_start_once(_timer, milliseconds * 1000ULL);
```

For your class

```c++
#include "esp_timer.h"

esp_timer_handle_t RPM_timer;
void encoder_ISR() { enc_pulse++; }
void calc_RPM() { // RPM calculation code }

// In Motor()
  esp_timer_create_args_t _timerConfig;
  _timerConfig.arg = 0; // If you aren't passing an argument, this should be ok        
  _timerConfig.callback = reinterpret_cast<esp_timer_cb_t>calc_RPM; // Not sure if this is correct
  // per esp_timer.h, only ESP_TIMER_TASK is supported right now
  _timerConfig.dispatch_method = ESP_TIMER_TASK; 
  _timerConfig.name = "calc_RPM"; // name displayed by esp_timer_dump

  pinMode(enc_pin, INPUT);
  attachInterrupt(digitalPinToInterrupt(enc_pin), std::bind(&Motor::encoder_ISR, this), FALLING);

  esp_timer_create(&_timerConfig, &RPM_timer);
  esp_timer_start_periodic(RPM_timer, 1000000); // this should run calc_RPM every second
```

It runs as a task in FreeTOS, but should run the callback pretty reliably every 1e6us in the background without fail. You can see either the Ticker library(part of the arduino-esp32 core) or esp_timer.h for ways to pause, re-start, deactivate or check the running timer. Alternatively, you make be able to include "Ticker.h" and define your timer as 

```c++
Ticker RPM_timer
// from Motor()

RPM_Timer.attach(1000, calc_RPM); // time is taken as milliseconds(uint32_t) or seconds(float)
// there is no microsecond precision with the timer as yet due to lack of implementation in the code. 
```

Hope this helps.

closed 2020-03-31 by stale[bot]

radiantBear, 2023-07-17:
I'm running into the same problem... I created a MRE to test the solution from @nakanaela, but it doesn't work either:

```
#include "esp_timer.h"

class Test {
    public:
        Test(void) {
            esp_timer_handle_t testTimer;
            esp_timer_create_args_t timerConfig;
            timerConfig.arg = 0;
            timerConfig.callback = reinterpret_cast<esp_timer_cb_t>(testFn);
            timerConfig.dispatch_method = ESP_TIMER_TASK;
            timerConfig.name = "Test_Timer";
            esp_timer_create(&timerConfig, &testTimer);
            esp_timer_start_periodic(testTimer, 1000000);
        }

        void testFn(void) {
            Serial.print("Test succeeded! Time elapsed: ");
            Serial.printf("%lu.%03lu sec\n", millis() / 1000, millis() % 1000);
        }
};

void setup() {
    // put your setup code here, to run once:

    Serial.begin(115200);
    while(!Serial);

    Test justATestObject;
}

void loop() {
    // put your main code here, to run repeatedly:
}
```

When I try to compile this, I get the error **`Compilation error: invalid use of member function 'void Test::testFn()' (did you forget the '()' ?)`** 

_What would be the correct way to cast the member function?_

**P.S.**
Interestingly, `attachInterrupt()` (located in [FunctionalInterrupt.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-timer.c)) at least allows the use of lambda function callbacks (making it possible to call a member function by piping`this`), which `timerAttachInterrupt()` does not. I assume this is because that function uses the argument type `std::function<void(void)>` for its callback, while `timerAttachInterrupt()` uses `void (*fn)(void)`, though I don't know why this difference exists. Would it be possible to change this?