espressif/arduino-esp32 #3467: boot crash: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!

jdpnlapperry1, 2019-11-09:
I am setting up a new Arduino ESP32 Feather board with BME680 sensor.  The sample sketches I tried are working, and I have started to adapt the sample code.

My code compiles fine, but crashes with the following output:

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40089019 on core 1

Backtrace: 0x4008cc18:0x3ffb1db0 0x4008ce49:0x3ffb1dd0 0x40089019:0x3ffb1df0 0x4014780a:0x3ffb1e30 0x40147ae6:0x3ffb1e50 0x40136708:0x3ffb1e70 0x40136771:0x3ffb1e90 0x4013dcc9:0x3ffb1eb0 0x4013dcee:0x3ffb1ed0 0x40136567:0x3ffb1ef0 0x400d6bb7:0x3ffb1f10 0x400d14ac:0x3ffb1f80 0x400d7483:0x3ffb1fb0 0x4008932d:0x3ffb1fd0

Rebooting...

I installed the Exception Decoder in the Arduino IDE and get the following:

Decoding stack results
0x4008cc18: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008ce49: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40089019: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
0x4014780a: sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 78
0x40147ae6: sys_arch_protect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 469
0x40136708: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 302
0x40136771: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 398
0x4013dcc9: udp_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 1145
0x4013dcee: udp_new_ip_type at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 1177
0x40136567: sntp_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c line 547
0x400d6bb7: configTime at /Users/awsjames/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-time.c line 55
0x400d14ac: setup() at /Users/awsjames/Desktop/IoT Downloads/bme680test/bme680test.ino line 93
0x400d7483: loopTask(void*) at /Users/awsjames/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x4008932d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Here is the current code:
//INCLUDES FOR ESP32 FEATHER + BME680 TEST TO FUNCTION PROPERLY
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_Sensor.h>
#include "Adafruit_BME680.h"

//ADDITIONAL INCLUDES FOR AWS IOT INTEGRATION
//#include <WiFiAP.h> //Code won't compile with this include file...
#include <WiFiMulti.h>
#include <WiFiUdp.h>
#include <WiFiScan.h>
#include <ETH.h>
#include <WiFiClient.h>
#include <WiFiSTA.h>
#include <WiFiServer.h>
#include <WiFiType.h>
#include <WiFiGeneric.h>
#include <ArduinoJson.h>
#include <AWS_IOT.h>
#include <WiFi.h>
#include "time.h"
#include <string.h>

//CONFIGURATION FOR BME680 SENSOR USING SPI
#define BME_SCK 5
#define BME_MISO 19
#define BME_MOSI 18
#define BME_CS 21
struct tm timeinfo;
[... 156 lines cut ...]
        JSONencoder["Humidity"] = bme.humidity;
        JSONencoder["Gas"] = bme.gas_resistance / 1000.0;

        char JSONmessageBuffer[1000];
        JSONencoder.printTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));
        Serial.println("Sending message to MQTT topic..");
        Serial.println(JSONmessageBuffer);
        
        if(sensor.publish(TOPIC_NAME,JSONmessageBuffer) == 0)
        {        
            Serial.print("Publish Message:");
            Serial.println(JSONmessageBuffer);
        }
        else
        {
            Serial.println("Publish failed");
            tick = 15;
        }
      
    }  
    vTaskDelay(1000 / portTICK_RATE_MS); 
    tick++;
}

I am not a developer and am trying to hack my way through fixing this.  Any help would greatly be appreciated!

lbernstone, 2019-11-09:
You cannot call configTime until WiFi has been initialized.

jdpnlapperry1, 2019-11-09:
lbernstone: Thank You!  Thank You!  Thank You!  That was the issue...

lbernstone, 2019-11-09:
It is there in the traceback if you know how to read it.  Please close the issue.

closed 2020-01-22 by stale[bot]

ro85ac, 2020-10-04:
> You cannot call configTime until WiFi has been initialized.

@lbernstone 

This issue happens also when i am using sim7000g. What should I modify to make it working. 
[SIM7000G_Azure_MQTT.txt](https://github.com/espressif/arduino-esp32/files/5324275/SIM7000G_Azure_MQTT.txt)

lbernstone, 2020-10-04:
https://github.com/espressif/arduino-esp32/commit/f41beb92bf4434f52209bd15655c1b2f2447447f

ro85ac, 2020-10-04:
@lbernstone 

Now I am receiving:
23:41:16.577 -> Info: Initializing SNTP

23:41:16.577 -> assertion "Operating mode must not be set while SNTP client is running" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c", line 600, function: sntp_setoperatingmode
23:41:16.577 -> abort() was called at PC 0x400e7fd7 on core 1

ro85ac, 2020-10-04:
@lbernstone 

I've made the modification you've suggested, but I still receive the same error.

I also added "WiFi.mode(WIFI_MODE_AP);" just before "if (!Esp32MQTTClient_Init((const uint8_t*)connectionString))" and the last thing I see in the monitor is "Info: Initializing SNTP". Probably the Azure MQTT is waiting for the WiFi connection to be ready.

ro85ac, 2020-10-06:
> @lbernstone
> 
> I've made the modification you've suggested, but I still receive the same error.
> 
> I also added "WiFi.mode(WIFI_MODE_AP);" just before "if (!Esp32MQTTClient_Init((const uint8_t*)connectionString))" and the last thing I see in the monitor is "Info: Initializing SNTP".
> 
> PS. I also modified sntp_lwip.c
> 
> I added "#include "tcpip_adapter.h" and
> int SNTP_Init()
> {
> LogInfo("Initializing SNTP");
> _tcpip_adapter_init(); // Should not hurt anything if already inited_
> sntp_setoperatingmode(SNTP_OPMODE_POLL);
> 
> and remains here, just like adding "WiFi.mode(WIFI_MODE_AP);"
> 
> Probably SNTP is waiting for the WiFi connection to be ready and I am using the GSM.

Indeed the problem is not the Azure MQTT client. Although the modem is connected to internet and I obtain an IP, the whole connection process is not completed.