espressif/arduino-esp32 #3472: Guru Meditation Error: Core 0 panic'ed (Load Prohibited). Exception was handled.

OmarAlkassab, 2019-11-11:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		11/Nov/2019
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                      no?
Upload Speed:				        921600
Computer OS:                                         Windows 10

### Description:
I use PCA9555 IO expander with ESP32 module to read Pushbuttons, I make polling to the buttons to know which one is pressed then make the correspondent action. After some times, and randomly, an Exception occurs.

I decoded the exception and the results are in the following snapshot.

Anyone can help?

[image]

stickbreaker, 2019-11-11:
@OmarAlkassab are you accessing `Wire()` from an interrupt context?
The `Load Prohibited` is caused when accessing an illegal memory address, usually caused by a NULL pointer. I have seen this error when people attempt to use `Wire()` from an interrupt. 

`Wire()` is not thread safe.  verify that all access to `Wire()` is serialized.

I believe that you have multiple tasks overwriting each other when they use `Wire()`.  The explosion in `i2cAddQueue()` is usually caused when one task is building an i2c command and another task is executing an I2C operation.

Your semaphore code needs to block access to `Wire()` for complete transactions:
* **write transactoin:**
lock semaphore
Wire.beginTransmission();
Wire.write()
Wire.endTransmission();
unlock semaphore
* **read transaction:**
lock semaphore
Wire.requestFrom();
Wire.read();
unlock semaphore

I'd have to see your code for the exact problem.

Chuck.

OmarAlkassab, 2019-11-11:
Dear @stickbreaker,

I don’t use it from an interrupt context.
I use xSemaphoreTake(baton, portMAX_DELAY) and xSemaPhoreGive(baton), before each call of the the functions of each library (PCA9555 and LCD_I2C libraries). Isn’t the semaphore enough?

stickbreaker, 2019-11-11:
Please post a link to the PCA9555 and the LCD_i2c library.  I'll look through their code and see.

Please post your code so that I can see how you use the libraries.
>offLineLoop()
offLineSetup()
setButtonFlags()

The semaphore has to cover the complete transaction.  I suspect that the semaphore does not cover enough.

Chuck.

OmarAlkassab, 2019-11-11:
Regarding PCA9555 library: https://github.com/nicoverduin/PCA9555
Regarding LCD_I2C library: https://github.com/mathertel/LiquidCrystal_PCF8574

```
SemaphoreHandle_t I2CSemaphoreBaton = xSemaphoreCreateMutex();
PCA9555 expander(0x20, SDA2, SCL2); // SDA2 and SCL2 are the I2C pins used
LiquidCrystal_PCF8574 lcd(0x27); // set the LCD address to 0x27 for a 16 chars and 2 line display
```

```
void setButtonFlags()
{
    for (int i = 0; i < numberOfButtons; i++)
    {
        xSemaphoreTake(I2CSemaphoreBaton, portMAX_DELAY);
        int reading = expander.digitalRead(inputButtons[i]);
        xSemaphoreGive(I2CSemaphoreBaton);
        if (reading != lastButtonState[i])
            lastDebounceTime[i] = millis();
        if (millis() - lastDebounceTime[i] > debounceDelay)
        {
            if (reading != buttonState[i])
            {
                buttonState[i] = reading;
                if (buttonState[i] == HIGH)
                    buttonFlags[i] = HIGH;
            }
        }
        lastButtonState[i] = reading;
    }
}
```

```
void offlineSetup(void* parameter)
{
    EEPROM.begin(EEPROMSize);
    if(!EEPROM.read(settingsConfigurationsAddress))
    {
        saveSettingsConfigurations();
        EEPROM.write(settingsConfigurationsAddress, 1);
        ////Debug.println("settings saved successfully");
    }
    EEPROM.commit();
    loadSettingsConfiguration();
    InitializeIOs();
    InitializeLCD();
    while(true) offlineLoop();
}
```

```
void InitializeIOs()
{
    LoadOutputsStates();

    for (byte i = 0; i < numberOfButtons; i++)
    {
        xSemaphoreTake(I2CSemaphoreBaton, portMAX_DELAY);
        expander.pinMode(inputButtons[i], INPUT_PULLUP);
        xSemaphoreGive(I2CSemaphoreBaton);
    }

    for (byte i = 0; i < numberOfOutputs; i++)
    {
        if (isExpander[i])
        {
            xSemaphoreTake(I2CSemaphoreBaton, portMAX_DELAY);
            expander.pinMode(outputsPins[i], OUTPUT);
            expander.digitalWrite(outputsPins[i], outputsValues[i]);
            xSemaphoreGive(I2CSemaphoreBaton);
        }
        else
        {
            pinMode(outputsPins[i], OUTPUT);
            digitalWrite(outputsPins[i], outputsValues[i]);
        }
    }

    for (byte i = 0; i < numberOfInputs; i++)
    {
        pinMode(inputsPins[i], INPUT_PULLDOWN);
    }
    
}
```

```
void InitializeLCD()
{
  xSemaphoreTake(I2CSemaphoreBaton, portMAX_DELAY);
  Wire.begin(SDA2, SCL2, 400000);
  lcd.begin(LCD_COLUMNS, LCD_ROWS);
  lcd.setBacklight(255);
  lcd.setCursor(0, 0);lcd.print("                    ");
  lcd.setCursor(0, 1);lcd.print("*   Hello World    *");
  lcd.setCursor(0, 2);lcd.print("--------------------");
  lcd.setCursor(0, 3);lcd.print("                    ");
  xSemaphoreGive(I2CSemaphoreBaton);
  delay(3000);
}
```

```
void offlineLoop()
{
    setButtonFlags();
    resolveButtonFlags();
    ResolveOutputs();
    if (needUpdateShowScreen && currentMode == SHOW_MODE)
    {
        needUpdateShowScreen = false;
        UpdateShowScreen(currentShowScreen);
    }
    if (needUpdateControlScreen && currentMode == CONTROL_MODE)
    {
        needUpdateControlScreen = false;
        UpdateControlScreen(currentControlScreen);
    }
    if(millis() > timeNowOffline + 2000)
    {
      timeNowOffline = millis();
      currentShowScreen++;
      currentShowScreen %= numberOfShowScreens;
    }
}
```

I remove unnecessary codes and functions such as "UpdateControlScreen() and UpdateShowScreen" because they are similar to "InitializeLCD()" function, they start with xSemaphoreTake() and end with xSemaphoreGive() functions.

stickbreaker, 2019-11-11:
try testing the return value of xSemaphoreTake();
```c++
if( xSemaphoreTake(I2CSemaphoreBaton, (TickType_t) 100) == pdTRUE ){
  //i2c code
        int reading = expander.digitalRead(inputButtons[i]);
        xSemaphoreGive(I2CSemaphoreBaton);
        if (reading != lastButtonState[i])
            lastDebounceTime[i] = millis();
        if (millis() - lastDebounceTime[i] > debounceDelay)
        {
            if (reading != buttonState[i])
            {
                buttonState[i] = reading;
                if (buttonState[i] == HIGH)
                    buttonFlags[i] = HIGH;
            }
        }
        lastButtonState[i] = reading;
    }
   else {
      Serial.printf("Failed to take Semaphore ");
   }
```

Chuck.

OmarAlkassab, 2019-11-11:
The exception occurred 2 times during one month, so I cannot reproduce the exception now. Anyway, do you have any expected enhancements to my code to enhance it, and see if the exception will occur again!

stickbreaker, 2019-11-12:
Check the return values on the `xSemaphoreTake()` your code assumes it will always succeed. 

verify every `lcd()` and `expander()` command is inside the semaphore. 

I have seen this error reported before, in every instance, `Wire()` was being accessed by multiple threads. If you implement the semaphore to cover all the i2c calls it will work.

Chuck.

OmarAlkassab, 2019-11-12:
I should make while loop for each xSemaphoreTake()? Like the following?
```
While(xSemaphoreTake(I2CSemaphoreBaton, (TickType_t) 100) != pdTRUE)
    delay(1);
```

stickbreaker, 2019-11-12:
@OmarAlkassab 
No, I would not use an endless loop. `xSemaphoreTake()` should succeed within the specified timeout, but if it doesn't you can't access `Wire()`.  In my code example I output the failure to `Serial()` as an error, so that this event is recorded.

You said this error is happening on an irregular basis, I want to see if `xSemaphoreTake()` is failing,  If it is failing something else is the problem, and the "failed" `xSemaphoreTake()` is just a symptom. 

The every time your error "Load Prohibited" in `i2cAddQueue()` has been reported, it has always been traced back to concurrent multithread access  to `Wire()`.

I would enable **ERROR** level logging and capture the `Serial()` output and see if there are prior errors that are actually the genesis of your problem.  An out of memory, or stalled thread, something.

Chuck.

OmarAlkassab, 2019-11-12:
Dear @stickbreaker ,
I will close the issue and test the endless loop with taking care about all I2C functions calls from all threads, and if the exception occurs I will reopen the issue and posting the debug message.
Many thanks for your great support.

closed 2019-11-12 by OmarAlkassab