espressif/arduino-esp32 #3474: package_esp32_index.json not accessible

tjmc, 2019-11-12:
Command to run
`curl -v https://dl.espressif.com/dl/package_esp32_index.json`

```
*   Trying 119.9.91.16...
* TCP_NODELAY set
* Connected to dl.espressif.com (119.9.91.16) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/cert.pem
  CApath: none
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to dl.espressif.com:443 
* stopped the pause stream!
* Closing connection 0
curl: (35) LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to dl.espressif.com:443 
```

diogosm, 2019-11-12:
Same here, so, I can't install the board :(

rdvojmoc, 2019-11-12:
Same issue here. When we can expect a working server?

chegewara, 2019-11-13:
Should be fixed by now.

tjmc, 2019-11-13:
Thanks for the quick response!

closed 2019-11-13 by tjmc

jramonlab, 2019-11-17:
Hello,
I think server is down again.
I get error when setting the ESP32 board on the board manager.
**Error downloading https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json**

jramonlab, 2019-11-18:
Hi,
I´ve found this link in other issue:
[https://dl.espressif.com/dl/package_esp32_index.json](url)

It seems to work. But not sure about the difference between this and the one on the installation instructions.
This is version 1.0.4.