espressif/arduino-esp32 #3476: Not getting (Correct) IP address (DHCP) after switching hard coded AP and re-flashing.

wimmmb, 2019-11-13:
Make your question, not a Statement, inclusive. Include all pertinent information:

I am trying to make a light controller using the ESP32 with a hard coded AP (SSID and PASS).
Setup:
HW: ESP32 module Adafruit, USB powered.
SW, Arduino, latest version and latest 1.04 release ESP.
Simple webserver example code.

Describe what is failing.
If i change my code (AP SSID and PASS) and when it connects it holds the old IP address and the MASK seems very wrong. I am testing with a AP and switching to another or my Phone as an AP.
"SimpleWIFIserver example code " is producing the error every time.

I did try and disconnect, switch off wifi, .. as found in other Issues reported, the only thing helping is the use of the ESP tool and erasing the chip fully before loading the new FW.

Also tested with the full IDE and the IDF example "httpserver/simple" it seems to work, maybe the tooling also erases before flashing the module.

Hope it's a known problem and a easy fix,

Thanks

wimmmb, 2019-11-13:
Debug information:

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
WIFI status = 1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
WIFI status = 1
Reset WiFi memeory [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY

Connecting to WLAN-PUB
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4 timestamp:5013
.[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 10.57.32.78, MASK: 255.255.248.0, GW: 10.57.32.1
WiFi connected IP address: 10.57.32.78 duration 6146
.
WiFi connected.
IP address: 
10.57.32.78

********************************************************
Changing AP SSID and PASS to my Phones and re flash the module
******************************************************

WiFi NOT connected.
IP address: 
0.0.0.0
Cleaning WiFi memeory [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
WIFI status = 1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
WIFI status = 1
Reset WiFi memeory [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Connecting to wim's iPhone
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
..[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4 timestamp:6348
...[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 10.57.32.78, MASK: 255.0.0.0, GW: 10.0.0.1
WiFi connected IP address: 10.57.32.78 duration 7771
.
WiFi connected.
IP address: 
10.57.32.78

*************************************************************************
Result
The same IP (10.57.32.78) and very strange and wrong MASK (255.0.0.0,)
************************************************************************

lbernstone, 2019-11-13:
Do the two APs have the same SSID?  What IP address do you get when connecting to your phone using the IDF example?  The mask does in fact match the gateway address given, so this could just be very weird behavior on your hotspot.
You say you tried disconnect, but provide no code or logs for that.  To erase the stored AP information, you should call `WiFi.disconnect(false, true);` before WiFi.begin.

wimmmb, 2019-11-13:
Thanks for the response,
- The SSID is defferent, first it is WPLAN-PUB, later it is wims Iphone.
- When i use the IDF example and conect to my phone i get : 172.20.10.7, mask: 255.255.255.240, gw: 172.20.10.1
- It also holds for the reverse, starting from my phone AP and moving to the WLAN-PUB the same problem, in that case it holds the IP recieved from the phone AP.

- the code for clearing and erasing i tested, gives no solution:
void wifiClear()
{
    WiFi.disconnect(false,true);// should erase connection information
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    
    WiFi.disconnect(true);                     // disconnects STA Mode
    delay(1000);
    WiFi.softAPdisconnect(true);           // disconnects AP Mode 
    delay(1000);  
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);
    //DO NOT TOUCH
    //  This is here to force the ESP32 to reset the WiFi and initialise correctly.
    Serial.print("WIFI status = ");
    Serial.println(WiFi.getMode());
    WiFi.disconnect(true);
    delay(1000);
    WiFi.mode(WIFI_STA);
    delay(1000);
    Serial.print("WIFI status = ");
    Serial.println(WiFi.getMode());
    // End silly stuff !!!

}
//--------------------------------------------------------------------------------------------------

Strange, tested it now on multiple AP and multiple android / laptop /ide setups, the problem keeps on happening.

lbernstone, 2019-11-13:
Can you try the WiFiMulti example and see if that causes the same issue?

wimmmb, 2019-11-14:
WiFiMulti example has the same problem, i entered first two AP credentials, tested it. disabled one of them in the code, re-flash and the IP problem is the same as above

wimmmb, 2019-11-14:
Also did try and only switch of one AP, and reboot or re-compline and flash, same problem.:
D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 10.57.32.78, MASK: 255.0.0.0, GW: 10.0.0.1
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: wim's iPhone
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 10.57.32.78
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 86:6D:12:8E:7A:E2
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 6
.57.32.78

wimmmb, 2019-11-14:
After a clean with the ESP32 download tool the same example MultiWifi code works:
onnecting to wim's iPhone
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
..[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4 timestamp:6407
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 172.20.10.7, MASK: 255.255.255.240, GW: 172.20.10.1
WiFi connected IP address: 172.20.10.7 duration 6435
.
WiFi connected.
IP address: 
172.20.10.7

lbernstone, 2019-11-14:
A wipe would have been my next suggestion.  For reference, it is easier for most people to use the [esptool](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) included with the board package to wipe the device.  This likely happened because you used a newer version of the code (esp-idf), and then downgraded to what is included in arduino-esp32.

wimmmb, 2019-11-15:
The wipe as explained is the only solution, still helps only ones. After the wipe the bug can be repeated.

lbernstone, 2019-11-15:
Start from a clean version of SimpleWiFiServer.  Add the following just before WiFi.begin:
```
  WiFi.mode(WIFI_STA);
  WiFi.disconnect(false,true);
```
This will remove any dhcp information.  The IP address should also be cleared, so it is hard to understand how that would persist.  I am unable to reproduce your issue, so I can't help much more than that.

wimmmb, 2019-11-20:
Ibernstone, did try again a clean pc and new install and SimpleWifiServer example, same results. after a clean with the ESP download tool it works.
So chaning hard coded the AP will result in not getting a valid IP, it holds the old IP and the mask is very strange indicating the process to get a new IP is not working.
Thanks,

closed 2020-02-02 by stale[bot]

Mellina1999, 2020-02-04:
By entering the code that the serial monitor gives me it shows me that someone, who can help me with that problem??
basic_string::_M_construct null not valid258EAFA5-E914-47DA-95CA-C5AB0DC85B110123456789abcdefABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789+/upgradeWebsocketconnectionSec-WebSocket-AcceptGET HTTP/1.1 Host: Upgrade: websocket Connection: Upgrade Sec-WebSocket-Key: Sec-WebSocket-Version: 13 HTTP/1.1 101httpws://wswss://wsshttps://httpsR @XR @X�@�G @tO @�G @`�@��������������Pb @hb @ConnectionUpgradewebsocketSec-WebSocket-Version13Sec-WebSocket-KeyHTTP/1.1 101 Switching Protocols Sec-WebSocket-Accept: tc @�c @��@�c @�@�c @�@C:\Users\willi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi\src\WiFiClient.cpp[E][%s:%u] %s(): Not enough memory to allocate buffer [E][%s:%u] %s(): %X : %d [E][%s:%u] %s(): fail on fd %d, errno: %d, "%s" [D][%s:%u] %s(): Disconnected: RES: %d, ERR: %d [I][%s:%u] %s(): Unexpected: RES: %d, ERR: %d [E][%s:%u] %s(): socket: %d [E][%s:%u] %s(): select on fd %d, errno: %d, "%s" [I][%s:%u] %s(): select returned due to timeout %d ms for fd %d [E][%s:%u] %s(): getsockopt on fd %d, errno: %d, "%s" [E][%s:%u] %s(): socket error on fd %d, errno: %d, "%s" [E][%s:%u] %s(): connect on fd %d, errno: %d, "%s" connectedflushavailablepeekfillBufferreadwritesetOptionconnect�u @v @D�@�q @�r @`�@ u @s @l� @��@�� @�y @�y @Pt @�u @�s @,�@��@Lo @�o @Xo @��@��@do @|o @po @��@C:\Users\willi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi\src\WiFiGeneric.cpp[W][%s:%u] %s(): Network Event Queue Send Failed! [E][%s:%u] %s(): Network Event Group Create Failed! [E][%s:%u] %s(): Network Event Queue Create Failed! network_event[E][%s:%u] %s(): Network Event Task Start Failed! [W][%s:%u] %s(): WiFi not started [W][%s:%u] %s(): STA has not been started [W][%s:%u] %s(): Neither AP or STA has been started [E][%s:%u] %s(): DNS Failed for %s [D][%s:%u] %s(): Event: %d - %s [W][%s:%u] %s(): Reason: %u - %s [D][%s:%u] %s(): STA IP: %u.%u.%u.%u, MASK: %u.%u.%u.%u, GW: %u.%u.%u.%u [D][%s:%u] %s(): ETH IP: %u.%u.%u.%u, MASK: %u.%u.%u.%u, GW: %u.%u.%u.%u [E][%s