espressif/arduino-esp32 #3480: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

mehrdad987, 2019-11-14:
Hardware:
Board: ESP32 Dev Module
IDE name: Arduino 1.8.10
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Windows 10
CODE Description:

IM USING A 1 SEC INTERRUPT FOR CALLING Some TASK
1- reading sensors data
2- write on sd
3- POST Http to my web site

ERR Description:

with turning off my modem I am getting this guru

`16:11:51.908 -> --APPEND--TASK 1
16:11:52.042 -> --DATA--TASK2
16:11:52.580 -> 11/14/2019 16:11:20
16:11:52.580 -> 2019-11-14
16:11:52.917 -> --CAN--T5
16:11:52.917 -> [D][HTTPClient.cpp:265] beginInternal(): host: *******.net port: 80 url: /api/esp32
16:11:52.917 -> Message appended
16:11:52.917 -> 130101,11/14/2019 16:11:20,32,16,29,1745,,
16:11:52.917 -> 
16:11:52.917 -> --APPEND--TASK 1
16:11:53.055 -> --DATA--TASK2
16:11:53.562 -> 11/14/2019 16:11:21
16:11:53.597 -> 2019-11-14
16:11:53.900 -> --CAN--T5
16:11:53.900 -> Message appended
16:11:53.936 -> 130101,11/14/2019 16:11:21,26,16,29,1703,,,3544,5117,0
16:11:53.936 -> 
16:11:53.936 -> --APPEND--TASK 1
16:11:54.073 -> --DATA--TASK2
16:11:54.442 -> --GPS--T6
16:11:54.577 -> 11/14/2019 16:11:22
16:11:54.577 -> 2019-11-14
16:11:54.915 -> --CAN--T5
16:11:54.915 -> Message appended
16:11:54.915 -> 130101,11/14/2019 16:11:22,25,16,29,1700,,
16:11:54.915 -> 
16:11:54.915 -> --APPEND--TASK 1
16:11:55.017 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
16:11:55.050 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
16:11:55.050 -> [D][HTTPClient.cpp:1003] connect():  connected to ********.net:80
16:11:55.050 -> [D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 113
16:11:55.050 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
16:11:55.050 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
16:11:55.086 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
16:11:55.086 -> [W][HTTPClient.cpp:1288] returnError(): error(-1): connection refused
16:11:55.086 -> Error on sending POST: [D][HTTPClient.cpp:369] disconnect(): tcp is closed
16:11:55.086 -> 
16:11:55.086 -> --HTTP--TASK3
16:11:55.560 -> 11/14/2019 16:11:23
16:11:55.560 -> 2019-11-14
16:11:55.899 -> --CAN--T5
16:11:55.899 -> Message appended
16:11:55.932 -> 130101,11/14/2019 16:11:23,27,16,29,1745,,
16:11:55.932 -> 
16:11:55.932 -> --APPEND--TASK 1
16:11:56.069 -> --DATA--TASK2
16:11:56.546 -> 11/14/2019 16:11:24
16:11:56.546 -> 2019-11-14
16:11:56.919 -> --CAN--T5
16:11:56.919 -> Message appended
16:11:56.919 -> 130101,11/14/2019 16:11:24,23,16,29,1706,,
16:11:56.919 -> 
16:11:56.919 -> --APPEND--TASK 1
16:11:57.055 -> --DATA--TASK2
16:11:57.359 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
16:11:57.359 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
16:11:57.359 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
16:11:57.392 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
16:11:57.392 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
16:11:57.560 -> 11/14/2019 16:11:25
16:11:57.560 -> 2019-11-14
16:11:57.899 -> --CAN--T5
16:11:57.899 -> Message appended
16:11:57.933 -> 130101,11/14/2019 16:11:25,29,16,29,1794,,
16:11:57.933 -> 
16:11:57.933 -> --APPEND--T 1
16:11:58.067 -> --DATA--T2
16:11:58.539 -> 11/14/2019 16:11:26
16:11:58.539 -> 2019-11-14
16:11:58.912 -> --CAN--T5
16:11:58.912 -> Message appended
16:11:58.912 -> 130101,11/14/2019 16:11:26,28,16,29,1700,,
16:11:58.912 -> 
16:11:58.912 -> --APPEND--T 1
16:11:59.048 -> --DATA--T2
16:11:59.520 -> 11/14/2019 16:11:27
16:11:59.520 -> 2019-11-14
16:11:59.691 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
16:11:59.691 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
16:11:59.691 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
16:11:59.691 -> Core 1 register dump:
16:11:59.691 -> PC      : 0x4017b4ec  PS      : 0x00060034  A0      : 0x80081450  A1      : 0x3ffbe780  
16:11:59.725 -> A2      : 0x3ffbec2c  A3      : 0x00000000  A4      : 0x3ffc46ac  A5      : 0x00000000  
16:11:59.725 -> A6      : 0x3ffc46a8  A7      : 0xffffffff  A8      : 0x80081014  A9      : 0x3ffbe750  
16:11:59.725 -> A10     : 0x3ffbee28  A11     : 0x3ffba2c4  A12     : 0x00000001  A13     : 0x00000001  
16:11:59.725 -> A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000007  
16:11:59.725 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
16:11:59.760 -> Core 1 was running in ISR context:
16:11:59.760 -> EPC1    : 0x40083a83  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4017b4ec
16:11:59.760 -> 
16:11:59.760 -> Backtrace: 0x4017b4ec:0x3ffbe780 0x4008144d:0x3ffbe7a0 0x400828b9:0x3ffbe7c0 0x40083a80:0x3ffba1f0 0x4008515f:0x3ffba210 0x4008956d:0x3ffba230
16:11:59.760 -> 
16:11:59.760 -> Rebooting...
`
AND ALSO with turning on my modem(TP-LINK-AC750)

`16:20:22.899 -> --APPEND--T 1
16:20:23.032 -> --DATA--T2
16:20:23.066 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
16:20:23.066 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
16:20:23.066 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
16:20:23.066 -> Core 1 register dump:
16:20:23.066 -> PC      : 0x4017b4ec  PS      : 0x00060034  A0      : 0x80081450  A1      : 0x3ffbe780  
16:20:23.100 -> A2      : 0x3ffbec2c  A3      : 0x00000000  A4      : 0x3ffc46ac  A5      : 0x00000000  
16:20:23.100 -> A6      : 0x3ffc46a8  A7      : 0xffffffff  A8      : 0x80081014  A9      : 0x3ffbe750  
16:20:23.100 -> A10     : 0x3ffbee28  A11     : 0x3ffba2c4  A12     : 0x00000001  A13     : 0x00000001  
16:20:23.100 -> A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
16:20:23.100 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
16:20:23.134 -> Core 1 was running in ISR context:
16:20:23.134 -> EPC1    : 0x40083a81  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4017b4ec
16:20:23.134 -> 
16:20:23.134 -> Backtrace: 0x4017b4ec:0x3ffbe780 0x4008144d:0x3ffbe7a0 0x400828b9:0x3ffbe7c0 0x40083a7e:0x3ffba1f0 0x4008515f:0x3ffba210 0x4008956d:0x3ffba230
16:20:23.134 -> 
16:20:23.134 -> Rebooting...
16:20:23.134 -> ets Jun  8 2016 00:22:57
16:20:23.169 -> 
16:20:23.169 -> rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
16:20:23.169 -> configsip: 0, SPIWP:0xee
16:20:23.169 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:20:23.169 -> mode:DIO, clock div:1
16:20:23.169 -> load:0x3fff0018,len:4
16:20:23.169 -> load:0x3fff001c,len:1100
16:20:23.169 -> load:0x40078000,len:9232
16:20:23.169 -> load:0x40080400,len:6400
16:20:23.169 -> entry 0x400806a8
16:20:23.580 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
16:20:23.613 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
16:20:23.613 -> [I][WiFiMulti.cpp:202] APlistAdd(): [WIFI][APlistAdd] add SSID: HIOTS
16:20:23.613 -> [I][WiFiMulti.cpp:202] APlistAdd(): [WIFI][APlistAdd] add SSID: HIOTS30550
16:20:23.613 -> [I][WiFiMulti.cpp:202] APlistAdd(): [WIFI][APlistAdd] add SSID: your_SSID3
16:20:23.613 -> [I][WiFiMulti.cpp:202] APlistAdd(): [WIFI][APlistAdd] add SSID: your_SSID4
16:20:26.431 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 1 - SCAN_DONE
16:20:26.431 -> [I][WiFiMulti.cpp:72] run(): [WIFI] scan done
16:20:26.431 -> [I][WiFiMulti.cpp:77] run(): [WIFI] 9 networks found
`

IF  any information is needed just tell me
regards

lbernstone, 2019-11-14:
Interrupts must be short.  1 seconds is way too long to hold up processing.  Set a flag or a semaphore in your interrupt and handle the activity in the normal processing loop.

mehrdad987, 2019-11-14:
`hw_timer_t * timer = NULL;

portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR onTimer() {
  
  xTaskCreate( taskOne, "TaskOne",  20000,  NULL,   tskIDLE_PRIORITY,   NULL);     
  
}`

mehrdad987, 2019-11-14:
18:34:23.200 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
18:34:23.200 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
18:34:23.236 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

lbernstone, 2019-11-14:
This error indicated that you are trying to access normal memory spaces from an interrupt context.  Interrupts must be very short and not try to interact directly with the hardware.
If you don't understand this concept, then I recommend you use the [Ticker library](https://github.com/espressif/arduino-esp32/tree/master/libraries/Ticker) instead of trying to use FreeRTOS.

closed 2019-11-26 by mehrdad987