espressif/arduino-esp32 #3482: Secure Connection to Gmail SMTP server fails

Al-Imran, 2019-11-14:
### Hardware:
Board:							WEMOS LOLIN32
Core Installation/update date:		15/Nov/2019 - esp32 v1.0.4
IDE name:						Arduino IDE v1.8.7 and Visual Studio w Visual Micro Ext.
Flash Frequency:					80Mhz
PSRAM enabled:                                      Default
Upload Speed:					921600
Computer OS:                                          Windows 10

### Description:
I can not establish a secure connection to the gmail smtp server. It always fails.

This section is the issue.
`client.connect(_GMailServer, 465)`

// Arduino Code
`#include<esp_wifi.h>	// ESP32 WiFi
#include<WiFiClientSecure.h>

WiFiClientSecure client;

const char* _GMailServer = "smtp.gmail.com";

void setup() {
	Serial.begin(115200);
	delay(10);

	// Connect to WiFi network
	Serial.print("Connecting to ");
	Serial.println(_ssid);

	WiFi.begin(_ssid, _password);
	while (WiFi.status() != WL_CONNECTED) {
		delay(500);
		Serial.print(".");
	}
	Serial.println("");
	Serial.println("My IP address: ");
	Serial.println(WiFi.localIP());

	delay(1000);
	Serial.println("Attempting to connect to GMAIL server"); 
	if (client.connect(_GMailServer, 465) == 1) {
		Serial.println(F("Connected"));
	}
	else {
		Serial.print(F("Connection failed:")); 
		return 0;
	}

        // Rest of the code is irrelevant to my issue.
}

void loop() {
	// Empty
}

`

atanisoft, 2019-11-14:
unless you put in some certs the WiFiClientSecure will always reject the SSL connection as it can't validate the connection.

Al-Imran, 2019-11-14:
How do I obtain the certificate? Thanks @atanisoft

lbernstone, 2019-11-14:
0 is a successful connection.  If there is no cert, it will accept whatever the server hands out.  Turn on verbose debugging.

Al-Imran, 2019-11-14:
@lbernstone It might be a foolish question but how would I do so? What exactly am I looking for?

lbernstone, 2019-11-14:
Use the "ESP32 dev module" board.  Core debug level is in the tools menu.  You aren't looking for anything in particular, it will just show you the SSL transaction (and any errors).

Al-Imran, 2019-11-15:
This is what the verbose output looks like:
`Attempting to connect to GMAIL server
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 265456
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[E][ssl_client.cpp:87] start_ssl_client(): Connect to Server failed!
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.`

Al-Imran, 2019-11-15:
Apparently, the code failed to connect to the server since I was on my organization network. Probably, because some special ports are blocked on the firewall for security purposes. Creating a WiFi hotspot on my phone solved the issue. Thanks all. Hope it will help someone in the future.

closed 2019-11-15 by Al-Imran