espressif/arduino-esp32 #3484: [FR] Include SoftwareSerial.h as required by teemuatlut TMCStepper library

Hedda, 2019-11-15:
[Feature Request] Please include SoftwareSerial.h in Arduino core for ESP32 as it is required by teemuatlut TMCStepper library.

For reference see https://github.com/teemuatlut/TMCStepper/issues/35

SoftwareSerial.h is required by teemuatlut TMCStepper Arduino library for Trinamic TMC stepper motor drivers used by Marlin Firmware (the optimized firmware for RepRap 3D printers based on the Arduino platform, see http://www.marlinfw.org ) as Marlin 2.0 will be supported to run on ESP32).

atanisoft, 2019-11-15:
With three hardware uarts on the esp32 it doesn't make much sense to use SoftwareSerial. Per the issues and PR you linked to that library already supports using a real uart so it would be better to use that approach.

lbernstone, 2019-11-15:
There is a software serial [library](https://github.com/plerup/espsoftwareserial) in the Arduino library manager.

Hedda, 2019-11-19:
@atanisoft UART does not fully replace Serial interface for all scenarios in the world of stepper motors as used by robots, 3D-printers, CNC machines, laser and vinyl cutters, etc..  

Having the option to simply choose if Serial should be used instead of UART and vice versa is a very convenient and user-friendly usability feature.

Just saying that you can use UART instead of Serial is not a solution to this feature request, that can at best only be seen as a workaround. Please don't close this feature request with only that statement as the final answer.

atanisoft, 2019-11-19:
@Hedda the HardwareSerial interface works quite well for almost any application. There are a few usecases for not using it on the esp32 but they are very specialized protocols typically. As @lbernstone mentions there is a SoftwareSerial lib available today in the library manager inside the Arduino IDE so it doesn't necessarily need to be included as part of the arduino-esp32 core and be maintained as such.

me-no-dev, 2019-11-21:
@Hedda at the same time you do not have enough pins for many steppers, so maybe look into TMC2209 and chain them properly (one UART used and each has it's own ID, much like I2C)

me-no-dev, 2019-11-21:
Oh... you can also switch the pins of the UART around and use again only 1

me-no-dev, 2019-11-21:
To clarify why we are against SoftSerial: ESP32 runs on FreeRTOS, which does not guarantee any time for your code. SoftSerial uses delays to send and receive data and the ESP kernel can at any point switch to another task and make the delay incorrect.

closed 2020-02-03 by stale[bot]