espressif/arduino-esp32 #3488: no matching function for call to 'WiFiClass::hostByName - Problems ping my external MySQL database

Janelis, 2019-11-17:
What you are trying to do?
Hey, I'm running Arduino IDE with the official esp32 libraries and i'm getting an error report, failing to compile a section of my code. I'm trying to connect my esp32 to a MySQL I have running on an external server and for that i imagine i need the external IP address so I can log in to it. For that i ran "IPAddress URL = WiFi.hostByName(test);" to try getting the IP and all that.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFi.h>             
#include <MySQL_Connection.h>
#include <MySQL_Cursor.h>
#include <SPI.h>
#include <Dns.h>

IPAddress dns(8, 8, 8, 8);
IPAddress dns2(8, 8, 4, 4);
IPAddress local_IP(192, 168, 1, 184);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 0, 0);

//i've deleted the logins for the mysql and my home wifi, since that isn't necessary, and is working

WiFiClient client;
MySQL_Connection conn((Client *)&client);

 Serial.begin(115200);
 while (!Serial);
 WiFi.config(local_IP, gateway, subnet, dns, dns2);
  Serial.print("WiFi setting configured.\n");

  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
       delay(500);
       Serial.print(".");
  }
    Serial.println("Connected to network");
    IPAddress ip = WiFi.localIP();
    Serial.print("My IP address is: ");
    Serial.println(ip);
Serial.println("Connecting...");
  char test[] = "url for mysql dot com";
  IPAddress URL = WiFi.hostByName(test);
//there's more code on connecting to the MySQL after that but the error is there on the last line
}

```
C:\Users\Janela\Documents\Arduino\TestSQL2\TestSQL2.ino: In function 'void setup()':

TesteSQL2:91:40: error: no matching function for call to 'WiFiClass::hostByName(char [19])'

   IPAddress URL = WiFi.hostByName(teste);
```

Thanks for the help! (I might have some libraries that aren't necessary included there but the code is a WIP, obviously, so don't flame me hahah)

lbernstone, 2019-11-17:
WiFi.hostByName takes two parameters- the string and an IPAddress to hold the result.  The method itself returns an int which provides an error code.

Janelis, 2019-11-17:
Oh, nice!
Tested it and it works. Ty!!

closed 2019-11-17 by Janelis