espressif/arduino-esp32 #3489: esp_restart() after setCpuFrequencyMhz() causes halt after reboot.

ginrei, 2019-11-18:
esp_restart()  sequence seems to affected by setCpuFrequencyMhz() in setup section.
Sometimes restart OK, sometimes it goes quiet after boot.
as the current slightly down, so it may be halt or unexpected sleep.

### Sketch:  
```cpp
#include <stdio.h>
#include "esp_system.h"
#include "esp_spi_flash.h"

void setup() {
    printf("Setup Start...");
    setCpuFrequencyMhz(160);
    printf("...End\n");
}

void loop() {
    // Print chip information
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is ESP32 chip with %d CPU cores, WiFi%s%s, ",
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("CPU Frequency: %dMhz\n", getCpuFrequencyMhz());

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
Setup Start......End
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 0, 4MB external flash
CPU Frequency: 160Mhz
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
Setup Start......End
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 0, 4MB external flash
CPU Frequency: 160Mhz
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
```
monitor suddenly turns quiet here.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Both on Arduino IDE and Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200?
Computer OS:                                        Mac OSX

me-no-dev, 2019-11-21:
why do you think that setCpuFrequencyMhz is causing this? on 1.0.4 tthe bootloaders are compiled actually for 160MHz and setCpuFrequencyMhz is called before setup to set it to whatever you have in the menu. Have you tried selecting 160MHz there and not using setCpuFrequencyMhz? if that shows any difference, then you could be on the right track :)

asetyde, 2019-11-21:
sorry for question , @me-no-dev  is problem use board at 80mhz if you compile at 160mhz ? we use to have less heat

me-no-dev, 2019-11-21:
not a problem

closed 2020-02-03 by stale[bot]