espressif/arduino-esp32 #3492: Sampling Rate is going down in Mutitasking

famop, 2019-11-18:
Hi everyone,

in a project, I got to send sensor data over WiFi while I scan the Sensor with 10kHz simultaneously.
If I scan first an send afterwards, i get sampling rates up to 90kHz. But if I separate the Read and Send tasks on Core 0 and 1 so I can scan "forever" without running out of storage, I just get a maximum sampling rate of 950Hz.
Maybe anyone knows why multitasking slows it down so much?

Here is my Code: 
`#include <Arduino.h>                      //Declaration of extern libraries
#include <CircularBuffer.h>
#include <WiFi.h>

CircularBuffer <uint8_t, 4096*10> Buffer;    //Buffer Declaration of Type uint8_t and size of 1024 places
TaskHandle_t WriteInBuffer;               //Declaration of separate tasks to run on Core 0 and 1
TaskHandle_t ReadFromBuffer;

int analogPin = 32;                       //Declaration of Pin where analog Data is read

const char *ssid = "ESP32ap";             //set WiFi settings
const char *password = "12345678";
WiFiServer wifiServer(80);                //WiFi Server on Port 80

uint8_t value = 0;                        // = value from ADC before Buffer
uint8_t element = 0;                      // = value from ADC after Buffer
int packSize = 4096;                       // = number of Data Points send in 1 Package
int nPacks = 4;                           // = number of Packs send after each other
bool connect = false;                     // = variable neccesary to communicate between Tasks

void WriteIn(void * pvParameters){        //Task to write in buffer
  for(;;){                                //never ending loop
    if (connect == true){                 //read analog value as long as the client is connected
      analogReadResolution(8);
      value = analogRead(analogPin);
      Buffer.unshift(value);              //write at the beginning of the buffer --> [here,3,2,1]
      vTaskDelay(1);
    }
    else{
      vTaskDelay(5);                          //neccesary to not trigger the WatchDog
    }
  }
}

void ReadFrom( void * pvParameters){              //Task to Read from Buffer and Send over WiFi
  for(;;){                                        //never ending loop
    WiFiClient client = wifiServer.available();   //Declaration of client 

    if (client){                                  //if client is connected
      int Bytes = 0;                              // = Bytes send in this session
      int i = 0;                                  // = counter for loop
      connect = true;                             //let other Task know that client is connected
      Serial.println("Client connected");
      while(client.connected()){                  //Maybe this loop is not neccesary?
        while(i < nPacks){                        //while not all Packs are sent
          //Serial.println(Buffer.size());          
          if(Buffer.size() >= packSize){          //if Buffer is bigger than the Pack size, send a Pack
            for (int j = 0; j < packSize; j++){
              element = Buffer.pop();             //get entry fom Buffers end [4,3,2,1(here)] --> FIFO
              Bytes += client.write(element);     //send elements over WiFi
            }
            i ++;                                 //increment the Packs Loop after sending a Pack
            Serial.println("Test");
          }
          vTaskDelay(5);
        }
        client.stop();                            //Disconnect Client after sending all Data 
        connect = false;                          //let other Task know about it
        Serial.println("Client disconnected");
        Serial.print("Bytes sent: ");
        Serial.println(Bytes);                    //number of Bytes sent
        Serial.println(Buffer.size());
        Buffer.clear();
      }
      connect = false;
    }
    vTaskDelay(10);                               //neccesary to not trigger the WatchDog
  }  
}

void setup() {
  Serial.begin(9600);                   //Configuring Serial Monitor
  Serial.println("Configuring access point...");  
  WiFi.softAP(ssid, password);          //opening Access Point

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);                 //Print IP-Adress of ESP
  wifiServer.begin();                   //Start WiFi Server

  xTaskCreatePinnedToCore(
                  WriteIn,              /* Task function. */
                  "WriteInBuffer",      /* name of task. */
                  10000,                /* Stack size of task */
                  NULL,                 /* parameter of the task */
                  1,                    /* priority of the task */
                  &WriteInBuffer,       /* Task handle to keep track of created task */
                  0);

  xTaskCreatePinnedToCore(
                  ReadFrom,             /* Task function. */
                  "ReadFromBuffer",     /* name of task. */
                  10000,                /* Stack size of task */
                  NULL,                 /* parameter of the task */
                  1,                    /* priority of the task */
                  &ReadFromBuffer,      /* Task handle to keep track of created task */
                  1);
}

void loop() {
  
}`

Thanks in advance!

lbernstone, 2019-11-18:
I didn't analyze too closely, but passing variables between cores takes time.  That circular buffer is probably the main source of your issue.  Use a FreeRTOS [queue](https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/system/freertos.html#queue-api) or [ring buffer](https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/system/freertos_additions.html#ring-buffers) for maximum speed.

famop, 2019-11-18:
Ah perfect, thank you @lbernstone !!
I changed it to the FreeRTOS queue and had a sampling rate of about 57 kHz.
Tomorrow I will test the ring buffer. 
You made my day!

lbernstone, 2019-11-18:
If the queue is working for you, stick with that instead of ring buffer.  Ring buffer is really intended for cases where you can't know the size of the incoming data.

closed 2019-11-18 by famop

Sebastiangka, 2020-02-16:
> I didn't analyze too closely, but passing variables between cores takes time. That circular buffer is probably the main source of your issue. Use a FreeRTOS [...]

@Ibernstone, I would like to have a look too. It may help me with my project. Thanks.