espressif/arduino-esp32 #3493: Build files do not work

debendramodi, 2019-11-18:
### Hardware:
Board:							ESP32 WROOM
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                        Windows 7

### Description:
I download the core and build this as a component to the blink project in IDF (without changing any options).  The build completes successfully.  When I replace the *.a files from the Build folder to my Arduino IDE, I am unable to compile my code (bleserver example code).  If I use the *.a files from within the component/arduino folder the code works fine.

I notice that the build files are of different size from the downloaded files within the component folder.  For example, libbt.a within component/arduino/tools/sdk/lib folder is 12 MB whereas the new build file in /build/bt/libbt.a is 8 MB.

Should not the new build files be the same as the files in the component folder ?

Thanks

lbernstone, 2019-11-18:
Arduino core is currently at 1.0.4.  The compiled IDF libraries need to match the expected version.  Download the latest version of arduino-esp32 with board manager, the follow the directions at https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md

debendramodi, 2019-11-18:
Thanks for your prompt response.

I built 1.0.4 and still the same thing.

For example, 
\examples\get-started\blink\components\arduino\tools\sdk\lib\libbt.a is 12,860 KB
The build file \build\bt\libbt.a is 8MB

Should not these be the same ?

lbernstone, 2019-11-18:
Pretty likely an artifact from the msys environment.  On linux, my libraries are nearly identical to the released versions.

debendramodi, 2019-11-19:
Thanks Ibernstone.  Followed the directions as given in lib_builder.md and was able to build this successfully.

Thanks again.

closed 2019-11-19 by debendramodi