espressif/arduino-esp32 #3496: Can I use a keyboard with an ESP32 ? Or a keypad ? How ?

GeorgeFlorian, 2019-11-19:
Hello !

I am interested into connecting a keyboard to an ESP32. I would need to to have all the number keys, an Enter and a Backspace. Much like an ATM keyboard.

I couldn't find anything useful online, other than users saying that is very hard to implement this.

Any thoughts or ideas ? Thank you !

beegee-tokyo, 2019-11-19:
It is old, but might work
[Arduino Playground Keypad](https://playground.arduino.cc/Code/Keypad/)
[Github Keypad](https://github.com/Chris--A/Keypad)
[Arduino Library Info Keypad](https://www.arduinolibraries.info/libraries/keypad)

GeorgeFlorian, 2019-11-19:
> It is old, but might work
> [Arduino Playground Keypad](https://playground.arduino.cc/Code/Keypad/)
> [Github Keypad](https://github.com/Chris--A/Keypad)
> [Arduino Library Info Keypad](https://www.arduinolibraries.info/libraries/keypad)
> 
> **[EDIT] I didn't try the libray!**

Should this work with an ESP ? https://www.banggood.com/Small-2_4GHz-Wireless-Numeric-Keypad-Mini-Suspension-Number-Pad-Keyboard-for-Laptop-PC-p-1412420.html?rmmds=buy&cur_warehouse=CN

From what I've found there can be wired, Bluetooth and WiFi keypads. Is this correct ? Do I simply plug in the keypad to the ESP and it should work with some code ?

GeorgeFlorian, 2019-11-20:
> It is old, but might work
> [Arduino Playground Keypad](https://playground.arduino.cc/Code/Keypad/)
> [Github Keypad](https://github.com/Chris--A/Keypad)
> [Arduino Library Info Keypad](https://www.arduinolibraries.info/libraries/keypad)
> 
> **[EDIT] I didn't try the libray!**

I actually need a bigger keypad, something more like a keyboard. Something that has all the alphanumerical keys, an enter and a backspace.

Can this be done in ESP32 ? Or should I move to a Raspberry Pi Zero ?

I need something like this:
[image]

closed 2020-02-02 by stale[bot]