espressif/arduino-esp32 #3501: getFreeSketchSpace

Adrianotiger, 2019-11-20:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I changed my partition table, to have OTA_0 with 1.5MB and OTA_1 with just 1MB:
[image]

The idea is to create an "update" partition with a small sketch that will update my big sketch on the other partition.

And now the 2 problems on `getFreeSketchSpace` function:
1- It does not return the free space, but the size of the entire partition.
2- It does not return the space of the entire partition, but the space of the next partition.

The output if I call getFreeSketchSpace and getSketchSize is:

> Sketch size: 880 kb
> Free size: 1024 kb
> We are on: OTA 1

But the partition should be OTA 0 and with 1536kb!

So I begun to check the code.

The code on the ESP.cpp core is this one:
### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
uint32_t EspClass::getFreeSketchSpace () {
    const esp_partition_t* _partition = esp_ota_get_next_update_partition(NULL);
    if(!_partition){
        return 0;
    }

    return _partition->size;
}
```

The function name is completly wrong! It would have sense on the Update library. But if you call this function on the ESP, you think about free space after the sketch was uploaded and on the current partition.

**But it returns the available space for a new sketch upload.**

Probably it is too late to change the name or change the function code. But the returned value is not "Free Sketch Space", but "OTA Update Available Space".

To get the space of the active partition, `esp_ota_get_running_partition()`should be called.

atanisoft, 2019-11-20:
@Adrianotiger looks like a rather ugly bug. Feel free to submit a PR to fix the API call (at least) and to return the correct free space for the partition and a new API for getPartitionSize(). Typically the OTA partitions should be the same size, but your usecase seems closer to the factory setup described in the IDF docs, a failsafe bootstrapping app of sorts that can install/config the actual app.

Jeroen88, 2019-11-20:
@atanisoft in my opinion ::getFreeSketchSpace() was ment to determine what size is available for an update. Also in my opinion whether or not the name is correct is less relevant than changing the result of a function that has already been there for quite some time and risking to break sketches that rely on that behavior. The requested functionality should be in a new function with a new name.

atanisoft, 2019-11-20:
@Jeroen88 fair enough, but the name of the function is a bit misleading. Renaming to getOTASketchSpace() would make sense for the current function implementation.

@me-no-dev thoughts?

Jeroen88, 2019-11-20:
@atanisoft I would not break the API. Maybe adding a function with the suggested name and a new function with the new functionality would be a good compromise?

atanisoft, 2019-11-20:
@Jeroen88 sure, that would be fine.

Adrianotiger, 2019-11-20:
@Jeroen88 After I changed the partitions, my application does not work anymore, since I used "ESP.getFreeSketchSpace() - ESP.getSketchSize()" to calculate the remaining space. Now, this will give a negative number (or give something like 4.29GB!!! of free memory), if the OTA_0 is bigger than OTA_1 and the sketch is bigger than the next partition.

Probably the same will happen to other too.

Probably, remove completly this function would be the best solution, if you want a perfect solution. Not just a workaround.

So you just have to implement the new function in your code, since the current name really let you think about free space on the sketch partition.

Jeroen88, 2019-11-20:
@Adrianotiger the function was already there on the ESP8266 and [this](http://arduino.esp8266.com/Arduino/versions/2.1.0-rc2/doc/ota_updates/ota_updates.html) documentation (search for "Basic requirements") led to the introduction of the function as it is now. Feel free to submit a PR to introduce a new function returning the running sketch free space.

Adrianotiger, 2019-11-20:
@Jeroen88 Yes, I see. The purpose was already to find out the available size for an OTA update. No, I will not submit a PR, because I do not know how to solve it. And probably it will not be accepted, if I just rename it to `getOtaSketchSpace()` for example. I also does not know if it is the right place to put new OTA features, since probably all OTA functions should be written in the library [Update](https://github.com/espressif/arduino-esp32/tree/master/libraries/Update).

me-no-dev, 2019-11-21:
this function is a left-over from esp8266. You do not need to know the space between the end of the sketch and the end of the partition on esp32. That space is unusable anyway.
If you want to have a smaller app to pull the actual apps to OTA, then you should add a factory partition for that. It can have a smaller size and will not be overwritten by OTA. Keep OTA partitions the max size that you expect the real firmwares to be.

Adrianotiger, 2019-11-21:
@me-no-dev A factory partition can't be updated. I want to upload a mini-sketch (WiFiManager + link to new bin file) over my smartphone, to update the main sketch (1.3MB). I have different mini-sketches, so I can update the ESP32 with an alpha version, a beta version, test version, stable and so on. 

I understand that this function was already implemented in the esp8266. But when you use this function, you just expect another behaviour, see:  
- https://github.com/espressif/arduino-esp32/issues/3131  

As already written by @Jeroen88, it is possible to find this function in the documentation (for the Update functionality) => it is not a bug. Just an enhancement ;)

me-no-dev, 2019-11-22:
So you want to OTA a small sketch to OTA another sketch? If that is the case, I still do not see why you will need getFreeSketchSpace? You OTA the first sketch from the phone, when it reboots it will pull the big sketch. You do need both OTA partitions to be able to fit the large app though... no way around that ;) It works on 8266 because we do not use partitions there and it's the bootloader that copies the new firmware to the proper start address. I vote for removing any reference to such functions from esp32 or this will drag forever on all future chips as well...

Adrianotiger, 2019-11-22:
I know, the solution is probably not so good, but it was just a case.
I need getFreeSketchSpace (partition size), to know if I am on the right partition.
If (I don't know any reasons, but could happens) the small sketch is on the big partition, it has to copy itself (or download the mini sketch again) on the next partition, before downloading the big sketch on the right partition.

What I want to say is just that I used this `getFreeSketchSpace`-function believing it would return the free space available on the current partition (getSketchSize + getFreeSketchSpace = getPartitionSize), but this was not the case...

mouridis, 2019-11-29:
@Adrianotiger Really sorry for the offtopic but can you please share what is the tool you are using for the partitions with the screenshot on the original post? Googling the window title didn't help.

Adrianotiger, 2019-11-29:
> @Adrianotiger Really sorry for the offtopic but can you please share what is the tool you are using for the partitions with the screenshot on the original post? Googling the window title didn't help.

It is just a private project:
https://github.com/Adrianotiger/ESPToolsGUI

But the conversions of the python scripts will always give anti virus alerts....

closed 2020-02-11 by stale[bot]