espressif/arduino-esp32 #3504: [request] factory reset bootloader

chegewara, 2019-11-21:
Hi folks,
as far as i understand we have few precompiled bootloaders. I have question then, we are close to upgrade to esp-idf v3.3, could we have at least 1 bootloader with enabled factory reset option?
Of course i can build one by myself, that's not a problem, and at the moment i don't need it, but maybe there is more users interested in such option.

Thanks.

me-no-dev, 2019-11-22:
if there are no conequences to the rest of the functionality, I see no reason not to add it as default :)
you can create a PR in the lib builder repo with the new option in the sdkconfig.
https://github.com/espressif/esp32-arduino-lib-builder

closed 2019-11-22 by me-no-dev

chegewara, 2019-11-22:
Hi @me-no-dev 
what version of esp-idf should i use? v3.2 or v3.3?

me-no-dev, 2019-11-22:
3.3 :) I do not plan to keep updating 3.2 ;)

me-no-dev, 2019-11-22:
there is a branch there for 3.3 . use that :)