espressif/arduino-esp32 #3505: Board Manager not showing esp32 to install

usmanshahid001, 2019-11-22:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

chegewara, 2019-11-22:
Try this one:
https://dl.espressif.com/dl/package_esp32_dev_index.json

me-no-dev, 2019-11-22:
don't ... we are trying to move away from dl.espressif.com. the json is hosted on github, if that is down.... the package is down as well

me-no-dev, 2019-11-22:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json this link works fine for me and shows all version (1.0.4 included)

jramonlab, 2019-11-22:
Hi Me-no-dev,

The provided link is down:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

Why is it always down ?
Is it down on purpose for any reason?
Is there any problem on using the one from dl.espressif.com ?

Regards, Ramon.

atanisoft, 2019-11-22:
try this: https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

me-no-dev, 2019-11-22:
have no idea what to tell you.. link is working fine here. It's hosted on github. Maybe check your internet? What happens when you try to open it in a browser?

atanisoft, 2019-11-22:
@me-no-dev the above link is a better link to use and will work more reliably since it is cached via CDN.

jramonlab, 2019-11-22:
> have no idea what to tell you.. link is working fine here. It's hosted on github. Maybe check your internet? What happens when you try to open it in a browser?

It does not load on a browser.

jramonlab, 2019-11-22:
> @jramonlab try this: https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json
> or: https://espressif.github.io/arduino-esp32/package_esp32_index.json

The first link is loaded ok in the browser, the second one (dev) is not .

atanisoft, 2019-11-22:
> The first link is loaded ok in the browser, the second one (dev) is not .

In that case use the one of those that does work. It is the same as the raw link that @me-no-dev suggested *BUT* it sits behind a CDN with cache. Very likely your system is unable to reach the raw link due to a device blocking connections to it somewhere between your system and the GitHub backend servers.

jramonlab, 2019-11-22:
Hi atanisoft, thanks for the info and the link :)

closed 2020-02-04 by stale[bot]

tyeth, 2023-02-21:
Can we promote the boards manager link to *not use* the raw content url, and to use the github pages version instead to benefit from CDN and avoid throttling. My arduino ide v2 is always stalling getting the esp32 package list using the raw url. I've switched now to the github.io version and it flys along with no delay/pausing.

TLDR: (update docs page https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide to use github.io pages link instead of rawcontent.github.com)

VojtechBartoska, 2023-02-23:
Hello @tyeth,

Thanks for your comment.

what do you think @pedrominatel?

Sahil-Kashyap-in, 2023-05-04:
Guys, I'm new to IoT can anyone help me with this? 
I'm actually trying to install the esp32 board manager on the Arduino IDE software with all these links what am I supposed to do to get the esp32 installed on my PC?

tyeth, 2023-05-04:
@Sahil-Kashyap-in 
Best place to start is the docs!
Here's the specific page to add the esp32 board support packages to the arduino IDE for your reference:
https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/installing.html

Sahil-Kashyap-in, 2023-05-05:
> @Sahil-Kashyap-in Best place to start is the docs! Here's the specific page to add the esp32 board support packages to the arduino IDE for your reference: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/installing.html

Thank you @tyeth I actually figured it out by adding the links on the Arduino IDE, btw thanks for the documentation link !