espressif/arduino-esp32 #3506: Can BLE be made to use PSRAM?

thehellmaker, 2019-11-22:
# Hardware:
Board:							?ESP32 Wrover B Dev Module? 
Core Installation version:			1.11.1 (PlatformIO)
IDE name:						?Platform.io?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?yes?
Upload Speed:				      ?921600?
Computer OS:                                        ?Mac OSX?

What you are trying to do?
I am trying to do a BLEDevice.init and would want it to use PSRAM

Describe your system( Hardware, computer, O/S, core version, environment).
I am using ESP WROVEB Dev Kit with 8MB Flash and 8MB PSRam

Describe what is failing.
I have enabled the build flags to use PSRAM in platform.io but still when I try to do 
BLEDevice::init("test"); the system Watchdog times out. and free space is about 50KB
 
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp-wrover-8
framework = arduino
upload_speed = 921600
build_flags =
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO
monitor_speed = 115200
#build_flags = -DLOG_LOCAL_LEVEL=ESP_LOG_INFO
lib_deps =
  ArduinoJson@6.11.1
  AsyncTCP@1.1.1
  Crypto@0.2.0
  ESP Async WebServer@1.2.3
  Ticker-esp32@1.1.0
[platformio]
src_dir=src
boards_dir=boards

Show the EXACT error message(it doesn't work is not enough).
I have looked into multiple other issues like 
https://github.com/espressif/arduino-esp32/issues/1915
https://github.com/platformio/platform-espressif32/issues/185 
https://www.reddit.com/r/esp32/comments/b9zxux/how_to_enable_the_4mb_of_psram_on_the/

chegewara, 2019-11-22:
Yes and no. 
With platformio you can change sdkconfig settings that allow bluetooth to allocate in PSRAM first. This way part of driver will be allocated in external memory.

closed 2020-02-05 by stale[bot]

m-menon, 2020-07-11:
> With platformio you can change sdkconfig settings that allow bluetooth to allocate in PSRAM first

@chegewara is there any notes or documentation that you can point me which does the above? I have enabled PSRAM on the esp32 board but couldn't figure out how to get the Bluetooth to use PSRAM.

chegewara, 2020-07-11:
`BT_ALLOCATION_FROM_SPIRAM_FIRST=y` or `CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y`

You can also try this:
`BT_BLE_DYNAMIC_ENV_MEMORY=y` or `CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y`

m-menon, 2020-07-15:
@chegewara thanks. I have been trying to reduce the memory usage on my [ESP32 (with 4MB PSRAM)](https://sea.banggood.com/LILYGO-TTGO-BTC-Ticker-ESP32-Module-Source-Bitcoin-Price-Ticker-Program-4-MB-SPI-Flash-4-MB-Psram-LCD-Display-p-1303223.html) as it crashes after a few hours of use. I am using AWS IoT (along with BLE library). tls seems to be the main culprit. The free heap keeps getting smaller and smaller with every tls handshake/negotiation.

In the beginning it's like this
```
Total heap: 257756
Free heap: 90092
Total PSRAM: 4194236
Free PSRAM: 4193404
```

And after about a few hours it ends up like this
```
Total heap: 238836
Free heap: 23540
Total PSRAM: 4194236
Free PSRAM: 4193404
```

Also I noticed that the following 2 defines are present in the sdkconfig.h and already set to true. 
```
#define CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST 1
#define CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY 1
```
Probably this is the default setting in the newest versions of platformio-arduino framework. 

The weird thing is this; I tried to change the below define in the same file but there is absolutely no effect. I reduced it to something like 2048 which would have prevented the aws iot from even connecting. The AWS IoT still connects :-)

`#define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 2048`

I tried changing 
`#define CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC 1`
to
`#define CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC 1`

but this seems to have no effect. The free PSRAM still remains the same. I am not sure if this is the right place for this issue as it has more to do with tls / https / mbedtls.

The sdkconfig.h has this define as well. I was wondering if this needs to be changed as well, for the above to work.
`#define CONFIG_SPIRAM_SIZE -1`

ESP32 boards with PSRAM is fairly common these days. This is the [one](https://sea.banggood.com/LILYGO-TTGO-BTC-Ticker-ESP32-Module-Source-Bitcoin-Price-Ticker-Program-4-MB-SPI-Flash-4-MB-Psram-LCD-Display-p-1303223.html) that I [have](https://sea.banggood.com/LILYGO-TTGO-BTC-Ticker-ESP32-Module-Source-Bitcoin-Price-Ticker-Program-4-MB-SPI-Flash-4-MB-Psram-LCD-Display-p-1303223.html).

Can someone point me to some reliable documentation regarding this? I am on this for the last 2 days without success :-D
I am using VS Code with PlatformIO extension (core 4.3.4, Home 3.2.3) and Arduino framework.

Sanketa54, 2020-08-28:
@thehellmaker : Hey, You got this working fine.
How you moved BLE memory to PSRAM ?
It'll be reallly helpful

noukee, 2020-10-26:
Anyone get to move BLE memory to PSRAM? Is It possible?

sanketa-54, 2020-12-04:
Yes, I tried it and was successfull using NimBLE library instead of Bluedroid.
This library reduces the heap and moves the stack memory to psram.
Can refer this : https://github.com/h2zero/NimBLE-Arduino/issues/107