espressif/arduino-esp32 #3508: Is it possible to multiplex additional UART pins?

DeltaVetal26, 2019-11-23:
Hi,

I have an ESP32 DevKit V4.

3 UART buses:
1) Communication with flash memory
2) Communication with Arduino UNO # 1
3) Communication with Arduino UNO # 2

I plan to connect another Arduino UNO # 3,
but the UART contacts are over.

Tell me, is it possible to programmatically create another UART bus using multiplexing? Is it a good idea to handle a lot of UART?

I have an active ESP32 pin schematic (22, 21, 15 pins are free)

(http://joxi.ru/Y2LXM16f7O3d42.png)

stickbreaker, 2019-11-23:
@DeltaVetal26 how close physically are the Uno's to the Esp32?  

If they are within 18inches (30cm) **And** the ESP32 can initiate all comm cycles you could use I2C (`Wire()`) and have all of your UNO's share the I2C bus on pins 21,22 also, please note, the ESP32 is 3.3v and the Arduino-Uno's are 5v so you would need a voltage shifter between then ESP32 and the Uno's. Like an [Adafruit 757](https://www.adafruit.com/product/757).

  Your code for the Uno's would just use `Wire.onRequest()` and `Wire.onReceive()`.  the ESP32 would send commands via `Wire.beginTransmission(), Wire.write(), Wire.endTransmission()` and read results with `Wire.requestFrom()`.

  It is potentially slower than `Serial()`, max I2C is about 40k bytes/second of transfer about 400k baud  `Serial()` equivalent.  

Chuck.

DeltaVetal26, 2019-11-23:
@stickbreaker,

Thanks for the answer!

UNO is not very far from ESP (within 50 cm).

As a level shift, I use a similar circuit on BSS138 transistors.

I wanted to use I2C, as it really reduces the number of wires.

What confused me was the stability of this tire. I found many topics that talked about the timeout problem. If the slave device loses power or other failure during transmission, then the master may freeze in a while loop and only a reboot will help. This is undesirable for my devices.

stickbreaker, 2019-11-23:
Freezing would be dependent on your code. With the ESP32, any I2C transaction has a maximum idle timeout of 13.1ms. If the Esp32 is communicating with a UNO over i2c and the bus freezes (power failure of uno, code failure of uno) for 13.1ms the ESP32's `Wire()` command will fail with a `timeout` error.  

Your ESP32 code would have to handle the error:
```c++
/* from esp32-hal-i2c.h

// External Wire.h equivalent error Codes
typedef enum {
    I2C_ERROR_OK=0,
    I2C_ERROR_DEV,
    I2C_ERROR_ACK,
    I2C_ERROR_TIMEOUT,
    I2C_ERROR_BUS,
    I2C_ERROR_BUSY,
    I2C_ERROR_MEMORY,
    I2C_ERROR_CONTINUE,
    I2C_ERROR_NO_BEGIN
} i2c_err_t;
*/
Wire.beginTransmission(unoSlave);
Wire.write(1);
uint8_t error = Wire.endTransmission();
if( error){ // something failed on I2C comm, 
   if (error == I2C_ERROR_ACK) { // unoSlave did not answer?
   }
   else if (error == I2C_ERROR_TIMEOUT) { // comm timeout ?
   }
   else if (error == I2C_ERROR_BUSY) { // bus is hung, or someother I2C master is usings bus?
       if( !Wire.begin()){ // Failed resetting bus, power fail on Slave is holding bus low
       Serial.printf(" I2C bus is hung probably power failure of slave device");
       }
       else { // bus reset, retry comm
   }
}
else { // comm succeeded
}
```

Chuck.

DeltaVetal26, 2019-11-23:
Thanks for the code!

I will try to connect via the I2C bus.

Will the bus return to work if the UNO (slave) is restored?

stickbreaker, 2019-11-23:
yes,you can use `Wire.busy()` to test it bus is hung:
```c++
if (Wire.busy()) { // bus is in busy state, either hung or other master i2c device is using bus
}
```
to reset the bus just call `Wire.begin()` without any parameters, it will reuse the configured pins and bus speed.  It returns `TRUE` if the bus is successfully reset.

Chuck.

DeltaVetal26, 2019-11-23:
@stickbreaker ,

The issue is resolved, many thanks for the help!

stickbreaker, 2019-11-23:
You could close the issue.

closed 2019-11-23 by stickbreaker