espressif/arduino-esp32 #3509: Serial.print dumps repeat block of past prints periodically.

unihedron, 2019-11-23:
### Hardware:
Board:							FireBeetle-ESP32
Core Installation version:			esp32/1.0.4
Additional Bopards Manager URL:          https://dl.espressif.com/dl/package_esp32_index.json
IDE name:						Arduino IDE  1.8.10
Upload Speed:					115200
Computer OS:                                        Linux Debian

### Description:
Something in the hardware is dumping a repeat repeat block of past serial.prints periodically.

I am trying to test the RepeatTimer example sketch (code below).
Connected serial port to Arduino monitor / Linux Cutecom / Mac CoolTerm.

The example performs a serial.print every one second. However, after a few minutes, a block of repeated text lines gets sent from the ESP32 device to the serial monitor for no apparent reason.

I originally tested this using millis() checking in another sketch to get the one second printouts, then changed to the timer event code in the RepeatTimer example. Both sketches exhibited the same serial print repeat block issue.

The following is an excerpt of the logged data from a serial terminal monitor with received data timestamps enabled:
```
[18:18:31:621] onTimer no. 1663 at 1663017 ms
[18:18:32:617] onTimer no. 1664 at 1664017 ms
[18:18:33:619] onTimer no. 1665 at 1665017 ms
[18:18:34:621] onTimer no. 1666 at 1666017 ms
[18:18:35:617] onTimer no. 1667 at 1667017 ms
[18:18:36:619] onTimer no. 1668 at 1668017 ms
[18:18:37:621] onTimer no. 1669 at 1669017 ms
[18:18:38:617] onTimer no. 1670 at 1670017 ms
[18:18:39:619] onTimer no. 1671 at 1671017 ms
[18:18:40:621] onTimer no. 1672 at 1672017 ms
[18:18:41:617] onTimer no. 1673 at 1673017 ms
[18:18:42:619] onTimer no. 1674 at 1674017 ms
[18:18:43:621] onTimer no. 1675 at 1675017 ms
[18:18:44:617] onTimer no. 1676 at 1676017 ms
[18:18:45:619] onTimer no. 1677 at 1677017 ms
[18:18:46:621] onTimer no. 1678 at 1678017 ms
[18:18:47:617] onTimer no. 1679 at 1679017 ms
[18:18:48:619] onTimer no. 1680 at 1680017 ms
[18:18:49:621] onTimer no. 1681 at 1681017 ms
[18:18:50:617] onTimer no. 1682 at 1682017 ms
[18:18:51:618] onTimer no. 1683 at 1683017 ms
[18:18:52:621] onTimer no. 1684 at 1684017 ms
[18:18:53:617] onTimer no. 1685 at 1685017 ms
[18:18:54:619] onTimer no. 1686 at 1686017 ms
[18:18:55:622] onTimer no. 1687 at 1687017 ms
[18:18:56:617] onTimer no. 1688 at 1688017 ms
[18:18:57:619] onTimer no. 1689 at 1689017 ms
[18:18:58:621] onTimer no. 1690 at 1690017 ms
[18:18:59:617] onTimer no. 1691 at 1691017 ms
[18:19:00:619] onTimer no. 1692 at 1692017 ms
[18:19:01:621] onTimer no. 1693 at 1693017 ms
[18:19:02:617] onTimer no. 1694 at 1694017 ms
[18:19:03:619] onTimer no. 1695 at 1695017 ms
[18:19:04:621] onTimer no. 1696 at 1696017 ms
[18:19:05:617] onTimer no. 1665 at 1665017 ms
[18:19:05:623] onTimer no. 1666 at 1666017 ms
[18:19:05:623] onTimer no. 1667 at 1667017 ms
[18:19:05:623] onTimer no. 1668 at 1668017 ms
[18:19:05:623] onTimer no. 1669 at 1669017 ms
[18:19:05:623] onTimer no. 1670 at 1670017 ms
[18:19:05:623] onTimer no. 1671 at 1671017 ms
[18:19:05:623] onTimer no. 1672 at 1672017 ms
[18:19:05:623] onTimer no. 1673 at 1673017 ms
[18:19:05:623] onTimer no. 1674 at 1674017 ms
[18:19:05:623] onTimer no. 1675 at 1675017 ms
[18:19:05:623] onTimer no. 1676 at 1676017 ms
[18:19:05:623] onTimer no. 1677 at 1677017 ms
[18:19:05:623] onTimer no. 1678 at 1678017 ms
[18:19:05:623] onTimer no. 1679 at 1679017 ms
[18:19:05:623] onTimer no. 1680 at 1680017 ms
[18:19:05:623] onTimer no. 1681 at 1681017 ms
[18:19:05:637] onTimer no. 1682 at 1682017 ms
[18:19:05:637] onTimer no. 1683 at 1683017 ms
[18:19:05:637] onTimer no. 1684 at 1684017 ms
[18:19:05:637] onTimer no. 1685 at 1685017 ms
[18:19:05:637] onTimer no. 1686 at 1686017 ms
[18:19:05:637] onTimer no. 1687 at 1687017 ms
[18:19:05:637] onTimer no. 1688 at 1688017 ms
[18:19:05:637] onTimer no. 1689 at 1689017 ms
[18:19:05:637] onTimer no. 1690 at 1690017 ms
[18:19:05:637] onTimer no. 1691 at 1691017 ms
[18:19:05:637] onTimer no. 1692 at 1692017 ms
[18:19:05:637] onTimer no. 1693 at 1693017 ms
[18:19:05:637] onTimer no. 1694 at 1694017 ms
[18:19:05:637] onTimer no. 1697 at 1697017 ms
[18:19:06:619] onTimer no. 1698 at 1698017 ms
```

From the above log file... The repeating starts at [18:19:05:617] with the large chunk of lines coming in at one time of [18:19:05:623] which was already transmitted previously at [18:18:33:619].

I have tested this by monitoring three separate methods: Linux Arduino 1.8.10 Serial Monitor, Linux Cutecom serial terminal on two separate USB ports as well as Mac Coolterm serial monitor. Showing the incoming timestamps helped identify that a large block of repeated data (not just one line) is coming in one time.

### Sketch:  

I had to change the RepeatTimer example to use IO3 since the Firebeetle uses IO0 for a download switch.

```cpp

/*
 Repeat timer example

 This example shows how to use hardware timer in ESP32. The timer calls onTimer
 function every second. The timer can be stopped with button attached to (IO0).

 This example code is in the public domain.
 */

// Stop button is attached to IO3
#define BTN_STOP_ALARM    3

hw_timer_t * timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

void IRAM_ATTR onTimer(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  isrCounter++;
  lastIsrAt = millis();
  portEXIT_CRITICAL_ISR(&timerMux);
  // Give a semaphore that we can check in the loop
  xSemaphoreGiveFromISR(timerSemaphore, NULL);
  // It is safe to use digitalRead/Write here if you want to toggle an output
}

void setup() {
  Serial.begin(115200);

  // Set BTN_STOP_ALARM to input mode
  pinMode(BTN_STOP_ALARM, INPUT);

  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Use 1st timer of 4 (counted from zero).
  // Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more
  // info).
  timer = timerBegin(0, 80, true);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer, true);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter)
  timerAlarmWrite(timer, 1000000, true);

  // Start an alarm
  timerAlarmEnable(timer);
}

void loop() {
  // If Timer has fired
  if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE){
    uint32_t isrCount = 0, isrTime = 0;
    // Read the interrupt count and time
    portENTER_CRITICAL(&timerMux);
    isrCount = isrCounter;
    isrTime = lastIsrAt;
    portEXIT_CRITICAL(&timerMux);
    // Print it
    Serial.print("onTimer no. ");
    Serial.print(isrCount);
    Serial.print(" at ");
    Serial.print(isrTime);
    Serial.println(" ms");
  }
  // If button is pressed
  if (digitalRead(BTN_STOP_ALARM) == LOW) {
    // If timer is still running
    if (timer) {
      // Stop and free timer    
      Serial.print("Timer stopped.");
      timerEnd(timer);
      timer = NULL;
    }
  }
}
```

lbernstone, 2019-11-23:
IO3 is used by Serial.  There are many, many other pins to choose from. https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap

unihedron, 2019-11-23:
> IO3 is used by Serial. 

I changed to IO27, that did not solve the problem.

stickbreaker, 2019-11-23:
@unihedron 
Your problem is really weird. The HAL layer UART code does not use an output buffer.  The hardware UART has an internal 128 byte FIFO, when you do a `Serial.print()` it transfers bytes from the buffer supplied by `print()` into the FIFO one byte at a time.  If the Fifo fills, it hangs in a loop until space is cleared by the UART sending bytes out.  In your example the block of repeated data is almost 1k bytes, by my count it is 929 bytes of repeated data. 

So, the `Serial()` code cannot be source of the problem, it would seem that the problem is either in `Stream()` or `Print()`. 

Chuck.

unihedron, 2019-11-23:
@stickbreaker
> In your example the block of repeated data is almost 1k bytes, by my count it is 929 bytes of repeated data.
> 
> So, the `Serial()` code cannot be source of the problem, it would seem that the problem is either in `Stream()` or `Print()`.

Thank you for the clues. My FireBeetle-ESP32 board contains an ST 32F042F6P6 micro-controller which I assume is being used for the USB interface. The specs for that part located here: https://www.st.com/en/microcontrollers-microprocessors/stm32f042f6.html#
indicate that the USB implementation contains a 1024 byte buffer.

unihedron, 2019-11-23:
@stickbreaker
I have changed the Additional Boards Manager URL: to the DF Robot suggested link:
  http://download.dfrobot.top/FireBeetle/package_esp32_index.json
and the problem persists.

By doing this, the esp platform has been degraded to 0.0.9:
Using core 'esp32' from platform in folder: ~/.arduino15/packages/esp32/hardware/DFRobot_FireBeetle-ESP32/0.0.9

unihedron, 2019-11-23:
This issue seems to be solved now after installing the specific DFRobot esp platform:
https://dl.espressif.com/dl/package_esp32_index.json

Unfortunately DF Robot has not updated their esp platform from version 0.0.9 and I kept getting:
 Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
I'll follow that up separately.

unihedron, 2019-11-24:
My positive conclusion was premature. The Firebeetle ESP32 v2.0 board still exhibits this same repeating serial output. 
I tried my code on a Wrover board and there was no problem.
I guess it has something to do with the Firebeeltle ESP32 v2.0 extra USB processor being faulty.
I've sent a tech support request to DFRobot about their firebeetle board.

closed 2020-02-07 by stale[bot]