espressif/arduino-esp32 #3510: ESP32-CAM rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) fault

halitekmekcioglu, 2019-11-24:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32-CAM WROVER MODULE
Core Installation/update date:			11/jul/2017
IDE name:							ARDUINO 1.8.9
Flash Frequency:					80 MHZ
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
I've got newly ESP-32CAM and I programmed it in the office for basically start it, everything was worked properly. Then I came home, and I want to connect it to home wifi,  I set up everything for my laptop and I installed same program. Only one thing I know happened, I've connected +5v to directly its 5V pin and board had started to be warm, I realized it and I changed it to 3.3V. After that, I started to get this script on serial monitor.  

### Sketch:  (leave the backquotes for [code formatting]#include "esp_camera.h"

#include "esp_camera.h"
#include <WiFi.h>
#include "soc/soc.h" //disable brownour problems
#include "soc/rtc_cntl_reg.h"  //disable brownour problems

//
//            or another board which has PSRAM enabled
//

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"

const char* ssid = "ssid";
const char* password = "pwd";

void startCameraServer();

void setup() {

  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  //initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    s->set_saturation(s, -2);//lower the saturation
  }
  //drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

Originally, properly working code is above, nothing much changed though. Only SSID and SSID PASSWORD changed.

This, I started to get it from the serial monitor:

### Debug Messages:
```
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

```

Any ideas, advice? Thanks for advance.

nurmayadi, 2019-11-25:
Hi mr. HALIT,

saya memiliki masalah yang hampir sama dengan anda.

saya menggunakan ESP32 Dev KIT, dan memiliki masalah  SW_RESET pada perangkat.

mungkin anda perlu mengupdate ESP SDK atau esptool yang ada pada arduino IDE.

lihat juga pada [https://github.com/espressif/arduino-esp32/issues/3461](url)

halitekmekcioglu, 2019-11-26:
Thanks, problem solved. This link helpful for this issue.
https://randomnerdtutorials.com/esp32-troubleshooting-guide/

closed 2019-11-26 by halitekmekcioglu

Tcoton, 2020-07-05:
I have found that one of my ESP32 modules had this exact issue, setting the flash frequency to 40MHz instead of 80MHz solved the issue, no more reboots.

Cyril995, 2020-07-08:
@Tcoton how do you change 80Mhz to 40Mhz ? any guidance ?

Tcoton, 2020-07-08:
Using Arduino IDE, select the ESP32 Dev Module as a board and a bit below, you will find the Flash Frequency set to 80 MHz by default, set it to 40 MHz and check other settings too. I personally use the following settings on the weirdo ESP32-CAM I have:

Board: ESP32 Dev Module
Upload Speed : 921600 (default)
CPU Frequency: 240MHz (default)
Flash Frequency: 40 MHz
Flash Mode: QIO (default)
Flash size: 4MB (default)
Partition Scheme: Huge APP (3MB No OTA/1MB SPIFFS)
Core Debug Level: None
PSRAM: Enabled

Again, this is specific to my weirdo board and it took me ages to figure why on earth it would not work with any other settings than the Flash Frequency set to 40 MHz, flashing with the AI-Thinker board which offers no settings was making the board to loop reset. The second module I have bought at the very same time at the vey same place did not have this issue at all.

Hishan98, 2021-04-26:
> I have found that one of my ESP32 modules had this exact issue, setting the flash frequency to 40MHz instead of 80MHz solved the issue, no more reboots.

thank you very much

meakdemir, 2021-12-26:
Means it is a HW problem folks, first check your USB to ESP32 cable, try with another one. Using another cable fixed my problem. FYI.

TENGYIHONG, 2022-03-12:
我的ESP32 -cam也出现这样的问题，自己换过flash。上面的方法试过，都无法解决问题。

日志：
Backtrace: 0x400e2c45:0x3ffbbac0 0x400e41e9:0x3ffbbae0 0x40084e3c:0x3ffbbb30 0x400e0bc5:0x3ffbbb60 0x400d11f8:0x3ffbbb80 0x400d1135:0x3ffbbba0 0x400d1379:0x3ffbbbc0 0x400e0d3e:0x3ffbbbf0 0x40089ad5:0x3ffbbc10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e2c40: 0b000000 ffffffff ffffffff
Core 0 register dump:
PC      : 0x400e2c45  PS      : 0x00060030  A0      : 0x800e41ec  A1      : 0x3ffbbac0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ff42034  
A6      : 0x0000001e  A7      : 0x0000001f  A8      : 0x0160a000  A9      : 0x0000000f  
A10     : 0x00000000  A11     : 0x0000000f  A12     : 0x00000004  A13     : 0x3ff42084  
A14     : 0x0000f500  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40087ba4  LEND    : 0x40087bc0  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Xiangyu-Fu, 2022-03-21:
I solved this question by connecting the board also to the 5v power instead of using 3.3v only. I guess 3.3v cannot provide enough power supplement for ESP32.

adeyemisaobaan, 2023-04-29:
i am trying to build a facial recognition trigger project but i keep getting this on the serial monitor :
SOMEONE PLEASE HELP!!!!!!!!!!!

rst: 0xc (SW_CPU RESET), boot: 0x13 (SPI FAST FLASH BOOT) configsip: 0, SPIWP: Oxee
c1k_drv: 0x00, 9_drv: 0x00,d_drv: 0x00, c0_drv: 0x00,hd_drv: 0x00, wp_drv: 0x00
mode: DIO, clock div:1
load:0x3fff0018, len: 4
load:0x3fff001c, len:1216
ho 0 tail 12 room 4
Load: 0x40078000, 1en: 10944
load: 0x40080400, 1en: 6388
entry 0x40080664

KahfiSmith, 2023-11-29:
I had the same error, I checked the esp32 board was the most updated. is there any other solution?

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0

KahfiSmith, 2023-11-29:
> Thanks, problem solved. This link helpful for this issue. https://randomnerdtutorials.com/esp32-troubleshooting-guide/

how? i'm still confused

Tsurematsu, 2024-03-02:
If the error is of any use to me because the cable I used did not provide enough voltage to the microcontroller,

cyberporkexception, 2025-02-12:
If it happens while experimenting with zigbee, maybe a wrong stack is selected in Arduino IDE.
https://github.com/espressif/arduino-esp32/issues/10742