espressif/arduino-esp32 #3518: ESP32 debounce

MaazSk, 2019-11-26:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do? 
Toggle led using inbuilt touch capabilities of the esp32
Describe your system( Hardware, computer, O/S, core version, environment).
Hardware ESP32 Dev Module
O/S windows 10
environment Arduino IDE 1.8.9
Describe what is failing.
Need some help how to debounce the key or read the touch key once

I just want to a help in which I could be able to debounce the touch key.
The touch examples are working perfectly but unable to implement debounce

closed 2020-02-08 by stale[bot]

SuGlider, 2022-01-26:
Check #6194 
There are new examples that will help you.

MaazSk, 2022-01-27:
Hi @SuGlider, 

Thank you for the update but I forgot about this issue and I am no longer working on this project